openvehicles/Open-Vehicle-Monitoring-System-3 #724: MG ZS EV (2020) SOC and estimated km wrong after BMS update

pascalsaul, 2022-08-29:
During the maintenance they updated the software versions as well as the BMS to 1027630EU4 which adds delayed charging and battery heating. Very nice but OVMS can't handle the changes all though the "Updated BMS Firmware" is checked and gave OVMS a reboot. The latest 3.3.002-46-g89963cee/ota_0/edge (build idf v3.3.4-848-g1ff5e24b1 Aug 26 2022 21:55:38) does give the same issues. MG changed the max voltage to be 449.75 instead of 456.25 and SOC changed to 98.9071 which is the new 100%. 

```
Before:
v.b.range.est                            255.5km
v.b.range.full                           263km
v.b.range.ideal                          263km
v.b.range.speed
v.b.soc                                  100%
v.b.soh                                  97.18%
v.b.temp                                 28°C
v.b.voltage                              456.25V

After:
v.b.range.est                            17km
v.b.range.full                           263km
v.b.range.ideal                          260.126km
v.b.range.speed
v.b.soc                                  98.9071%
v.b.soh                                  97.07%
v.b.temp                                 27°C
v.b.voltage                              449.75V

Softver:
DCDC F01R0A7355
ATC REL_DC_BEV_C036     
GWM 0026011080
BCM 2210 4176 
PEPS ZS11SW01
SCU 194136858 C0807
PLC 1770099611
ABS 1019000000
EPS Zs11ESVf18
SRM ZS11.E1.02
EPB       S001
FVCM Z5.0RC07  
TC F01R0A7354
TBOX SNMX007Z1E
FDR ACC_SW:5.0
VCU   blrv0450
RDRA 00000SW026
IPK S200425-09
ICE S160630.01
BMS 1027630EU4
TPMS QY1105A812
```

mossc001, 2023-02-05:
@pascalsaul , I've just had the BMS update on my MG ZS 2020 and have exactly the same problem with the estimated range, it just stays at 18km regardless of power cycles or distance. The "BMS Update" box in the features seems to have no affect; odd.

I haven't noticed a big change in SOC, but the estimated range is the very annoying one.

mossc001, 2023-02-09:
@pascalsaul , I currently have all the data logging to my own MQTT server, then into Home Assistant, so I might build my own estimation calculation until it's fixed.

GreyPeter, 2023-02-10:
Hello @pascalsaul and @mossc001. I believe this was a problem that I thought was resolved about 2 years ago but apparently not. I have found an old build by @KilveEng which I have built. If this resolves your issues, I should be able to implement it into the latest edge build. Can you try it out for me please. Peter 
[ovms3.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/10703253/ovms3.bin.zip)

pascalsaul, 2023-02-10:
> Hello @pascalsaul and @mossc001. I believe this was a problem that I thought was resolved about 2 years ago but apparently not. I have found an old build by @KilveEng which I have built. If this [...]

```
Download firmware from https://saul.nu/downloads/ovms3.bin to ota_1
Expected file size is 162
OTA Flash HTTP: Preparing flash partition...
Error: ESP32 error #5379 when writing to flash - state is inconsistent
```

For some reason I can't get the extracted file uploaded to OVMS through my webserver

pascalsaul, 2023-02-10:
Thatworks @GreyPeter since we got an option to choose the BMS version:
[image]

GreyPeter, 2023-02-10:
Thanks for testing it @pascalsaul . I will do a new version with the additional code and upload it here again.

mossc001, 2023-02-13:
@pascalsaul , did this resolve the SOC issue too? I haven't had chance to test the revised *.bin yet, I plan to do it this week.

[image]

pascalsaul, 2023-02-13:
> @pascalsaul , did this resolve the SOC issue too? I haven't had chance to test the revised *.bin yet, I plan to do it this week.
> 
> [image]

I just checked the estimated for now and will plan to do a full test with the new version from @GreyPeter (also with a full charge). I believe this will be fixed automatically with the additional code.

mossc001, 2023-02-19:
[image]
Looking good my end.

GreyPeter, 2023-02-19:
@mossc001 I am working on an updated version that I hope to push to the current version. It calculates the Estimated Range from the car's own consumption, battery size, state of health and state of charge. I am doing tests at the moment and hope to be able to release the code in the next couple of weeks.

dragon2611, 2023-02-28:
> [image] Looking good my end.
> 
> @GreyPeter , will the new version be a custom bin or wrapped into a 3.3.x release?

Does this mean the pre facelift MG5 *IS* supported as the docs seem to suggest it isn't although I saw mention of a possible early firmware build that might support it a couple years back in the mgev's forum.

GreyPeter, 2023-02-28:
@dragon2611 I am also working on a version for the ZS EV facelift vehicle which I also hope with the MG5. I am not finding much spare time at the moment so not much development getting done. Hopefully next week will be better.

dragon2611, 2023-02-28:
Do we know if it’s possible to get the SOC?
I’d need to buy the ovms hardware which is why I asked.

GreyPeter, 2023-03-01:
SOC should work on the MG5 as well but will need to be tested on a MG5. The ZS EV MK2 pids look very similar to the MG5.

dragon2611, 2023-03-07:
is it worth me ordering the hardware or should I hold off, sorry I'm unclear if there is already a build about that can read the PIDS on the Pre facelift 5.

Also the BMS on my car could be a weird version as it originally had issues where it kept coming up check HV battery and going into turtle (https://www.mgevs.com/threads/mg5-mystery-faults.7702/#post-125902) they had to release a new software to resolve it (Not sure if BMS or ECU)

GreyPeter, 2023-03-09:
@dragon2611 I would not advise that you buy the OVMS module until we can check if the version for the MG ZS EV facelift model works on the MG5. I believe there was some work done on an MG5 Version of the OVMS firmware but I don’t believe it was ever released to the public. If there is someone who can test a new version on an MG5, please let me know.

GreyPeter, 2023-03-10:
[ovms3.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/10937773/ovms3.bin.zip)
This is my latest build. It includes code to calculate the vehicle consumption and then calculates the estimated range from consumption.  It no longer uses the estimated range from the BMS so is unaffected by the BMS build. There is no longer a need to select the BMS Version. It may still need some fine tuning. Please test it and see if it performs correctly with Estimated Range.

dragon2611, 2023-03-10:
> @dragon2611 I would not advise that you buy the OVMS module until we can check if the version for the MG ZS EV facelift model works on the MG5. I believe there was some work done on an MG5 Version [...]

I may have to just risk it as I suspect there won't be many people who have the hardware sitting around and also happen to have an MG5 unless they either bought it previously with the hope of getting it working or have more than one car.

Worst case I'll have an expensive toy.

GreyPeter, 2023-03-11:
@dragon2611 could you join MG EV Hackers on slack so I can guide you through scanning PIDS when you get your OVMS Module.

dragon2611, 2023-03-11:
> @dragon2611 could you join MG EV Hackers on slack so I can guide you through scanning PIDS when you get your OVMS Module.

I think they may have set it to private as I found it via the docs site but it appeared to want an existing account and I didn’t see any options to join

GreyPeter, 2023-03-11:
Can you send your email address to prharry@hotmail.com so I can invite you?

dragon2611, 2023-03-11:
Done I think (mobile at the moment so using a tablet)

GreyPeter, 2023-03-11:
OK send me a Direct Message on slack when you can.

GreyPeter, 2023-03-11:
I have added some code to fix the SOC issue. Try doing a full charge and let me know if it works OK.
[ovms3.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/10947660/ovms3.bin.zip)

pascalsaul, 2023-03-11:
Looks good to me now. Getting the message topping off at 100% again. I must say the last 5% to 100% is going still quickly but that is probably something from the car itself.

pascalsaul, 2023-03-12:
Once the SOC is 100% it doesn't decrease anymore

GreyPeter, 2023-03-12:
OK @pascalsaul. Looks like I may have to approach it from a different slant. When you next do a full charge, after it finishes Topping Off, note the value of SOC Raw under MG/Charging Metrics

GreyPeter, 2023-03-14:
[ovms3.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/10963712/ovms3.bin.zip)
I have added some selections on the MG/Features please select BMS Firmware EU* and MG EV (UK/EU) as the vehicle under Config/Vehicle. Hopefully this should work correctly. Please Note: The estimated range uses a new calculated average consumption figure. I have initially set it to 165Wh/km (16.5kWh/100km or 265.54 Wh/mi) and it should update as you drive around. I may need to tweak how this averages consumption over time. Please let me know how it performs over time. You can view it in the shell with: me l xmg.avg.con

pascalsaul-copaco, 2023-03-15:
```
v.b.range.est                            214.196km
v.b.range.full                           251.849km
v.b.range.ideal                          259.004km
v.b.range.speed
v.b.soc                                  102.841%
v.b.soh                                  95.76%
v.b.temp                                 13°C
v.b.voltage                              449.5V
xmg.v.bat.voltage.bms                    449.5V
xmg.v.bat.voltage.vcu                    449.5V
xmg.v.range.raw                          194
xmg.v.soc.raw                            93%
xmg.avg.consumption                      294.662Wh/km

```

GreyPeter, 2023-03-15:
@pascalsaul-copaco thank you for these figures. I am still working on a better way to calculate the average consumption. I want it to average over a longer time and then the estimated range won't jump all over the place.

pascalsaul, 2023-03-15:
```
v.b.range.est                            214.196km
v.b.range.full                           251.849km
v.b.range.ideal                          259.004km
v.b.range.speed
v.b.soc                                  102.841%
v.b.soh                                  95.76%
v.b.temp                                 13°C
v.b.voltage                              449.5V
xmg.v.bat.voltage.bms                    449.5V
xmg.v.bat.voltage.vcu                    449.5V
xmg.v.range.raw                          194
xmg.v.soc.raw                            93%
xmg.avg.consumption                      294.662Wh/km
```

What about the strange SOC after full charge of 102.841% ;)

[image]

GreyPeter, 2023-03-15:
ovms3.bin.zip - Try this

GreyPeter, 2023-03-15:
[ovms3.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/10985321/ovms3.bin.zip)

pascalsaul, 2023-03-17:
`ota flash http http://home.saul.nu:8081/ovms3.bin`

I host the latest BIN here if others want to test as well. Uploaded and wil check in some hours again.

pascalsaul, 2023-03-17:
Charging is fixed now with SOC to 100%.

GreyPeter, 2023-03-18:
@pascalsaul thank you for hosting the .bin file. It will allow more people to test it. Is it still crashing?

GreyPeter, 2023-03-18:
I am currently working on a better way of calculating average consumption. 
If I can get it working properly, I will push it to the main build.

pascalsaul, 2023-03-19:
> @pascalsaul thank you for hosting the .bin file. It will allow more people to test it. Is it still crashing?

Looks ok for now. Maybe an older bug popped up.

mossc001, 2023-03-19:
@GreyPeter , I will get this tested today after a period of being offline.

GreyPeter, 2023-03-19:
Excellent @mossc001. The more testers the better.

mossc001, 2023-03-21:
I'll need to get the full output later but I'm only getting an SOC to 98.9% when full.

[image]

GreyPeter, 2023-03-22:
@mossc001 have you selected BMS Firmware EU* on the MG/Features page?

mossc001, 2023-03-24:
> @mossc001 have you selected BMS Firmware EU* on the MG/Features page?

Sorry, yes, seems it was on AU. I've now done a full charge since and can confirm 100% on SoC.

GreyPeter, 2023-03-25:
Ok - all good. I still want to do a bit more work on the code before I am ready to get it put into the main code.

GreyPeter, 2023-04-05:
OK. I think it is all done. I have added trip consumption (xmg.p.trip.consumption) which is calculated during a journey by checking the consumption every minute while the car is travelling over 5KPH. At the end of the trip, it is used to adjust the Average Consumption (xmg.p.avg.consumption) over several trips. The average consumption, weighted by the current trip consumption, is then used to calculate the estimated range. I have added selections on the MG/Features page to fix the low SOC readings. NOTE: The Estimated range will vary a little depending on the type of trips that you are doing but should be fairly accurate.
[ovms3_trip_consumption.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/11154860/ovms3_trip_consumption.bin.zip)

mossc001, 2023-04-05:
> OK. I think it is all done. I have added trip consumption (xmg.p.trip.consumption) which is calculated during a journey by checking the consumption every minute while the car is travelling over [...]

I will get this uploaded this Bank Holiday weekend and give it a test. Thanks!

pascalsaul, 2023-04-07:
> Thank you

```
Download complete (at 2785280 bytes)
Error: ESP32 error #5379 finalising OTA operation - state is inconsistent
```
Can't get the image uploaded. Tried Wednesday and today so looks corrupt to me. Can flash the other image on it.

GreyPeter, 2023-04-07:
I have done a rebuild and successfully installed it on my vehicle. Try this:
[ovms3_trip_consumption.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/11177565/ovms3_trip_consumption.bin.zip)

GreyPeter, 2023-04-07:
Should be 2,800,080 bytes when unzipped

pascalsaul, 2023-04-07:
Looking better and updated the image on my webserver ;)

mossc001, 2023-04-08:
Just installed it, will monitor over the long weekend :)

GreyPeter, 2023-04-08:
That's great. Thanks

pascalsaul, 2023-04-12:
> That's great. Thanks

v.b.soc                                  101.364%

You changed a bit too much again ;)

GreyPeter, 2023-04-12:
Can you tell me what SOC Raw is in MG/Charging Metrics or from the shell with me l xmg.v.soc.raw when fully charged.

GreyPeter, 2023-04-12:
OK Thanks guys. I will do a new build.

mossc001, 2023-04-12:
> [ovms3_trip_consumption.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/11208224/ovms3_trip_consumption.bin.zip)

Thanks! I've just updated but I will need to wait until the next partial discharge.

GreyPeter, 2023-04-12:
OK Thanks. Can you let me know how estimated range performs.

pascalsaul, 2023-04-12:
```
v.b.soc                                  100.78%
xmg.v.soc.raw                            93%
xmg.p.avg.consumption                    165Wh/km
xmg.p.trip.consumption                   165Wh/km
```

mossc001, 2023-04-27:
@GreyPeter over a few charges, I can confirm I'm getting the following:
```
v.b.soc = 100.78
xmg.v.soc.raw = 93
```

GreyPeter, 2023-04-27:
I will adjust it and do a rebuild and then upload it. Thanks for testing it

[13 more replies]