openvehicles/Open-Vehicle-Monitoring-System-3 #726: OVMS keeps crahing: abort() was called on core 0

pascalsaul, 2022-09-17:
Since long time and during tons of updates OVMS keeps crashing:

```
Last boot was 97284 second(s) ago
Time at boot: 2022-09-16 06:26:06 CEST
This is reset #14 since last power cycle
Detected boot reason: Crash (12/12)
Reset reason: Task watchdog (6)
Crash counters: 6 total, 0 early

Last crash: abort() was called on core 0
Backtrace:
0x4008ddca 0x4008e065 0x400e93dc 0x40084176
WDT tasks: IDLE1|OVMS DukTape
Version: 3.3.003-2-gd00de6c0/ota_1/edge (build idf v3.3.4-848-g1ff5e24b1 Sep 2 2022 17:00:44)

Hardware: OVMS WIFI BLE BT cores=2 rev=ESP32/1; MODEM SIM5360
Hardware: OVMS WIFI BLE BT cores=2 rev=ESP32/1; MODEM SIM5360
Firmware: 3.3.003-2-gd00de6c0/ota_1/edge (build idf v3.3.4-848-g1ff5e24b1 Sep 2 2022 17:00:44)
Running partition: ota_1
Boot partition: ota_1
Factory image: 3.2.015-192-g04a1f5d2
OTA_O image: 3.3.002-49-gffed9502
OTA_1 image: 3.3.003-2-gd00de6c0
```

mjkapkan, 2022-09-17:
I also observed this in past couple of months. Will dig deeper sometime this month.

dexterbg, 2022-09-17:
Still the same watchdog timeout issue as discussed previously:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/531

Please continue investigation there, closing this issue.

closed 2022-09-17 by dexterbg