openvehicles/Open-Vehicle-Monitoring-System-3 #728: [components/can] 2 small questions regarding naming

llange, 2022-09-30:
While browsing the source code I have 2 questions - not important things, but as I start to dig I try to understand the way the things are.

* The TAG for [the canformat `canswitch`](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/b70e4e3a8d46245f7564bc5f5479a97739cb1a90/vehicle/OVMS.V3/components/can/src/canformat_canswitch.cpp#L28) is `canformat-crtd` (i.e., the same as the canformat `crtd`), is this expected or a typo ? I can provide a PR to change it (to `canswitch` ? to `cs11` ?)

* Why is the [canformat `lawricel`] named as such ? It looks like the format from [Sweden Company Lawicel AB](http://www.lawicel.com/can232/can232.pdf) (not law**R**icel), is it expected or a typo ?)

Thanks for your help.

(PS: Please tell me also if it's the right place to ask, or would be better in the forum ? in the mailing-list ? somewhere else ?)

dexterbg, 2022-09-30:
You're probaly right on both, and you'll find some more copy/paste remnants, e.g. in the comments, as CRTD was the first formatter. You're welcome to post a pull request.

llange, 2022-09-30:
Thanks @dexterbg .
I've open 2 PRs:
* #729 for `cs11` (canswitch)
* #730 (Draft, not ready to be merged) form `lawricel` -> `lawicel`. Being a little bit concerned that it could break existing setup and user habits, I've duplicated the `RegisterCanFormat` call so that the format is registered twice and available as both `lawicel` and `lawricel`. I don't know if it's a good idea or not, neither if it will work... - let me know.

closed 2022-10-01 by llange