openvehicles/Open-Vehicle-Monitoring-System-3 #737: handle SCP possible sunset / implement SFTP

llange, 2022-10-13:
With OpenSSH 9.x+, OpenSSH has made **SFTP protocol** [the default protocol for the command-line `scp` tool](https://man.openbsd.org/scp.1#O) (_client-side_).
(See also e.g. https://www.redhat.com/en/blog/openssh-scp-deprecation-rhel-9-what-you-need-know)

It's possible that some day in the future, the support of **SCP protocol** will finally be completely from OpenSSH (_client-side and server-side_).

Users having OpenSSH 9.x+ client have a hard time copying files to and from OVMS, as the OVMS protocol for file transfer is **SCP protocol** and their `scp` tool uses **SFTP protocol** by default. (It's still possible to use the **SCP protocol** with the `-O` option to `scp`).

This issue is to engage a discussion about implementing (or not) the support for **SFTP protocol** (which is supported by the [WolfSSH library](https://github.com/wolfSSL/wolfssh)).

\+ documentation updates to explain the current situation.

llange, 2022-10-13:
Doc: https://www.wolfssl.com/documentation/manuals/wolfssh/chapter14.html

leres, 2022-10-13:
(Selfishly) this sounds like a good idea; I ran into this in June when I upgraded to 9.0p1 and in the last week switched to 9.1p1.

dexterbg, 2022-10-13:
Until then, adding a note to https://docs.openvehicles.com/en/latest/userguide/console.html#ssh-console would be good.

llange, 2022-10-13:
documentation updated.

leres, 2022-10-17:
I poked at this a little. First I had to understand that we're currently using mbedtls instead of wolfssl, I believe because the latter is too old to initiate a connection to a v2 or v3 server that uses modern crypto. And also that wolfssh is separate from wolfssl.

 Although I was able to get  WOLFSSH_SFTP to compile:
```
ice 1225 % git diff components/wolfssh/component.mk
diff --git a/vehicle/OVMS.V3/components/wolfssh/component.mk b/vehicle/OVMS.V3/components/wolfssh/component.mk
index 50228e3c..cfb23a1d 100644
--- a/vehicle/OVMS.V3/components/wolfssh/component.mk
+++ b/vehicle/OVMS.V3/components/wolfssh/component.mk
@@ -16,3 +16,7 @@ CFLAGS += -DWOLFSSL_USER_SETTINGS
 CFLAGS += -Wno-maybe-uninitialized
 #CFLAGS += -DDEBUG_WOLFSSH
 endif
+
+COMPONENT_OBJS += src/wolfsftp.o
+CFLAGS += -DWOLFSSH_SFTP
+CFLAGS += -DUSE_OSE_API
```
it doesn't change the "Connection closed" symptom seen when sftp is not supported so clearly I'm missing something.

leres, 2022-10-17:
I tried building with wolfssl 5.5.1 but I was hitting too many struct/typedef size errors.