openvehicles/Open-Vehicle-Monitoring-System-3 #741: Continous crashes

jollyjinx, 2022-10-21:
I bought a used module (board says 2022-03-05) and I'm seeing constant crashes like this:

I've tries eap and edge versions:
```
***ERROR*** A stack overflow in task tiT has been detected.
abort() was called at PC 0x4008e07c on core 1

ELF file SHA256: 96847e4f13e6f1ec

Backtrace: 0x4008ddca:0x3ffcc0e0 0x4008e065:0x3ffcc100 0x4008e07c:0x3ffcc120 0x40090048:0x3ffcc140 0x40091974:0x3ffcc160 0x4009192a:0x3ffcc1b0 0x400f53d5:0x000000fe

Rebooting...
```

dexterbg, 2022-10-21:
You need to tell the exact firmware version for that backtrace to make sense.

jollyjinx, 2022-10-21:
I just used a second ovms module (2022-05-22) cause I had the suspicion that the module might be broken. But same.

I presume the following will help:
```
I (80277) ovms-server-v3: Tx event system.modem.netmode***ERROR*** A stack overflow in task tiT has been detected.
abort() was called at PC 0x4008e07c on core 1

ELF file SHA256: 96847e4f13e6f1ec

Backtrace: 0x4008ddca:0x3ffcc0f0 0x4008e065:0x3ffcc110 0x4008e07c:0x3ffcc130 0x40090048:0x3ffcc150 0x40091974:0x3ffcc170 0x4009192a:0x3ffcc1c0 0x400f53d5:0x000000fe

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4796
load:0x40078000,len:0
load:0x40078000,len:14896
entry 0x40078d74
I (1056) psram: This chip is ESP32-D0WD
I (1056) spiram: Found 64MBit SPI RAM device
I (1056) spiram: SPI RAM mode: flash 40m sram 40m
I (1060) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1067) cpu_start: Pro cpu up.
I (1071) cpu_start: Application information:
I (1076) cpu_start: Project name:     ovms3
I (1081) cpu_start: App version:      3.3.003
I (1086) cpu_start: Compile time:     Sep  3 2022 11:32:11
I (1092) cpu_start: ELF file SHA256:  96847e4f13e6f1ec...
I (1098) cpu_start: ESP-IDF:          v3.3.4-848-g1ff5e24b1b
I (1104) cpu_start: Starting app cpu, entry point is 0x400818d4
I (1096) cpu_start: App cpu up.
I (1982) spiram: SPI SRAM memory test OK
```

jollyjinx, 2022-10-21:
and with another firmware on that module:

```
I (48230) gsm-ppp: Initialising...
D (48240) events: Signal(system.modem.netmode)
I (48240) ovms-server-v3: Tx event system.modem.netmode***ERROR*** A stack overflow in task tiT has been detected.
abort() was called at PC 0x4008e07c on core 1

ELF file SHA256: 07ebb07ee8901d89

Backtrace: 0x4008ddca:0x3ffcc100 0x4008e065:0x3ffcc120 0x4008e07c:0x3ffcc140 0x40090048:0x3ffcc160 0x40091974:0x3ffcc180 0x4009192a:0x3ffcc1d0 0x401103d1:0x000000fe

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4796
load:0x40078000,len:0
load:0x40078000,len:14896
entry 0x40078d74
I (1067) psram: This chip is ESP32-D0WD
I (1067) spiram: Found 64MBit SPI RAM device
I (1067) spiram: SPI RAM mode: flash 40m sram 40m
I (1071) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1078) cpu_start: Pro cpu up.
I (1082) cpu_start: Application information:
I (1087) cpu_start: Project name:     ovms3
I (1091) cpu_start: App version:      3.3.003-17-g5d8f5771
I (1098) cpu_start: Compile time:     Oct  7 2022 13:54:39
I (1104) cpu_start: ELF file SHA256:  07ebb07ee8901d89...
I (1110) cpu_start: ESP-IDF:          v3.3.4-848-g1ff5e24b1b
I (1116) cpu_start: Starting app cpu, entry point is 0x400818d4
I (1108) cpu_start: App cpu up.
I (1994) spiram: SPI SRAM memory test OK
I (1994) heap_init: Initializing. RAM available for dynamic allocation:
I (1995) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (2001) heap_init: At 3FFBF7B0 len 00020850 (130 KiB): DRAM
I (2007) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (2014) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (2020) heap_init: At 4009CCFC len 00003304 (12 KiB): IRAM
I (2026) cpu_start: Pro cpu start user code
I (2031) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (38) ovms_main: Set default logging level for * to INFO
I (38) ovms_main: Initialising WATCHDOG...
I (39) ovms-duktape: Initialising DUKTAPE Registry (1000)
I (46) command: Initialising COMMAND (1010)
I (51) command: Expanding DUKTAPE javascript engine
I (56) boot: Initialising BOOT (1100)
I (60) boot: Boot #4 reasons for CPU0=12 and CPU1=12
I (66) boot: Reset reason Exception/panic (4)
E (71) boot: Early crash #1 detected
I (75) events: Initialising EVENTS (1200)
I (80) config: Initialising CONFIG (1400)
```

jollyjinx, 2022-10-21:
I don't know the code and I can't debug the files (just have a mac with lldb) 
I'm comparing the output good vs. bad 
It looks like the internet connection got up and the ovms server gets the signal and tries to send before the cellular has the initialized the ppp callback ? then POOF ?

Warning:  that's just an guess - I might be completly wrong here.

```
D (43211) cellular: State transition NetStart => NetMode
I (43211) cellular: State: Enter NetMode state
I (43211) gsm-ppp: Initialising...
I (43211) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/v/e/parktime=925
I (43221) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/freeram=3496600
D (43221) events: Signal(system.modem.netmode)
I (43221) ovms-server-v3: Tx event system.modem.netmode
I (43351) gsm-ppp: StatusCallBack: None
I (43351) gsm-ppp: status_cb: Connected
I (43351) gsm-ppp:    our_ipaddr  = 10.156.209.132
I (43351) gsm-ppp:    his_ipaddr  = 10.64.64.64
```

```
I (48230) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/net/wifi/sq=-56.3
I (48230) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/net/sq=-56
D (48230) cellular: State transition NetStart => NetMode
I (48230) cellular: State: Enter NetMode state
I (48230) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/v/e/parktime=854
I (48230) gsm-ppp: Initialising...
D (48240) events: Signal(system.modem.netmode)
I (48240) ovms-server-v3: Tx event system.modem.netmode***ERROR*** A stack overflow in task tiT has been detected.
abort() was called at PC 0x4008e07c on core 1
```

dexterbg, 2022-10-21:
Sorry, I don't use MQTT due to the protocol overhead, so you need to help tracing this down.

- Please post the output of `module summary`.
- Try booting with server v3 disabled.
- If server v3 starts & runs without crashing after network setup is complete, please then try `server v3 update all`.
- If you're using TLS, try disabling.

jollyjinx, 2022-10-21:
@dexterbg Thanx for looking into this. I'm more than happy to help.

I'm not using TLS for now.

My findings: 

module summary ( when modem enabled )
```
OVMS MODULE SUMMARY

Module
  Version:  3.3.003-17-g5d8f5771/ota_0/eap (build idf v3.3.4-848-g1ff5e24b1b Oct  7 2022 13:54:39)
  Hardware: OVMS WIFI BLE BT cores=2 rev=ESP32/3; MODEM SIM7600
  12v:      12.9v

MODEM Status
  PIN: SIM card PIN code required.
  Model: SIM7600
  Network Registration: RegisteredHome
    Provider: Drillisch
    Signal: -61 dBm
    Mode: LTE,Online
  State: NetMode
    Ticker: 229
    User Data: 0
  UART:
    FIFO overflows: 0
    Buffer overflows: 0
    Parity errors: 0
    Frame errors: 0
    Driver Buffer overflows: 0
  Mux: Status up
    Open Channels: 4
    Framing Errors: 0
    RX frames: 359
    TX frames: 50
    Last RX frame: 3 sec(s) ago
  PPP: Connected on channel: #2
  GPS: Connected on channel: #1
     Status: enabled
     Time: enabled

Configuration
  [auto]
    dbc: no
    egpio: no
    ext12v: no
    init: yes
    modem: yes
    obd2ecu: 
    ota: no
    scripting: yes
    server.v2: yes
    server.v3: yes
    vehicle.type: BMWI3
    wifi.mode: apclient
    wifi.ssid.ap: ovms
    wifi.ssid.client: jinx.de IOT
  [can]
  [dbc]
  [egpio]
    monitor.ports: 
  [http.plugin]
  [http.server]
  [locations]
  [log]
    file.enable: yes
    file.keepdays: 0
    file.path: /sd/logs/ovms.log
    level: debug
[... 105 lines cut ...]
OVMS# server v3 update all
Server V3 data update for all metrics has been scheduled
I (123208) housekeeping: System considered stable (RAM: 8b=64844-67240 32b=12344 SPI=3444340-3449572)
D (123248) ovms-duktape: Duktape: Compacting DukTape memory done in 30 ms
I (142198) ovms-server-v2: Send MP-0 S0,K,0,0.00,stopped,standard,0,0,0,0,0,0,0,21,0,0,0,0,0.00,0,0,0,0,-1,0,0,0,0,0,0,0,0.00,0.00,0,0.00,0.00,0.00,0.0
I (142198) ovms-server-v2: Send MP-0 D0,0,5,0,0,0,0,0,0,139,0,0,0,0,12.91,0,13,0,0,0,0
I (171198) ovms-server-v3: Connection is mqtt.jinx.de:1883 WBY8P21010VD27889/ovms topic ovms/WBY8P21010VD27889/
I (171198) ovms-server-v3: Status: Connecting...
D (171198) events: Signal(server.v3.connecting)
I (171338) ovms-server-v3: Connection successful
I (171358) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/s/v3/connected=yes
I (171358) ovms-server-v3: Status: OVMS V3 MQTT login successful
D (171358) events: Signal(server.v3.connected)
I (171358) ovms-server-v3: Tx event server.v3.connected
I (172198) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/monotonic=169
I (172198) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/time/utc=1666383451
I (172198) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/net/wifi/sq=-58.3
I (172198) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/net/sq=-58
I (172198) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/v/e/parktime=169
I (172198) ovms-server-v3: Subscribe to MQTT topics
I (172208) ovms-server-v3: Transmit all metrics
I (172208) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/egpio/input=0,1,2,3,4,5,6,7,8,9
I (172208) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/egpio/output=4,5,6,7,8,9
```
but then continues to reconnect.

jollyjinx, 2022-10-21:
I did try the *server v3 update all * command with the system up and modem enabled - and it worked fine.
For whatever reason It got through the cellular connection phase this time fine (no crash) and seems to keep on running.

dexterbg, 2022-10-22:
The BMWI3 adapter creates a range of custom metrics, most of which will only be filled if the car has been switched on:
- https://docs.openvehicles.com/en/latest/components/vehicle_bmwi3/docs/index.html#custom-metrics

I assume the amount of data / transmission requests needed for MQTT in your case is on the edge of too much, when sent via the slow cellular connection.

The MQTT client does full retransmissions on each reconnect, I think that exceeds the TCP/IP stack size by chance depending on the actual metrics data (e.g. BMS cell data).

Most probably no other BMWI3 user yet uses MQTT, so this didn't come up before.

@elbow Steve, can you confirm / replicate this? If so, please test if raising the TCP/IP stack size would help.

dexterbg, 2022-10-22:
…that's configured in the LWIP section, defaults to 2560 (LWIP generally adds another 512 bytes to that).

jollyjinx, 2022-10-22:
I found out that 'tiT' stands for the tcp/ip stack process esp32 and found the following thread with a person having the same issue:  http://lists.openvehicles.com/pipermail/ovmsdev/2018-March/011592.html 
A increase in stacksize resolved the persons problem. 

The stacktrace leading to the crash is always the same here maybe I have to figure out how to get an development environment to see the function names to see the error.

jollyjinx, 2022-10-22:
> The BMWI3 adapter creates a range of custom metrics, most of which will only be filled if the car has been switched on:
>
> * [...]

The problem arises on my desk with no BMW connected and the system is in WiFi range.

jollyjinx, 2022-10-22:
As I'm on ventura ( which brew currently does not support) I created a docker container on my raspberry pi ( espressif/idf:latest ) which does not include gdb , but at least **aarch64-linux-gnu-addr2line**

With that I was able to translate the crash to readable addresses:
```
root@833e4f270b1d:/project/project/edge# aarch64-linux-gnu-addr2line -e ovms3.elf --pretty-print 
0x4008ddca:0x3ffcc100
/home/balzer/esp/esp-idf/components/esp32/panic.c:156
0x4008e065:0x3ffcc120
/home/balzer/esp/esp-idf/components/esp32/panic.c:171
0x4008e07c:0x3ffcc140
/home/balzer/esp/esp-idf/components/esp32/panic.c:122
0x40090048:0x3ffcc160
/home/balzer/esp/esp-idf/components/freertos/tasks.c:2770
0x40091974:0x3ffcc180
/home/balzer/esp/esp-idf/components/freertos/portasm.S:406
0x4009192a:0x3ffcc1d0
/home/balzer/esp/esp-idf/components/freertos/portasm.S:206
0x401103d1:0x000000fe
/home/balzer/esp/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/main/ovms_netmanager.cpp:742
```

Looking at the source code:

it just returned from **netif_set_default** , but I can't seem to find the source for that function - yet.

dexterbg, 2022-10-22:
That's why I asked you to get the `module tasks` output, as that includes stack usage. It doesn't show a usage anywhere near the limit for the TCP/IP task, but that may happen in a fraction of a second.

Stack sizes are fixed and need to be internal RAM, that's why we need to be careful about raising them. The thread on the developer list ended with an unanswered question about the sufficient stack size, so the change wasn't incorporated into the standard build config.

As you're going to dig into the code: use the `a2l` script (see support directory) instead, as addr2line doesn't work correctly in our case.

dexterbg, 2022-10-22:
Also, to change the stack size for your own build, you can simply adjust `CONFIG_TCPIP_TASK_STACK_SIZE=2560` in file `sdkconfig` manually, or you can use `make menuconfig` and find that config in Component config → LWIP.

jollyjinx, 2022-10-23:
I might get into compiling it myself at a later stage, but I haven't yet.

Stack is not heap. Why might it use up so much stack ? Stack is for stack variables and return addresses , registers and heap for dynamic allocated memory - that's how I understand it.  That would mean the code allocates huge amounts of stack variables (bad coding practice) or goes into a recursive loop somewhere - but that should then be visible by logging - right ?

dexterbg, 2022-10-23:
And that's why I'd like to know if changing the stack size helps, and then how much stack is actually used in this case.

LWIP is especially optimized for embedded systems with small stacks, it normally does a good job at that as you can see from the stack usage stats.

Logging may or may not help, as logging is asynchronous & slow.

dexterbg, 2022-10-23:
Attached is a build with +2K stack size for tiT. Please try if that helps, if so report the maximum stack usage you see by `module tasks`.
[ovms3.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/9846107/ovms3.bin.zip)

jollyjinx, 2022-10-23:
Wow! great news. I will give it a try :-D 
Thank you a lot for your support @dexterbg

jollyjinx, 2022-10-23:
I've just launched it and it seems to work after the first 2 boots.
As you are for sure interested in the ```mo ta``` output - here it comes:
```
OVMS# mo ta
Number of Tasks = 18        Stack:  Now   Max Total    Heap 32-bit SPIRAM C# PRI CPU% BPR/MH
3FFAFB88    1 Blk esp_timer         436   708  4096   41736    644  46208  0  22   0%  22/ 0
3FFC2C18    2 Blk OVMS DukTape      500  9252 12288     188      0 524288  1   5   1%   5/ 0
3FFC46F4    3 Blk eventTask         512  1888  4608     104      0      0  0  20   0%  20/ 0
3FFC6B78    4 Blk OVMS Events       484  2980  8192   81472      0  44492  1   8   1%   8/ 0
3FFC9CDC    5 Blk OVMS CanRx        472   472  4096    3240      0  55076  0  23   0%  23/ 0
3FFCAAD8    6 Blk ipc0              420   420  1024    7804      0      0  0  24   0%  24/ 0
3FFCB0DC    7 Blk ipc1              424   520  1024     120      0      0  1  24   0%  24/ 0
3FFCCF4C   10 Rdy IDLE0             408   584  1024       0      0      0  0   0  98%   0/ 0
3FFCD4E4   11 Rdy IDLE1             416   512  1024       0      0      0  1   0  95%   0/ 0
3FFCE27C   12 Blk Tmr Svc           376  2952  3072     324      0      0  0  20   0%  20/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6188  1  18   1%  18/ 0
3FFD39D8   20 Blk OVMS Cellular     644  3620  4096    6764      0  10388  0  20   0%  20/ 0
3FFD69EC   21 Blk wifi              472  2664  3584   38076      0   5740  0  22   1%  22/ 1
3FFE3694   22 Blk OVMS Vehicle      496   496  8192       0      0      0  1  10   0%  10/ 0
3FFE7FAC   23 Rdy OVMS Console      840  2680 10240     356  12100  15960  1   5   3%   5/ 1
3FFEA944   24 Blk OVMS NetMan       912  2944 10240     316      0   2172  1   5   0%   5/ 0
3FFEBF80   25 Blk mdns              508  1404  4096     104      0      4  0   1   0%   1/ 0
3FFF0474   26 Blk OVMS FileLog      592  1600  3072      40      0      0  1   2   0%   2/ 0
```

and a grep over my terminal buffer reveals:
```
tin>pbpaste |grep 'Blk tiT'                                                                                                                (15:39:04)
3FFCBA7C   17 Blk tiT               536  2040  4608    1256      0   2164  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  2040  4608    1256      0   2164  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1120      0   6464  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1120      0   6448  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1120      0   6444  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1120      0   6448  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608     784      0   5288  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608     784      0   5288  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608     784      0   5288  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  2040  4608     808      0   1136  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   7756  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6188  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6184  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6204  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6204  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6204  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6200  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6188  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6188  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6188  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6188  1  18   0%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   6188  1  18   1%  18/ 0
3FFCBA7C   17 Blk tiT               536  3064  4608    1008      0   7748  1  18   1%  18/ 0
```

so max is 3064 - i wonder why it did crash then with the 3072 max setting from the normal build.

dexterbg, 2022-10-23:
Maybe it will occasionally need some more stack? Please keep this running for 1-2 days, try to create as much stress on it as possible and keep an eye on the maximum stack usage recorded. Thanks!

jollyjinx, 2022-10-23:
Ok, first crash encountered when stopping v2 and v3 servers. But was not able to reproduce it by starting/stoping v2 and v3 servers after the reboot.

console showed ( i did a ```mo ta``` a couple of seconds before ):

```
OVMS# mo ta
Number of Tasks = 18        Stack:  Now   Max Total    Heap 32-bit SPIRAM C# PRI CPU% BPR/MH
3FFAFB88    1 Blk esp_timer         436   708  4096   41452    644  46208  0  22   0%  22/ 0
3FFC2C18    2 Blk OVMS DukTape      500  9252 12288     188      0 524288  1   5   0%   5/ 0
3FFC46F4    3 Blk eventTask         480  1936  4608     104      0      0  0  20   0%  20/ 0
3FFC6B78    4 Blk OVMS Events       484  2980  8192   81508      0  44532  1   8   1%   8/ 0
3FFC9BC4    5 Blk OVMS CanRx        480   480  4096    3320      0  55076  0  23   0%  23/ 0
3FFCA9C4    6 Blk ipc0              432   432  1024    7804      0      0  0  24   0%  24/ 0
3FFCAFC8    7 Blk ipc1              420   516  1024     120      0      0  1  24   0%  24/ 0
3FFCCE38   10 Rdy IDLE0             404   580  1024       0      0      0  0   0  97%   0/ 0
3FFCD3D0   11 Rdy IDLE1             412   508  1024       0      0      0  1   0  96%   0/ 0
3FFCE168   12 Blk Tmr Svc           372  2948  3072     324      0      0  0  20   0%  20/ 0
3FFCC8A0   17 Blk tiT               540  3068  4608   10580      0  20392  1  18   1%  18/ 0
3FFD3838   20 Blk OVMS Cellular     644  3620  4096    6768      0  10388  0  20   0%  20/ 0
3FFD68C4   21 Blk wifi              480  2720  3584   38108      0   7440  0  22   2%  22/ 1
3FFE3498   22 Blk OVMS Vehicle      484   484  8192       0      0      0  1  10   0%  10/ 0
3FFE7E1C   23 Rdy OVMS Console      840  2888 10240     356  12100  15964  1   5   0%   5/ 1
3FFEA7E8   24 Blk OVMS NetMan       900  3700 10240    2240      0   5080  1   5   1%   5/ 0
3FFEBE24   25 Blk mdns              512  1408  4096     108      0      4  0   1   0%   1/ 0
3FFF278C   26 Blk OVMS FileLog      584  1592  3072      40      0      0  1   2   0%   2/ 0
OVMS# Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40114b41  PS      : 0x00060630  A0      : 0x80114bc8  A1      : 0x3ffea590  
A2      : 0x0000002c  A3      : 0xffffffff  A4      : 0x3ffc48c8  A5      : 0x00000000  
A6      : 0x00000028  A7      : 0x00000000  A8      : 0x8023644c  A9      : 0x3ffea560  
A10     : 0x3f8b5e70  A11     : 0x400848e0  A12     : 0x3ffb00c8  A13     : 0x00000000  
A14     : 0x000000fe  A15     : 0x3ffae7e8  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x4008a84c  LEND    : 0x4008a868  LCOUNT  : 0x00000000  

ELF file SHA256: d6d4abda45d0d8cb

Backtrace: 0x40114b41:0x3ffea590 0x40114bc5:0x3ffea5b0 0x401622da:0x3ffea5d0 0x40167697:0x3ffea600 0x4012cf91:0x3ffea620 0x4012d0f8:0x3ffea660 0x4012f033:0x3ffea680 0x4012b1da:0x3ffea6e0 0x401140cb:0x3ffea700 0x40114179:0x3ffea750

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4796
load:0x40078000,len:0
load:0x40078000,len:14896
entry 0x40078d74
I (1067) psram: This chip is ESP32-D0WD
I (1067) spiram: Found 64MBit SPI RAM device
I (1067) spiram: SPI RAM mode: flash 40m sram 40m
I (1071) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1078) cpu_start: Pro cpu up.
I (1082) cpu_start: Application information:
I (1087) cpu_start: Project name:     ovms3
I (1091) cpu_start: App version:      3.3.003-33-g3915a27d-dirty
I (1098) cpu_start: Compile time:     Oct 23 2022 11:06:34
I (1104) cpu_start: ELF file SHA256:  d6d4abda45d0d8cb...
I (1110) cpu_start: ESP-IDF:          v3.3.4-848-g1ff5e24b1b
I (1117) cpu_start: Starting app cpu, entry point is 0x400818d4
I (1109) cpu_start: App cpu up.
I (1995) spiram: SPI SRAM memory test OK
``` 

Looks like it crashed in ```Blk OVMS Vehicle``` from the stacktrace.

jollyjinx, 2022-10-23:
And again - stopping v2 server lead to a crash:
```
OVMS# mo ta
Number of Tasks = 18        Stack:  Now   Max Total    Heap 32-bit SPIRAM C# PRI CPU% BPR/MH
3FFAFB88    1 Blk esp_timer         436   708  4096   41736    644  46208  0  22   0%  22/ 0
3FFC2C18    2 Blk OVMS DukTape      500  9252 12288     188      0 524288  1   5   1%   5/ 0
3FFC46F4    3 Blk eventTask         480  1936  4608     104      0      0  0  20   0%  20/ 0
3FFC6B78    4 Blk OVMS Events       516  3268  8192   80884      0  40160  1   8   1%   8/ 0
3FFC9CDC    5 Blk OVMS CanRx        472  2040  4096    3240      0  55008  0  23   0%  23/ 0
3FFCAAD8    6 Blk ipc0              420   500  1024    7804      0      0  0  24   0%  24/ 0
3FFCB0DC    7 Blk ipc1              424   520  1024     120      0      0  1  24   0%  24/ 0
3FFCCF4C   10 Rdy IDLE0             408   504  1024       0      0      0  0   0  97%   0/ 0
3FFCD4E4   11 Rdy IDLE1             416   512  1024       0      0      0  1   0  92%   0/ 0
3FFCE27C   12 Blk Tmr Svc           376  2952  3072     756      0     60  0  20   0%  20/ 0
3FFD0B5C   17 Blk tiT               540  3068  4608    1120      0   6460  1  18   1%  18/ 0
3FFD3A44   20 Blk OVMS Cellular     656  3632  4096    6464      0    112  0  20   0%  20/ 0
3FFD6AD0   21 Blk wifi              476  2716  3584   38108      0   7440  0  22   2%  22/ 1
3FFE36EC   22 Blk OVMS Vehicle      488   488  8192       0      0      0  1  10   0%  10/ 0
3FFE8038   23 Rdy OVMS Console      836  2676 10240     364  12100  15960  1   5   3%   5/ 1
3FFEAA04   24 Blk OVMS NetMan       912  4336 10240    2420      0   6244  1   5   2%   5/ 0
3FFEC040   25 Blk mdns              508  1404  4096     104      0      4  0   1   0%   1/ 0
3FFF0570   26 Blk OVMS FileLog      588  2124  3072     252      0      0  1   2   0%   2/ 0
I (1805287) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/monotonic=1802
I (1805287) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/m/time/utc=1666538609
....
I (1833467) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/s/v2/connected=no
V (1833467) ovms-server-v2: OvmsServerV2MongooseCallback(MG_EV_CLOSE)
I (1833467) ovms-server-v2: Status: Disconnected
I (1833467) ovms-server-v3: Tx metric ovms/WBY8P21010VD27889/metric/s/v2/peers=0
D (1833477) events: SignaGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40114b41  PS      : 0x00060f30  A0      : 0x80114bc8  A1      : 0x3ffea7a0  
A2      : 0x0000002c  A3      : 0xffffffff  A4      : 0x3ffc48c8  A5      : 0x00000000  
A6      : 0x00000028  A7      : 0x00000000  A8      : 0x8023644c  A9      : 0x3ffea770  
A10     : 0x3f831e6c  A11     : 0x400848e0  A12     : 0x3ffb00c8  A13     : 0x00000000  
A14     : 0x000000fe  A15     : 0x3ffae7e8  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x4008a84c  LEND    : 0x4008a868  LCOUNT  : 0x00000000  

ELF file SHA256: d6d4abda45d0d8cb

Backtrace: 0x40114b41:0x3ffea7a0 0x40114bc5:0x3ffea7c0 0x401622da:0x3ffea7e0 0x40167697:0x3ffea810 0x4012cf91:0x3ffea830 0x4012d0f8:0x3ffea870 0x4012f033:0x3ffea890 0x4012b1da:0x3ffea8f0 0x401140cb:0x3ffea910 0x40114179:0x3ffea960

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4796
load:0x40078000,len:0
load:0x40078000,len:14896
entry 0x40078d74
I (1067) psram: This chip is ESP32-D0WD
I (1067) spiram: Found 64MBit SPI RAM device
I (1067) spiram: SPI RAM mode: flash 40m sram 40m
I (1071) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1078) cpu_start: Pro cpu up.
I (1082) cpu_start: Application information:
I (1087) cpu_start: Project name:     ovms3
```

dexterbg, 2022-10-23:
That's possibly a race condition on stopping the v2/v3 servers, which is rarely done during normal operation. It's also rare that both are running at the same time, which probably leads to your near max timer service stack.

Attached is a new build that tries to address both issues.
[ovms3.bin.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/9846991/ovms3.bin.zip)

jollyjinx, 2022-10-23:
Another wow ! I just installed it and can't seem to get it to crash immediately. Lets see how well it fares when it's up and running a couple of minutes.

Thank you very much for your quick help and responses @dexterbg !

jollyjinx, 2022-10-23:
Well, got excited too early:

```
W (353271) websocket: WebSocketHandler[0x3f8bf43c]: job queue overflow detected
OVMS# mo taE (356201) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (356201) task_wdt:  - IDLE1 (CPU 1)
E (356201) task_wdt: Tasks currently running:
E (356201) task_wdt: CPU 0: IDLE0
E (356201) task_wdt: CPU 1: OVMS NetMan
E (356201) task_wdt: Aborting.
abort() was called at PC 0x400e86ac on core 0

ELF file SHA256: e030ee54cc1c1ffc

Backtrace: 0x4008ddce:0x3ffb0a90 0x4008e069:0x3ffb0ab0 0x400e86ac:0x3ffb0ad0 0x4008417a:0x3ffb0af0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4796
load:0x40078000,len:0
load:0x40078000,len:14896
entry 0x40078d74
I (1067) psram: This chip is ESP32-D0WD
I (1067) spiram: Found 64MBit SPI RAM device
I (1067) spiram: SPI RAM mode: flash 40m sram 40m
I (1071) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1078) cpu_start: Pro cpu up.
I (1082) cpu_start: Application information:
I (1087) cpu_start: Project name:     ovms3
I (1092) cpu_start: App version:      3.3.003-34-gc3ca4159-dirty
I (1098) cpu_start: Compile time:     Oct 23 2022 18:06:52
I (1104) cpu_start: ELF file SHA256:  e030ee54cc1c1ffc...
I (1110) cpu_start: ESP-IDF:          v3.3.4-848-g1ff5e24b1b
I (1117) cpu_start: Starting app cpu, entry point is 0x400818d8
I (1109) cpu_start: App cpu up.
I (1995) spiram: SPI SRAM memory test OK
```

jollyjinx, 2022-10-23:
I can't reproduce that crash even though it started/stopped v2, v3 and cellular and refreshed the website a couple of times.
stack looks large enough:

```
OVMS# mo ta
Number of Tasks = 18        Stack:  Now   Max Total    Heap 32-bit SPIRAM C# PRI CPU% BPR/MH
3FFAFB88    1 Blk esp_timer         436   708  4096   41736    644  46208  0  22   0%  22/ 0
3FFC2C18    2 Blk OVMS DukTape      500  9252 12288     188      0 524288  1   5   0%   5/ 0
3FFC46F4    3 Blk eventTask         480  1888  4608     104      0      0  0  20   0%  20/ 0
3FFC6B78    4 Blk OVMS Events       484  3268  8192   79632      0  39728  1   8   1%   8/ 0
3FFC9CDC    5 Blk OVMS CanRx        472   472  4096    3240      0  55008  0  23   0%  23/ 0
3FFCAAD8    6 Blk ipc0              420   500  1024    8316      0      0  0  24   0%  24/ 0
3FFCB0DC    7 Blk ipc1              424   520  1024     120      0      0  1  24   0%  24/ 0
3FFCCF4C   10 Rdy IDLE0             408   584  1024       0      0      0  0   0  96%   0/ 0
3FFCD4E4   11 Rdy IDLE1             416   592  1024       0      0      0  1   0  89%   0/ 0
3FFCE47C   12 Blk Tmr Svc           376  2952  3584     324      0      0  0  20   0%  20/ 0
3FFCC890   17 Blk tiT               540  3068  3584    1236      0   8488  1  18   1%  18/ 0
3FFD375C   20 Blk OVMS Cellular     648  3752  4096    6780      0  10392  0  20   0%  20/ 0
3FFD67E8   21 Blk wifi              468  2708  3584   38108      0   7444  0  22   2%  22/ 1
3FFE33F4   22 Blk OVMS Vehicle      496   496  8192       0      0      0  1  10   0%  10/ 0
3FFE7D54   23 Rdy OVMS Console      848  2784 10240     356  12100  15960  1   5   5%   5/ 1
3FFEA720   24 Blk OVMS NetMan       908  4396 10240    5832      0  18052  1   5   3%   5/ 0
3FFEBD5C   25 Blk mdns              504  1400  4096     104      0      4  0   1   0%   1/ 0
3FFEF240   26 Blk OVMS FileLog      588  2076  3072     252      0      0  1   2   0%   2/ 0
```

dexterbg, 2022-10-23:
That's yet another crash type, unfortunately not as easy to find, we've been hunting for that for a while. The task watchdog gets triggered when a higher priority task is busy for too long (120 seconds), so the watchdog doesn't get reset in time.

The tasks running at the WDT moment are varying, so don't give a real hint. Chances are this is really occuring somewhere outside measurability, maybe interrupt related. I had the effect some days ago and was lucky to be able to login shortly before the WDT reset. The module was nearly unresponsive, but no task showed any unusual numbers.

Your stack usage on the cellular task bothers me, that's much higher than normal, and uncomfortably close to the max as well. I'll do some more stack changes.

jollyjinx, 2022-10-24:
The module ran through till now (60k seconds!) and has not crashed. This is clearly a big advancement for me.

```
OVMS# mo ta
Number of Tasks = 18        Stack:  Now   Max Total    Heap 32-bit SPIRAM C# PRI CPU% BPR/MH
3FFAFB88    1 Blk esp_timer         436   708  4096   41452    644  46208  0  22   0%  22/ 0
3FFC2C18    2 Blk OVMS DukTape      500  9252 12288     188      0 524288  1   5   0%   5/ 0
3FFC46F4    3 Blk eventTask         480  1888  4608     104      0      0  0  20   0%  20/ 0
3FFC6B78    4 Blk OVMS Events       484  3444  8192   79824      0  39704  1   8   1%   8/ 0
3FFC9BC4    5 Blk OVMS CanRx        480  2048  4096    3320      0  55008  0  23   0%  23/ 0
3FFCA9C4    6 Blk ipc0              432   464  1024    8316      0      0  0  24   0%  24/ 0
3FFCAFC8    7 Blk ipc1              420   516  1024     120      0      0  1  24   0%  24/ 0
3FFCCE38   10 Rdy IDLE0             404   500  1024       0      0      0  0   0  98%   0/ 0
3FFCD3D0   11 Rdy IDLE1             412   588  1024       0      0      0  1   0  97%   0/ 0
3FFCE368   12 Blk Tmr Svc           372  2948  3584     324      0      0  0  20   0%  20/ 0
3FFCB33C   17 Blk tiT               540  3068  3584    1124      0   6444  1  18   1%  18/ 0
3FFD3648   20 Blk OVMS Cellular     644  3620  4096    6768      0  10388  0  20   0%  20/ 0
3FFD66D4   21 Blk wifi              480  2672  3584   38076      0   5904  0  22   2%  22/ 1
3FFE32A4   22 Blk OVMS Vehicle      496   496  8192       0      0      0  1  10   0%  10/ 0
3FFE7C40   23 Rdy OVMS Console     1484  2684 10240     304  12100  15964  1   5   0%   5/ 1
3FFEA60C   24 Blk OVMS NetMan       904  4264 10240    1848      0   3700  1   5   1%   5/ 0
3FFEBC2C   25 Blk mdns              504  1400  4096     104      0      4  0   1   0%   1/ 0
3FFF0338   26 Blk OVMS FileLog      580  2148  3072     252      0      0  1   2   0%   2/ 0
OVMS# boot status
Last boot was 60084 second(s) ago
```

dexterbg, 2022-10-29:
New edge build `3.3.003-43-g3e5ac774` now includes these stack adjustments.

jollyjinx, 2022-11-01:
This is great news.

closed 2022-11-01 by jollyjinx