openvehicles/Open-Vehicle-Monitoring-System-3 #752: Explore possibility of wireguard (VPN) support

llange, 2022-10-23:
I was toying with the idea of having a network-enabled OVMS auto-registering in a dedicated network ; and being "locally" available, mdns working, web and ssh reachable... while exposing no local service on its main interface.

I've seen discussion around a "firewall", and with this approach there is no more need to have one if no services are exposed.

Don't know if it's feasible, a project like https://github.com/smartalock/wireguard-lwip is certainly interesting to look at (and also https://github.com/ciniml/WireGuard-ESP32-Arduino).

This issue is to discuss, gather interest and coordinate ideas.

llange, 2022-11-07:
some notes: I tried as a first step to integrate https://github.com/trombik/esp_wireguard which is an ESP-IDF component, but unfortunately our version of lwip seems to old for that (missing an udp function etc...)
Maybe #360 could help here.

werdnum, 2023-06-05:
Would love to see this! Actually what I'd *really* love to see is Tailscale (based on Wireguard), but standalone Wireguard I could live with too.

llange, 2023-06-05:
Hi @werdnum,
if you're feeling adventurous and not afraid of having your hands dirty compiling, please note that on this branch https://github.com/llange/Open-Vehicle-Monitoring-System-3/tree/752-wireguard you should be able to build an ESP-IDFv5 version of OVMSv3 with my Wireguard patches.
Once you manage to compile it, Wireguard works out of the box - I'm only testing SSH and Web server on it but no reason that other protocol do not work.
Let us know.

werdnum, 2023-06-26:
> Hi @werdnum, if you're feeling adventurous and not afraid of having your hands dirty compiling, please note that on this branch [...]

This looks really interesting, thanks for sharing. It looks like this branch is mostly about getting ESPIDF updated, which I understand the importance of. I'd be keen to try it out, but not 100% clear on what kind of a state things are in right now. I see a bunch of TODOs - most of which don't bother me as I'm not actively hacking on OVMS - but I do see there might be some TLS/SSH issues remaining.

In general not afraid to give it a whirl, and get my hands a _little_ dirty with it, but the way my schedule is I won't have a ton of time hands-on with the vehicle, so I just want to make sure I'm not going to break everything and need to spend a bunch of time recovering it :).

llange, 2023-06-26:
Hi @werdnum !
You're right, the prerequisite for WG support is to have an ESP-IDFv5 up and running. ESP-IDFv3 could certainly be made to work with WG but would require a rework of the WG library/component I integrated.
Besides I found having an ESP-IDF up to date was needed for other pet projects of mine.

There are indeed a few quirks with this ESP-IDFv5. From memory:
* I suspect a few issues are related to stack change / memory issues because few things changed between v3 and v5. I had, for example, to change a stack variable to static to prevent an overflow : 1111c99a9488c7ce4b778c09b0f28b3ca924fb6c
* There are as you saw issues with SSH. This one is a pain, and a showstopper so we need to fix it. My hypothesis is that it has to do with the lack of multithreading support in Mongoose (in v3, we used a custom fork of Mongoose with added thread-safety :  openvehicles/mongoose@085da15428347a00231859b94afd1f6fc4202ff1 and openvehicles/mongoose@1dfbd82f24796814c2b28a924a1c4c484a5269fe . However these do not port cleanly to Mongoose v7 (which changed a lot) so we're back to mainstream non-thread-safe Mongoose. The issues only shows in SSH but will certainly pop up elsewhere.

I'm not actively working on it (read: from time to time I can spare one or two hours in a busy schedule to try to fix the SSH bugs), and I certainly lacked a status update on the list. I'll try to fix this ASAP.

I can't give any guarantee that this branch won't break a thing. Generally I found it usable (if you can leave without SSL) for a few tests after boot, but I cannot use it in a vehicle (as I badly need SSH, and my vehicle is remote in another country).
So you'll certainly need to asses the risk yourself, and see if you want to take the chance. (Or you may have multiple modules, and dedicate one to testing).

I'll certainly be interested with help on the SSH / Mongoose v7 threading/crashing issues as it's a little bit outside of my experience :-)

GreyPeter, 2023-06-27:
I apologise for hijacking this thread but needed to ask a question to do with compiling using  ESP-IDFv5. I bought myself a new MacBook Air and now find I can now longer use ESP-IDFv3 to build OVMS. I have successfully compiled OVMS using sdkconfig.defaults.esp5.0.1 and after the mongoose patches but it won't install using the Firmware/Flash from file so am I correct in assuming that I will need to flash using the usb cable. If so, is it safe to do so. I don't want to damage my OVMS module.

llange, 2023-06-27:
Hi @GreyPeter,
I myself am using a MBP with an old 11.6.5, and the official procedure for ESP-IDF (whatever version) does work OK.
I do all my testing and flashes with the USB cable, which is (AFAIK) the most comprehensive method as it can allow you to recover from most of the issues you could have (flashing issue, flash layout change, etc...)

To speed up the operation, I use the `make app-flash` command on ESP-IDFv3 and `idf.py app-flash` for ESP-IDFv4+.

Let me know the issues you have (esp. error messages etc...) and do not hesitate to open a specific thread/issue for that.

GreyPeter, 2023-06-27:
@llange thank you for the reply. My laptop has the new M2 chip so I can only install ESP-IDFv4 and above. I will try to flash the build and see how it goes. I will of course advise of any issues that I have.

GreyPeter, 2023-07-01:
@llange I was able to copy mongoose V7.10 directly into the components/mongoose/mongoose directory and build/flash successfully. I am using esp v5.0.2 using this sdkconfig.defaults.esp5.0.2.
[sdkconfig.defaults.esp5.0.2.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/11924271/sdkconfig.defaults.esp5.0.2.zip)
Everything seems to be working correctly on the bench. I will now test it in my car.

werdnum, 2023-10-26:
I see the changes for ESP-IDF5 landed a little while ago: https://github.com/llange/Open-Vehicle-Monitoring-System-3/commit/4b6f8a70e2adddde49181a7541dfdeba65d95d57

Does that mean this is unblocked?

GreyPeter, 2023-11-02:
Sorry but compiling ESP-IDF V5 is still a work-in-progress. OVMS can be compiled but is still quite unstable.

llange, 2023-11-02:
Hi @GreyPeter , @werdnum ; sorry to have been silent such a long time - I unfortunately have been busy on other subjects, and I didn't have the time (and still don't) to make significant progresses on the ESPv5+ front.

Compiling is a first step, and I hope that we're going to maintain that "compilability" for the future.

Having a working, then usable, then stable system is another step that will necessitate some work. (and unfortunately I'm not able to coordinate it at the moment)

werdnum, 2023-11-02:
> Hi @GreyPeter , @werdnum ; sorry to have been silent such a long time - I unfortunately have been busy on other subjects, and I didn't have the time (and still don't) to make significant progresses [...]

Thanks for the update. Of course you are busy like all of us. I'm not sure how much time *I* have either, but if you do get the time to write down how community members might be able to help, perhaps I or some other interested party might be able to take a look.

Thanks for pursuing this, it's a very exciting prospect!

GreyPeter, 2023-11-03:
I am also happy to contribute if given some direction as to what to look into.
Thank you for all the work that you have done thus far.