openvehicles/Open-Vehicle-Monitoring-System-3 #756: Feature request: Help developers onboarding with a working docker container with ovms development

jollyjinx, 2022-10-28:
I really would like to improve OVMS, but the three machines I'm using ( raspberry pi with ubuntu arm64; macOS ventura arm64 and freebsd on x86-64 ) are not supported out of the box.
Maybe it's me, having decades of development on unix platforms, I always am taken aback when I have to start an install.sh and a second later I get greeted with some obscure error about outdated or not found stuff.
I tried with installing on the commandline inside docker containers but always hit a wall. Mostly cause the outdated idf toolchain ovms uses is not supporting modern arm processors.
And I know that I could get it running eventually, somewhat.
My motivation in getting something to work in OVMS is high, but my motivation to get the development environment is zero, especially when I read that OVMS is using old IDF tools and docker containers exist for newer tools already. My point is:

OVMS is a great platform, but if I want to program on it, it should be easy to get started. Docker is probably the easiest way today as it does not clutter up my machine like when I have to install things in /usr/local which will outdate so fast....

leres, 2022-10-31:
> I really would like to improve OVMS, but the three machines I'm using ( raspberry pi with ubuntu arm64; macOS ventura arm64 and freebsd on x86-64 ) are not supported out of the box.

I bought my first ovms module in 2018 and have only ever engaged in development under FreeBSD; I wrote the xtensa-esp32-elf port to provide the toolchain.

jollyjinx, 2022-11-02:
@leres - great work and exactly what I meant. Before one can start developing, setting up the development environment should not be a hurdle or port something before developing for ovms.

Anyways - maybe I can get this thing going in docker somehow. Right now I created a ubuntu 20.04 arm64 Docker with the necessary tools and mounted ovms into the docker container. After adding changes to the ```OVMS.V3/tools/esp-idf/tools/tools.json``` where to find updated compiler tools and changes to the ```OVMS.V3/tools/esp-idf/tools/idf_tools.py``` I was able to get ```install.sh``` and ```. ./startup.sh``` go through.

Then ```make menuconfig``` worked and nd now I'm stuck with c-compiler warnings and errors and erratic precompiler definitions when starting ```make```. 

I probably will not succeed cause the compiler is newer or - whatever. But without these hurdles I could have gotten right to the point and fix this stupid battery draining issue (ovms does not sleep when the battery is empty) which is the only reason I want to get it compile so I can make things better.

leres, 2022-11-02:
The toolchain (and compiler version) are tightly coupled. ovms currently uses v3.3.4 of esp-idf:
```
ice 29 % cat make/version.mk
IDF_VERSION_MAJOR := 3
IDF_VERSION_MINOR := 3
IDF_VERSION_PATCH := 4
```
And the [release notes](https://github.com/espressif/esp-idf/releases/tag/v3.3.4) say:
```
The 1.22.0-96-g2852398 GCC 5.2.0 toolchain version added in this release includes an improved PSRAM fix.
```
So that's the version you'll want to use.

jollyjinx, 2022-11-03:
@leres thanx a lot . I would have thought using a newer compiler ( the first I found compiled for arm64 ubuntu was way newer ) would work, but you are right, using the right compiler version got rid of the compile issues - doh .
It needs 1.22.0-97 and weirdly enough the gcc I'm using now in docker-arm64-ubuntu is an armel one, but it seems to work. It is currently still building but things are looking good. I wonder what it takes so long to build. And I thought compiling Swift is slow.
I keep my fingers crossed and hope I can get it working.

jollyjinx, 2022-11-04:
First Success!

I got an arm64 docker container with the ovms esp tools created that I then can compile OVMS.3 out of the box.

Dockerfile looks currently like this ( is a clone of the one insed esp/tools/docker ) but with a reference to a different esp version which includes references to the arm64 version.
```
FROM ubuntu:20.04

ARG DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    apt-utils \
    bison \
    ca-certificates \
    ccache \
    check \
    curl \
    dos2unix \
    flex \
    git \
    gperf \
    lcov \
    libncurses-dev \
    libusb-1.0-0-dev \
    make \
    cmake \
    ninja-build \
    python3 \
    python3-pip \
    python3.8-venv \
    unzip \
    wget \
    xz-utils \
    zip \
   && apt-get autoremove -y \
   && rm -rf /var/lib/apt/lists/* \
   && update-alternatives --install /usr/bin/python python /usr/bin/python3 10

#     cmake \ 

RUN python -m pip install --upgrade pip virtualenv

ARG IDF_CLONE_URL=https://github.com/jollyjinx/openvehicles-esp-idf.git
ARG IDF_CLONE_BRANCH_OR_TAG=v3.3.4-2
ARG IDF_CHECKOUT_REF=6d302b920

ENV IDF_PATH=/opt/esp/idf
ENV IDF_TOOLS_PATH=/opt/esp

RUN echo IDF_CHECKOUT_REF=$IDF_CHECKOUT_REF IDF_CLONE_BRANCH_OR_TAG=$IDF_CLONE_BRANCH_OR_TAG && \
    git clone --tags --recursive \
      ${IDF_CLONE_BRANCH_OR_TAG:+-b $IDF_CLONE_BRANCH_OR_TAG} \
      $IDF_CLONE_URL $IDF_PATH && \
    if [ -n "$IDF_CHECKOUT_REF" ]; then \
      cd $IDF_PATH && \
      git checkout $IDF_CHECKOUT_REF && \
      git submodule update --init --recursive; \
    fi

# Install all the required tools, plus CMake
RUN $IDF_PATH/tools/idf_tools.py --non-interactive install required \
  && $IDF_PATH/tools/idf_tools.py --non-interactive install cmake \
  && $IDF_PATH/tools/idf_tools.py --non-interactive install-python-env \
  && rm -rf $IDF_TOOLS_PATH/dist

RUN mkdir -p $HOME/.ccache && \
  touch $HOME/.ccache/ccache.conf

COPY entrypoint.sh /opt/esp/entrypoint.sh

ENTRYPOINT [ "/opt/esp/entrypoint.sh" ]
CMD [ "/bin/bash" ]

```

Which I build with:
```
# mkdir ovms
# cd ovms
# cp Dockerfile .
# docker build -t ovms:latest .
# git clone https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3.git ovms3
# docker run -v "$PWD/ovms3:/ovms" -ti ovms:latest
```
now in docker:
```
# cd /ovms3/vehicle/tools
# make menuconfig
# make flash
```

and after waiting a long time it's done building a omvs3.bin was in ovms3/vehicle/OVMS3/build which I copied on a sdcard, moved it in a ovms module and it correctly installed itself.
And the new binary is working !

It is a crude first working version but at least a working starting point.

kimmilde, 2023-02-22:
So if I just want to make some few new changes and try them. How can I do this? Can I clone ovms3 and make the changes in visual studio, and then build the changes (Will I get some error if the build failed?). I need to fix some faults, and be sure that they are working before making a pull request. This is for handling charging errors in soul

dexterbg, 2023-02-25:
Not sure if you're actually relating to this issue.

OVMS standard developer resources are:
- https://docs.google.com/document/d/1q5M9Lb5jzQhJzPMnkMKwy4Es5YK12ACQejX_NWEixr0
- http://lists.openvehicles.com/mailman/listinfo/ovmsdev

IamMattM, 2024-03-04:
Can a working docker container also be created for the ovms-server  V2 and V3(mqtt) ?

jamiejones85, 2025-02-03:
I'm struggling to build on my arm mac, works fine on my old intel mac.

Using this docker I'm getting a build error:

`In file included from /ovms/vehicle/OVMS.V3/components/ovms_cellular/src/ovms_cellular.h:43:0,
                 from /ovms/vehicle/OVMS.V3/main/ovms_peripherals.h:71,
                 from /ovms/vehicle/OVMS.V3/components/can/src/canlog_vfs.cpp:35:
/ovms/vehicle/OVMS.V3/components/ovms_cellular/src/gsmpppos.h:68:5: error: 'ppp_pcb' does not name a type
     ppp_pcb*     m_ppp;
     ^
`

Any tips or ideas?

markwj, 2025-02-06:
> I'm struggling to build on my arm mac, works fine on my old intel mac.
>
> Using this docker I'm getting a build error:
>
> `In file included from [...]

Can you show exactly what versions of tools and libraries you have installed. In particular:

1. OVMS firmware git repository:  git remote -v; git describe; git branch
2. Espressif IDF git repository:  git remote -v; git describe; git branch
3. extensa tools version

jamiejones85, 2025-02-06:
IDF git
`root@27b02d6f94f9:/opt/esp/idf# git remote -v; git describe; git branch
origin	https://github.com/jollyjinx/openvehicles-esp-idf.git (fetch)
origin	https://github.com/jollyjinx/openvehicles-esp-idf.git (push)
v3.3.4-2
* (no branch)`

`
root@27b02d6f94f9:/opt/esp/tools/xtensa-esp32-elf# ls
1.22.0-97-gc752ad5d-5.2.0
`

`
root@27b02d6f94f9:/ovms# git remote -v; git describe; git branch
origin	https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3.git (fetch)
origin	https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3.git (push)
3.3.003-1077-ge580b899
* master

`

I get the same error on my fork of OVMS that builds on my intel macbook, so I think it must be related to the tools/dev environment