openvehicles/Open-Vehicle-Monitoring-System-3 #766: CanLog: ability to automatically start (resume) logging after boot / reboot

llange, 2022-11-07:
I would like to be able to set in autostart the logging configuration, so that the logging starts right after the boot.
In this way, should the module reboot or go into sleep mode for various reasons, the logging would resume as soon as possible.

Use-case:  related to #748, #749, #750 : adding some datalogging capabilities to OVMS, in a fire & forget way.

Of course it's somehow related to #748 as we do not want to overwrite an existing log file. So there is an idea of cycling the name in some way to avoid that.

llange, 2022-11-08:
One possible solution : using the scripting system and events:

In the following files:
* `/store/events/system.shutdown/10-canlog`
* `/store/events/sd.unmounting//10-canlog`
* `/store/events/powermgmt.ovms.shutdown/10-canlog`
```
can log stop
```

In `/store/events/sd.mounted/10-canlog`:
```
can log start vfs crtd /sd/logs/autolog/autolog.crtd
```

dexterbg, 2022-11-09:
If you need to react to the system shutdown, you would attach to `system.shuttingdown` (begin shutdown) instead of `system.shutdown` (shutdown finished), and that will also be triggered by a power manager shutdown, so you don't need `powermgmt.ovms.shutdown`.

But you also don't need `system.shuttingdown`, as `sd.unmounting` will also be issued by any regular shutdown with the SD card still mounted.

Regarding the `sd.mounted` handler I would suggest renaming the `autolog.crtd` file and creating a new file, as the file may have become corrupted by a crash.

llange, 2022-11-20:
I think at this point we can now close this issue, as the event solution is working and there is no (for the moment) need for something more involved.

closed 2022-11-20 by llange