openvehicles/Open-Vehicle-Monitoring-System-3 #781: Using OVMS hardware with Arduino Framework

maxmacstn, 2022-11-23:
Hi, I'm currently doing research on vehicle telematics and I found that this project has amazing hardware that I can use for my research. However, I am more familiar with the Arduino framework and it's faster to develop using libraries on the internet. 

I use the schematics and the original code from the OVMS (written with ESP-IDF framework) as a base for initializing and utilizing the hardware. I'm now able to use the SIM7600 modem and IO expander MAX7317 without a problem.

One thing that I'm not able to use is the built-in CAN bus of the ESP32 (CAN 1).  I have checked with the oscilloscope and confirmed that the incoming data from the SN65 transceiver did arrive at IO26 of the ESP32.

Below is the example Arduino sketch that CAN bus of the OVMS is not receiving any incoming CAN messages, but this code works perfectly fine when I tested it with another ESP32 dev. board +[ SN65HVD230 module](https://www.waveshare.com/sn65hvd230-can-board.htm)

CAN Library and example code were taken from [Arduino-CAN](https://github.com/sandeepmistry/arduino-CAN)

```
#include <CAN.h>

void setup() {
  Serial.begin(115200);
  while (!Serial);
  CAN.setPins(26, 25); //OVMS CAN RX,TX
  // start the CAN bus at 500 kbps
  if (!CAN.begin(500E3)) {
    Serial.println("Starting CAN failed!");
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = CAN.parsePacket();

  if (packetSize) {
    // received a packet
    Serial.print("Received ");

    if (CAN.packetExtended()) {
      Serial.print("extended ");
    }

    if (CAN.packetRtr()) {
      // Remote transmission request, packet contains no data
      Serial.print("RTR ");
    }

    Serial.print("packet with id 0x");
    Serial.print(CAN.packetId(), HEX);

    if (CAN.packetRtr()) {
      Serial.print(" and requested length ");
      Serial.println(CAN.packetDlc());
    } else {
      Serial.print(" and length ");
      Serial.println(packetSize);

      // only print packet data for non-RTR packets
      while (CAN.available()) {
        Serial.print((char) CAN.read());
      }
      Serial.println();
    }

    Serial.println();
  }
}
```
In this code, I have omitted to set Rs pin of the SN65 transceiver to be HIGH because I needed only input data from CAN bus.

If someone can point out what I missed or what I should try, please let me know.

Bests,
Max

maxmacstn, 2022-11-24:
### Update: 
**I'm now able to use the ESP32's CAN bus on OVMS by using CAN driver from Espressif instead of using [Arduino-CAN](https://github.com/sandeepmistry/arduino-CAN) library.***

And Rs pin of the SN65 transceiver needs to be turned on in order to send ACK bit back to the CAN bus. However, If I turned on the TX of the transceiver while using the Arduino-CAN library. The whole CAN bus network crashed because ESP32 was continuously transmitting alternating LOW/HIGH signals out from CAN TX pin (IO25) as attached image.

I'm not sure why the library was causing this issue although it's working fine with the off-the-shelf CAN transceiver module. Perhaps there's something different between SN65HVD230 and SN65HVD233 on OVMS?

[image]

closed 2022-11-24 by maxmacstn

dexterbg, 2022-11-24:
Max, thanks for documenting the issue & solution.