openvehicles/Open-Vehicle-Monitoring-System-3 #791: Ioniq 28 Estimated range, average comsumption, TPMS

slajgaj, 2022-12-02:
Hello!
I found this project, when have TPMS data and average comsumption.
https://github.com/hokus15/pioniq

**Can you integrate from this to OVMS , and can you calculate in the OVMS the "estimated range" data from average comsumption and SOC? Where ideal range in 28kWh Ioniq = 200 **km****

dexterbg, 2022-12-02:
Yes & yes.
- TPMS integration is straight forward.
- Average consumption isn't available/decoded from the car. The "pioniq" project also doesn't read this from the car. It needs to be derived from speed & power or distance & energy used, so needs some more work, but is also possible.

slajgaj, 2022-12-02:
Hello MichaelI am very happy to hear this.  Many people believe that OVMS development for the Ioniq 28 has stopped.  I’m in contact with many curious Ioniq owners in Hungary, and many would be interested in OVMS!I believe that the OVMS team is very good and will solve it as soon as possible the average consumption and TPMSPs: i thinking in the “pionoq” project have calculating the av.compI saw the EV Notify project too and this project have an adjustable parameter for average consumption and this  parameter and SOC with calculateGood luck for you and for your teams, and we look forward to the developments!Best regardsLajos TothAz iPhone-omról küldve

dexterbg, 2022-12-09:
Range estimations & TPMS are now included in a test build, that can be installed from here:
- `ovms.dexters-web.de/f/firmware/test/ovms3.bin`

Feedback is welcome.

If you encounter issues, please set component `v-hyundaivfl` to log level `verbose`, `events` to `debug` and send me the log excerpt with your issue description.

slajgaj, 2022-12-09:
Michael!That is great!You are the genius programmer, thank you very much!2 question please:-can you received my bank transfer ? (I'm excited about it)- can you have a next idea for the Ioniq OVMS?