openvehicles/Open-Vehicle-Monitoring-System-3 #804: Nissan Leaf Climate Control command not working

nismoman240sx, 2023-01-18:
I have a 2015 Nissan leave with the OVMS module. Recently, the CC command has stopped working. Someone else is having the same problem. All other functions seem fine. The other user reset the OVMS module back to it's original firmware and says the CC function is now working properly. I'm updated to the most recent update that came out on 01/17/2023, but the issue persists. I tried contacting OVMS about it, but they said to post the issue here. Any help or updates would be greatly appreciated.

nismoman240sx, 2023-01-18:
I just reverted to the main firmware and I can also confirm that the issue is resolved by reverting to the main firmware version. When I switch it back to the newest firmware, the problem comes back. 

Also, I tried setting in the car when sending the command, and I can hear the car close and then open some relays, but the HVAC system never illuminates or does anything on the dash like it normally would when you use the remote command.

dexterbg, 2023-01-18:
- Please specify the full firmware versions you're referring to.
- Please provide all/any crash details.
- Please provide a verbose log excerpt of the faulty action.

nismoman240sx, 2023-01-18:
Factory firmware image that works; 3.3.001-33-g58d01654
OTA Firmware image that works; 3.3.002 (build idf v3.3.4-848-g1ff5e24b1)
OTA Firmware image that does NOT work; 3.3.003 (build idf v3.3.4-848-g1ff5325b1)

After switching between all of the firmware types to get the firmware info, it seems to work now, but that's with the network awake. Every time I've had an issue, it's been with it asleep. I'll try it again after it goes to sleep. 

I've never seen any crash details as I haven't seen a crash per-se. When you would send the command, the car just wouldn't respond appropriately. 

This is the log I have from just now, but as I said, it "failed to fail" this time. I'll try again later. 
ShellLog Monitor
I (258047) webserver: HTTP GET /status
I (272477) webserver: HTTP GET /home
I (273157) ovms-server-v2: Send MP-0 S79.0,M,0,0.00,stopped,,54,36,0,0,0,1,13,21,-1,0,0,0,51.61,0,0,0,0,-1,0,0,0,0,0,45,0,0.00,385.50,83,0.00,100.00,-0.50,0.4
I (273167) ovms-server-v2: Send MP-0 D64,8,4,6.11111,6.66667,5,0,273230,0,21287,3.5,0,1,1,13.74,0,12.6,16,0,0,4.44444
I (273167) ovms-server-v2: Send MP-0 L39.028275,-95.740067,0,302.9,1,1,0,0,0,-0.193,0.653,0.132,0,100,AA,15,0.7,0.0,100
I (274557) webserver: HTTP GET /edit
I (274707) webserver: HTTP POST /api/execute
I (274717) webcommand: HttpCommandStream[0x3f8c096c]: 3479128 bytes free, executing: vfs ls '/store'
I (286397) webserver: HTTP GET /home
I (287657) webserver: HTTP GET /shell
I (297487) webserver: HTTP GET /shell
I (303167) housekeeping: 2023-01-18 15:51:20 CST (RAM: 8b=78976-79808 32b=12344 SPI=3403636-3410312)
I (304667) webserver: HTTP GET /home
I (334157) ovms-server-v2: Send MP-0 S79.0,M,0,0.00,stopped,,54,36,0,0,0,1,13,21,-1,0,0,0,51.61,0,0,0,0,-1,0,0,0,0,0,45,0,0.00,385.50,83,0.00,100.00,-0.50,0.4
I (334157) ovms-server-v2: Send MP-0 D64,8,4,6.11111,6.66667,5,0,273230,0,21348,3.5,0,0,0,13.72,0,12.6,16,0,0,4.44444
I (334167) ovms-server-v2: Send MP-0 L39.028275,-95.740067,0,303.1,1,1,0,0,0,-0.193,0.653,0.132,0,100,AA,14,0.7,0.0,100
I (395157) ovms-server-v2: Send MP-0 S79.0,M,0,0.00,stopped,,54,36,0,0,0,1,13,21,-1,0,0,0,51.61,0,0,0,0,-1,0,0,0,0,0,45,0,0.00,385.50,83,0.00,100.00,-0.50,0.4
I (395157) ovms-server-v2: Send MP-0 D64,8,4,6.11111,6.66667,5,0,273230,0,21409,3.5,0,0,0,13.7,0,12.6,16,0,0,4.44444
I (395167) ovms-server-v2: Send MP-0 L39.028275,-95.740067,0,303.3,1,1,0,0,0,-0.193,0.653,0.132,0,100,AA,15,0.7,0.0,100
I (406507) ovms-server-v2: Incoming Msg: MP-0 Z2
I (406507) ovms-server-v2: One or more peers have connected
I (407157) ovms-server-v2: Send MP-0 S79.0,M,0,0.00,stopped,,54,36,0,0,0,1,13,21,-1,0,0,0,51.61,0,0,0,0,-1,0,0,0,0,0,45,0,0.00,385.50,83,0.00,100.00,-0.50,0.4
I (407157) ovms-server-v2: Send MP-0 D64,8,4,6.11111,6.66667,5,0,273230,0,21421,3.5,0,0,0,13.7,0,12.6,16,0,0,4.44444
I (407167) ovms-server-v2: Send MP-0 L39.028275,-95.740067,0,303.3,1,1,0,0,0,-0.193,0.653,0.132,0,100,AA,15,0.7,0.0,100
I (407167) ovms-server-v2: Send MP-0 Y4,FL,FR,RL,RR,4,255.1,255.1,256.8,256.8,1,0,-1,0,-1,0,-1
I (407167) ovms-server-v2: Send MP-0 W37,0,37.25,0,37,0,37.25,0,1
I (407177) ovms-server-v2: Send MP-0 F3.3.003/ota_1/main (build idf v3.3.4-848-g1ff5e24b1 Sep 1 2022 08:40:30),,19,0,NL,BoatGordHizy,-1,-1,OVMS WIFI BLE BT cores=2 rev=ESP32/3; MODEM SIM7600
I (449277) webserver: HTTP GET /home
I (450717) webserver: HTTP GET /shell
I (468157) ovms-server-v2: Send MP-0 S79.0,M,0,0.00,stopped,,54,36,0,0,0,1,13,21,-1,0,0,0,51.61,0,0,0,0,-1,0,0,0,0,0,45,0,0.00,385.50,83,0.00,100.00,-0.50,0.4
I (468167) ovms-server-v2: Send MP-0 D64,8,4,6.11111,6.66667,5,0,273230,0,21482,3.5,0,0,0,13.69,0,12.6,16,0,0,4.44444
I (468167) ovms-server-v2: Send MP-0 L39.028275,-95.740067,0,302.6,1,1,0,0,0,-0.193,0.653,0.132,0,100,AA,13,0.8,0.0,100
I (469377) ovms-server-v2: Incoming Msg: MP-0 Z1
I (476037) webserver: HTTP POST /api/execute
OVMS# climate control on 
Unrecognised command
Usage: climatecontrol off|on
I (476037) webcommand: HttpCommandStream[0x3f829b38]: 3479456 bytes free, executing: climate control on 
I (485227) webserver: HTTP POST /api/execute
OVMS# climatecontrol on 
Climate Control 
on
I (485237) webcommand: HttpCommandStream[0x3f829b4c]: 3481740 bytes free, executing: climatecontrol on 
I (485237) v-nissanleaf: CommandClimateControl
I (485237) v-nissanleaf: RemoteCommandHandler
I (485237) v-nissanleaf: Sending Wakeup Frame
E (485237) can: can1: intr=20671 rxpkt=20435 txpkt=190 errflags=0x8040d9 rxerr=0 txerr=130 rxinval=0 rxovr=0 txovr=0 txdelay=149 txfail=15 wdgreset=0 errreset=0
E (485237) can: can1: intr=20673 rxpkt=20435 txpkt=190 errflags=0x8040d9 rxerr=0 txerr=130 rxinval=0 rxovr=0 txovr=0 txdelay=149 txfail=16 wdgreset=0 errreset=0
E (485237) can: can1: intr=20675 rxpkt=20435 txpkt=190 errflags=0x8040d9 rxerr=0 txerr=130 rxinval=0 rxovr=0 txovr=0 txdelay=149 txfail=17 wdgreset=0 errreset=0
I (485337) v-nissanleaf: RemoteCommandTimer 24
I (485337) v-nissanleaf: OLD TCU on EV Bus
I (485337) v-nissanleaf: Sending Wakeup Frame
I (485337) v-nissanleaf: Enable Climate Control
I (485437) v-nissanleaf: RemoteCommandTimer 23
I (485437) v-nissanleaf: OLD TCU on EV Bus
I (485437) v-nissanleaf: Sending Wakeup Frame
I (485437) v-nissanleaf: Enable Climate Control
I (485537) v-nissanleaf: RemoteCommandTimer 22
I (485537) v-nissanleaf: OLD TCU on EV Bus
I (485537) v-nissanleaf: Sending Wakeup Frame
[... 423 lines cut ...]
I (614437) v-nissanleaf: Disable Climate Control
I (614537) v-nissanleaf: RemoteCommandTimer 4
[... cut ...]
I (658167) ovms-server-v2: Send MP-0 L39.028271,-95.740059,0,303,1,1,0,0,0,-0.193,0.653,0.132,0,100,AA,15,0.7,0.0,100
I (719157) ovms-server-v2: Send MP-0 S79.0,M,0,0.00,stopped,,54,36,0,0,0,1,13,21,-1,0,0,0,51.61,0,0,0,0,-1,0,0,0,0,0,45,0,0.00,385.50,83,0.00,100.00,-0.50,0.4
I (719157) ovms-server-v2: Send MP-0 D64,8,4,6.11111,6.66667,5,0,273230,0,21733,4,0,1,1,14.07,0,12.6,16,0,0,4.44444
I (719167) ovms-server-v2: Send MP-0 L39.028267,-95.740051,0,303.7,1,1,0,0,0,-0.193,0.653,0.132,0,100,AA,12,0.9,0.0,100
climatecontrol off

mjkapkan, 2023-01-19:
Hi, I see the last change to NL code was over a year ago. But it could be that the users usually dont use OTA feature and this got unnoticed. I will verify it myself and check at which change it stopped working. Meanwhile @nismoman240sx, have you tried to change model year in settings to like 2013 or 2018 to  see of that helpa?

nismoman240sx, 2023-01-19:
The vehicle is a 2015, but was built in 2014, so I did try to change it to 2014, but at that time, it didn't change anything. I didn't try any other years. I will try that.

mjkapkan, 2023-01-19:
Good to hear that it works now! There were no changes to the Nissan Leaf funcitonality for over a year, so no recent changes could have caused than. Most liekely something with the firmware itself got corrupted so hard reset cleared the issue.

nismoman240sx, 2023-01-20:
Remote CC failed to work this morning from my app. I then logged in via the wifi network and attempted to send the command directly from the shell editor. It still failed to activate. I tried again with no effect. I then sent the command from the app a couple of times. Same result. I then tried to send the wakeup command from the app. The car won't wake up. Here is the log;
ShellLog Monitor
I (70782868) webserver: HTTP GET /shell
I (70785648) webserver: HTTP POST /shell
I (70785658) webserver: HandleLogin: 'admin' logged in, sid f904df628026a840
I (70785688) webserver: HTTP GET /menu
I (70785698) webserver: HTTP GET /shell
I (70802198) housekeeping: 2023-01-20 08:04:46 CST (RAM: 8b=76548-79776 32b=12344 SPI=3386608-3409952)
OVMS# climatecontrol on 
undefined
I (70836188) ovms-server-v2: Send MP-0 S100.0,M,0,0.00,timerwait,,68,43,0,0,0,102,5,14,-1,0,0,0,51.61,58,0,0,0,-1,0,0,0,0,0,43,0,0.00,395.00,83,0.00,100.00,-1.00,0.8
I (70836198) ovms-server-v2: Send MP-0 D76,0,5,3.88889,4.44444,2,0,273320,0,52452,-2.5,0,0,0,13.29,0,12.6,16,0,0,-8.88889
I (70836198) ovms-server-v2: Send MP-0 L39.028233,-95.740021,0,302.8,1,1,0.0,0,0,-0.395,1.263,0.184,0,100,AA,15,0.7,0.0,100
I (70843878) webserver: HTTP POST /api/execute
I (70843878) webcommand: HttpCommandStream[0x3f8c0670]: 3477288 bytes free, executing: climatecontrol on 
OVMS# climatecontrol on 
Climate Control 
on
I (70843888) v-nissanleaf: CommandClimateControl
I (70843888) v-nissanleaf: RemoteCommandHandler
I (70843888) v-nissanleaf: Sending Wakeup Frame
I (70843988) v-nissanleaf: RemoteCommandTimer 24
I (70843988) v-nissanleaf: OLD TCU on EV Bus
I (70843988) v-nissanleaf: Sending Wakeup Frame
I (70843988) v-nissanleaf: Enable Climate Control
I (70844088) v-nissanleaf: RemoteCommandTimer 23
I (70844088) v-nissanleaf: OLD TCU on EV Bus
I (70844088) v-nissanleaf: Sending Wakeup Frame
I (70844088) v-nissanleaf: Enable Climate Control
I (70844188) v-nissanleaf: RemoteCommandTimer 22
I (70844188) v-nissanleaf: OLD TCU on EV Bus
I (70844188) v-nissanleaf: Sending Wakeup Frame
I (70844188) v-nissanleaf: Enable Climate Control
I (70844288) v-nissanleaf: RemoteCommandTimer 21
I (70844288) v-nissanleaf: OLD TCU on EV Bus
I (70844288) v-nissanleaf: Sending Wakeup Frame
I (70844288) v-nissanleaf: Enable Climate Control
I (70844388) v-nissanleaf: RemoteCommandTimer 20
I (70844388) v-nissanleaf: OLD TCU on EV Bus
I (70844388) v-nissanleaf: Sending Wakeup Frame
I (70844388) v-nissanleaf: Enable Climate Control
I (70844488) v-nissanleaf: RemoteCommandTimer 19
I (70844488) v-nissanleaf: OLD TCU on EV Bus
I (70844488) v-nissanleaf: Sending Wakeup Frame
I (70844488) v-nissanleaf: Enable Climate Control
I (70844588) v-nissanleaf: RemoteCommandTimer 18
I (70844588) v-nissanleaf: OLD TCU on EV Bus
I (70844588) v-nissanleaf: Sending Wakeup Frame
I (70844588) v-nissanleaf: Enable Climate Control
I (70844688) v-nissanleaf: RemoteCommandTimer 17
I (70844688) v-nissanleaf: OLD TCU on EV Bus
I (70844688) v-nissanleaf: Sending Wakeup Frame
I (70844688) v-nissanleaf: Enable Climate Control
I (70844788) v-nissanleaf: RemoteCommandTimer 16
I (70844788) v-nissanleaf: OLD TCU on EV Bus
I (70844788) v-nissanleaf: Sending Wakeup Frame
I (70844788) v-nissanleaf: Enable Climate Control
I (70844888) v-nissanleaf: RemoteCommandTimer 15
I (70844888) v-nissanleaf: OLD TCU on EV Bus
I (70844888) v-nissanleaf: Sending Wakeup Frame
I (70844888) v-nissanleaf: Enable Climate Control
I (70844988) v-nissanleaf: RemoteCommandTimer 14
I (70844988) v-nissanleaf: OLD TCU on EV Bus
I (70844988) v-nissanleaf: Sending Wakeup Frame
I (70844988) v-nissanleaf: Enable Climate Control
I (70845088) v-nissanleaf: RemoteCommandTimer 13
I (70845088) v-nissanleaf: OLD TCU on EV Bus
I (70845088) v-nissanleaf: Sending Wakeup Frame
I (70845088) v-nissanleaf: Enable Climate Control
I (70845188) v-nissanleaf: RemoteCommandTimer 12
[... 261 lines cut ...]
I (70942298) v-nissanleaf: Enable Climate Control
I (70942398) v-nissanleaf: RemoteCommandTimer 3
I (70942398) v-nissanleaf: OLD TCU on EV Bus
I (70942398) v-nissanleaf: Sending Wakeup Frame
I (70942398) v-nissanleaf: Enable Climate Control
I (70942498) v-nissanleaf: RemoteCommandTimer 2
I (70942498) v-nissanleaf: OLD TCU on EV Bus
I (70942498) v-nissanleaf: Sending Wakeup Frame
I (70942498) v-nissanleaf: Enable Climate Control
I (70942598) v-nissanleaf: RemoteCommandTimer 1
I (70942598) v-nissanleaf: OLD TCU on EV Bus
I (70942598) v-nissanleaf: Sending Wakeup Frame
I (70942598) v-nissanleaf: Enable Climate Control
I (70942698) v-nissanleaf: RemoteCommandTimer 0
I (70943498) v-nissanleaf: CcDisableTimer
I (70943498) v-nissanleaf: OLD TCU on EV Bus
I (70943498) v-nissanleaf: Sending Wakeup Frame
I (70943498) v-nissanleaf: Auto Disable Climate Control
I (70970298) ovms-server-v2: Incoming Msg: MP-0 C18
I (70970298) v-nissanleaf: Sending Wakeup Frame
I (70970298) ovms-server-v2: Send MP-0 c18,0
I (70994188) ovms-server-v2: Send MP-0 S100.0,M,0,0.00,timerwait,,68,43,0,0,0,102,5,14,-1,0,0,0,51.61,58,0,0,0,-1,0,0,0,0,0,43,0,0.00,395.00,83,0.00,100.00,-1.00,0.8
I (70994198) ovms-server-v2: Send MP-0 D76,0,5,3.88889,4.44444,2,0,273320,0,52610,-2.5,0,0,0,13.29,0,12.6,16,0,0,-8.88889
I (70994198) ovms-server-v2: Send MP-0 L39.028233,-95.740021,0,302.8,1,1,0.0,0,0,-0.395,1.263,0.184,0,100,AA,14,0.7,0.0,100
climatecontrol on

nismoman240sx, 2023-01-20:
I see the " OLD TCU on EV Bus" line repeated. Could it mean that I don't have the TCU properly unplugged? I thought I just unplugged everything, but maybe I'm remembering incorrectly?

biohead, 2023-01-22:
I see similar behaviour as described. Dropped back to main and it worked well the first time, but never worked again. Soft resets didn't recover it, but a hard power reset did. Another 2015 model.
I've switched back to edge OTAs, enabled logging and will report back. I also see "Old TCU..." repeated a lot in the verbose logs despite only just having enabled logging.

nismoman240sx, 2023-01-23:
I tried to get in to it again today to try messing with it again, and now the car's wifi network is gone. I've tried on my phone, my laptop, and my PC. None of them can find it now. I'm going to go get an SD card today and start logging to it to try to get more debugging information.

nismoman240sx, 2023-01-27:
There are others having issues as well now. Not with just the HVAC controls, but the OVMS unit no longer communicating via cellular or Wi-Fi. One user seemed to think that the EDGE firmware caused the issue and that moving to EAP resolved the issue.

mjkapkan, 2023-01-29:
@nismoman240sx, I don't have my Leaf currently. Will have it back in a week so can retest. I recently also had some strange issues with OVMS. I did a hard reset and installed Latest `main` firmware version. That seems to have fixed the issue.
If hard reset + main firmware is not helping, you could try taking out the modem and try using wifi only version. If even that does not help, it could be a faulty OVMS unit (happened to me once).

nismoman240sx, 2023-02-07:
I finally found time to get the OVMS unit pulled out so I could do a factory reset on it. So far, it seems to be working again, but we'll see how long it lasts. I'll come back and close this if it doesn't act up again.

nismoman240sx, 2023-02-09:
Annnd it's not working again. I didn't bother downloading the logs again, as it's the EXACT same log output as before. What it appears to be trying to do is to get the car to wake up so it can activate the climate control, but the car just seems to refuse to wake up. Even sending the wakeup command by itself will not wake up the car. I'm starting to lean towards a hardware issue, but the fact that rebooting gets it to work for a bit makes me question that.

nismoman240sx, 2023-02-10:
I have some different information now. I've got the issue narrowed down to something repeatable. First of all, the firmware versions I've tested this with are the same as above. The issue is repeatable across firmware versions. If you reboot and then watch the log monitor you can see something odd happening. So the first few times you send the climatecontrol on and off commands, it will work, but there are "errors" reported in the log. Then the commands stops working, but the errors also stop. It appears to me to be associated with the Wakeup Frame. This log happens to be with the latest main firmware, but it's the same for all version. This log is immediately after a reboot.

Log Monitor
I (47683) webserver: HTTP POST /cfg/firmware
I (47683) webserver: HandleLogin: 'admin' logged in, sid e069cb37751a8820
I (47723) webserver: HTTP GET /cfg/firmware
I (48203) cellular: Network Registration status: RegisteredRoaming
I (48203) cellular: Network Provider is: T-Mobile Hologram
I (48523) webserver: HTTP GET /menu
I (49173) cellular: State: Enter NetStart state
I (50253) cellular: PPP Connection is ready to start
I (51173) cellular: State: Enter NetMode state
I (51173) gsm-ppp: Initialising...
I (51263) gsm-ppp: StatusCallBack: None
I (51263) gsm-ppp: status_cb: Connected
I (51273) gsm-ppp: our_ipaddr = 100.70.11.207
I (51273) gsm-ppp: his_ipaddr = 10.64.64.64
I (51273) gsm-ppp: netmask = 255.255.255.255
I (51273) gsm-ppp: DNS#0 = 8.8.8.8
I (51273) gsm-ppp: DNS#1 = 8.8.4.4
I (51273) gsm-ppp: our6_ipaddr = ::
I (51273) netmanager: Set DNS#1 0.0.0.0
I (51273) netmanager: Set DNS#2 0.0.0.0
I (51273) netmanager: MODEM up (with WIFI client up): staying with WIFI client priority
I (51903) webserver: HTTP GET /home
I (54533) webserver: HTTP GET /shell
I (83783) webserver: HTTP POST /api/execute
OVMS# climatecontrol on 
Climate Control 
on
I (83783) webcommand: HttpCommandStream[0x3f8c0d68]: 3480788 bytes free, executing: climatecontrol on 
I (83783) v-nissanleaf: CommandClimateControl
I (83783) v-nissanleaf: RemoteCommandHandler
I (83783) v-nissanleaf: Sending Wakeup Frame
E (83783) can: can1: intr=1 rxpkt=0 txpkt=0 errflags=0x8000d9 rxerr=0 txerr=8 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=0 wdgreset=0 errreset=0
E (83783) can: can1: intr=8 rxpkt=0 txpkt=0 errflags=0x8000d9 rxerr=0 txerr=64 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=0 wdgreset=0 errreset=0
E (83783) can: can1: intr=11 rxpkt=0 txpkt=0 errflags=0x8000d9 rxerr=0 txerr=88 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=0 wdgreset=0 errreset=0
E (83783) can: can1: intr=13 rxpkt=0 txpkt=0 errflags=0x8040d9 rxerr=0 txerr=104 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=0 wdgreset=0 errreset=0
E (83783) can: can1: intr=15 rxpkt=0 txpkt=0 errflags=0x8040d9 rxerr=0 txerr=120 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=0 wdgreset=0 errreset=0
E (83783) can: can1: intr=18 rxpkt=0 txpkt=0 errflags=0x204800 rxerr=0 txerr=128 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=0 wdgreset=0 errreset=0
E (83783) can: can1: intr=18 rxpkt=0 txpkt=0 errflags=0x204800 rxerr=0 txerr=128 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=1 wdgreset=0 errreset=0
E (83783) can: can1: intr=20 rxpkt=0 txpkt=0 errflags=0x8040d9 rxerr=0 txerr=128 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=2 wdgreset=0 errreset=0
E (83783) can: can1: intr=22 rxpkt=0 txpkt=0 errflags=0x8040d9 rxerr=0 txerr=128 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=3 wdgreset=0 errreset=0
I (83893) v-nissanleaf: RemoteCommandTimer 24
I (83893) v-nissanleaf: OLD TCU on EV Bus
I (83893) v-nissanleaf: Sending Wakeup Frame
I (83893) v-nissanleaf: Enable Climate Control
I (83993) v-nissanleaf: RemoteCommandTimer 23
I (83993) v-nissanleaf: OLD TCU on EV Bus
I (83993) v-nissanleaf: Sending Wakeup Frame
I (83993) v-nissanleaf: Enable Climate Control
I (84093) v-nissanleaf: RemoteCommandTimer 22
I (84093) v-nissanleaf: OLD TCU on EV Bus
I (84093) v-nissanleaf: Sending Wakeup Frame
I (84093) v-nissanleaf: Enable Climate Control
I (84193) v-nissanleaf: RemoteCommandTimer 21
I (84193) v-nissanleaf: OLD TCU on EV Bus
I (84193) v-nissanleaf: Sending Wakeup Frame
I (84193) v-nissanleaf: Enable Climate Control
I (84293) v-nissanleaf: RemoteCommandTimer 20
I (84293) v-nissanleaf: OLD TCU on EV Bus
I (84293) v-nissanleaf: Sending Wakeup Frame
I (84293) v-nissanleaf: Enable Climate Control
I (84393) v-nissanleaf: RemoteCommandTimer 19
I (84393) v-nissanleaf: OLD TCU on EV Bus
I (84393) v-nissanleaf: Sending Wakeup Frame
I (84393) v-nissanleaf: Enable Climate Control
I (84493) v-nissanleaf: RemoteCommandTimer 18
I (84493) v-nissanleaf: OLD TCU on EV Bus
I (84493) v-nissanleaf: Sending Wakeup Frame
[... 565 lines cut ...]
I (196903) v-nissanleaf: Disable Climate Control
I (197003) v-nissanleaf: RemoteCommandTimer 14
I (197003) v-nissanleaf: OLD TCU on EV Bus
I (197003) v-nissanleaf: Sending Wakeup Frame
I (197003) v-nissanleaf: Disable Climate Control
I (197103) v-nissanleaf: RemoteCommandTimer 13
I (197103) v-nissanleaf: OLD TCU on EV Bus
I (197103) v-nissanleaf: Sending Wakeup Frame
I (197103) v-nissanleaf: Disable Climate Control
I (197203) v-nissanleaf: RemoteCommandTimer 12
I (197203) v-nissanleaf: OLD TCU on EV Bus
I (197203) v-nissanleaf: Sending Wakeup Frame
I (197203) v-nissanleaf: Disable Climate Control
I (197303) v-nissanleaf: RemoteCommandTimer 11
I (197303) v-nissanleaf: OLD TCU on EV Bus
I (197303) v-nissanleaf: Sending Wakeup Frame
I (197303) v-nissanleaf: Disable Climate Control
I (197403) v-nissanleaf: RemoteCommandTimer 10
I (197403) v-nissanleaf: OLD TCU on EV Bus
I (197403) v-nissanleaf: Sending Wakeup Frame
I (197403) v-nissanleaf: Disable Climate Control
I (197503) v-nissanleaf: RemoteCommandTimer 9
I (197503) v-nissanleaf: OLD TCU on EV Bus
I (197503) v-nissanleaf: Sending Wakeup Frame
I (197503) v-nissanleaf: Disable Climate Control

dexterbg, 2023-02-11:
I don't know if that's related to your Leaf issue, but the CAN1 error flags switching from 0x8… to 0x4… indicates a permanent bus-off state.

I'm having the very same CAN1 issue after switching Wifi AP mode off. Did you change anything about your Wifi config lately?

I posted my findings on the developer list and implemented a workaround, which is included in the current edge build. The workaround detects the issue and performs a bus reset to solve it. Please try.

See also http://lists.openvehicles.com/pipermail/ovmsdev/2023-January/016070.html

nismoman240sx, 2023-02-13:
I haven't made any changes before the issue started.  The module has been set to WiFi AP + client mode since the day it was installed and I've never changed that.

I installed the latest edge version firmware and tested. 

Here's the log from the shell monitor after updating firmware;
OVMS# climatecontrol on 
Climate Control 
on
I (1030214) v-nissanleaf: CommandClimateControl
I (1030214) v-nissanleaf: RemoteCommandHandler
I (1030224) v-nissanleaf: Sending Wakeup Frame
E (1030224) can: can1: intr=373858 rxpkt=373325 txpkt=435 errflags=0x8000d9 rxerr=0 txerr=39 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=3 wdgreset=0 errreset=0
E (1030224) can: can1: intr=373866 rxpkt=373325 txpkt=435 errflags=0x8440d9 rxerr=0 txerr=103 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=3 wdgreset=0 errreset=0
E (1030224) can: can1: intr=373868 rxpkt=373325 txpkt=435 errflags=0x8040d9 rxerr=0 txerr=119 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=3 wdgreset=0 errreset=0
E (1030224) can: can1: intr=373872 rxpkt=373325 txpkt=435 errflags=0x204800 rxerr=0 txerr=135 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=3 wdgreset=0 errreset=0
E (1030224) can: can1: intr=373872 rxpkt=373325 txpkt=435 errflags=0x204800 rxerr=0 txerr=135 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=4 wdgreset=0 errreset=0
I (1030324) v-nissanleaf: RemoteCommandTimer 24
I (1030324) v-nissanleaf: OLD TCU on EV Bus
I (1030324) v-nissanleaf: Sending Wakeup Frame
I (1030324) v-nissanleaf: Enable Climate Control
I (1030424) v-nissanleaf: RemoteCommandTimer 23
I (1030424) v-nissanleaf: OLD TCU on EV Bus
I (1030424) v-nissanleaf: Sending Wakeup Frame
I (1030424) v-nissanleaf: Enable Climate Control
I (1030524) v-nissanleaf: RemoteCommandTimer 22
I (1030524) v-nissanleaf: OLD TCU on EV Bus
I (1030524) v-nissanleaf: Sending Wakeup Frame
I (1030524) v-nissanleaf: Enable Climate Control
I (1030624) v-nissanleaf: RemoteCommandTimer 21
I (1030624) v-nissanleaf: OLD TCU on EV Bus
I (1030624) v-nissanleaf: Sending Wakeup Frame
I (1030624) v-nissanleaf: Enable Climate Control
I (1030724) v-nissanleaf: RemoteCommandTimer 20
I (1030724) v-nissanleaf: OLD TCU on EV Bus
I (1030724) v-nissanleaf: Sending Wakeup Frame
I (1030724) v-nissanleaf: Enable Climate Control
I (1030824) v-nissanleaf: RemoteCommandTimer 19
I (1030824) v-nissanleaf: OLD TCU on EV Bus
I (1030824) v-nissanleaf: Sending Wakeup Frame
I (1030824) v-nissanleaf: Enable Climate Control
I (1030924) v-nissanleaf: RemoteCommandTimer 18
I (1030924) v-nissanleaf: OLD TCU on EV Bus
I (1030924) v-nissanleaf: Sending Wakeup Frame
I (1030924) v-nissanleaf: Enable Climate Control
I (1031024) v-nissanleaf: RemoteCommandTimer 17
I (1031024) v-nissanleaf: OLD TCU on EV Bus
I (1031024) v-nissanleaf: Sending Wakeup Frame
I (1031024) v-nissanleaf: Enable Climate Control
I (1031124) v-nissanleaf: RemoteCommandTimer 16
I (1031124) v-nissanleaf: OLD TCU on EV Bus
I (1031124) v-nissanleaf: Sending Wakeup Frame
I (1031124) v-nissanleaf: Enable Climate Control
I (1031224) v-nissanleaf: RemoteCommandTimer 15
I (1031224) v-nissanleaf: OLD TCU on EV Bus
I (1031224) v-nissanleaf: Sending Wakeup Frame
I (1031224) v-nissanleaf: Enable Climate Control
I (1031324) v-nissanleaf: RemoteCommandTimer 14
I (1031324) v-nissanleaf: OLD TCU on EV Bus
I (1031324) v-nissanleaf: Sending Wakeup Frame
I (1031324) v-nissanleaf: Enable Climate Control
I (1031424) v-nissanleaf: RemoteCommandTimer 13
I (1031424) v-nissanleaf: OLD TCU on EV Bus
I (1031424) v-nissanleaf: Sending Wakeup Frame
I (1031424) v-nissanleaf: Enable Climate Control
I (1031524) v-nissanleaf: RemoteCommandTimer 12
I (1031524) v-nissanleaf: OLD TCU on EV Bus
I (1031524) v-nissanleaf: Sending Wakeup Frame
I (1031524) v-nissanleaf: Enable Climate Control
I (1031624) v-nissanleaf: RemoteCommandTimer 11
I (1031624) v-nissanleaf: OLD TCU on EV Bus
[... 344 lines cut ...]
I (1080524) v-nissanleaf: Disable Climate Control
I (1080624) v-nissanleaf: RemoteCommandTimer 5
I (1080624) v-nissanleaf: OLD TCU on EV Bus
I (1080624) v-nissanleaf: Sending Wakeup Frame
I (1080624) v-nissanleaf: Disable Climate Control
I (1080724) v-nissanleaf: RemoteCommandTimer 4
I (1080724) v-nissanleaf: OLD TCU on EV Bus
I (1080724) v-nissanleaf: Sending Wakeup Frame
I (1080724) v-nissanleaf: Disable Climate Control
I (1080824) v-nissanleaf: RemoteCommandTimer 3
I (1080824) v-nissanleaf: OLD TCU on EV Bus
I (1080824) v-nissanleaf: Sending Wakeup Frame
I (1080824) v-nissanleaf: Disable Climate Control
I (1080924) v-nissanleaf: RemoteCommandTimer 2
I (1080924) v-nissanleaf: OLD TCU on EV Bus
I (1080924) v-nissanleaf: Sending Wakeup Frame
I (1080924) v-nissanleaf: Disable Climate Control
I (1081024) v-nissanleaf: RemoteCommandTimer 1
I (1081024) v-nissanleaf: OLD TCU on EV Bus
I (1081024) v-nissanleaf: Sending Wakeup Frame
I (1081024) v-nissanleaf: Disable Climate Control
I (1081124) v-nissanleaf: RemoteCommandTimer 0
climatecontrol off

As you can see, the only issue occurs when the first "climatecontrol on" command is sent, and it's different than the other errors seen when using the "main" firmware version. After that first set of errors, no errors are ever reported again. Also, I noticed today, that if the car is plugged in, there are never error codes when the command is sent, but when it's unplugged is when the errors are reported. I'm going to leave it set on "edge" firmware for now and see how it does. I'll report back in a week or so regardless of how it's acting to keep everyone updates. So far, this latest firmware seems to resolve the issue.

nismoman240sx, 2023-02-16:
Still not working right. I went to start the car today as it's cold and icy out. Remote CC command would not work. I even restarted the module and it still wouldn't work. I had to go out to the car and just start it to get it going to melt the ice off the car and warm up. Here are the logs;
This is after I tried the command the first time and it didn't work. I logged onto the module, sent the command twice and got this;
I (244267710) webserver: HTTP GET /shell
I (244270330) webserver: HTTP POST /shell
I (244270340) webserver: HandleLogin: 'admin' logged in, sid 9454cd5902fe5468
I (244270360) webserver: HTTP GET /shell
I (244270370) webserver: HTTP GET /menu
I (244285000) webserver: HTTP POST /api/execute
I (244285010) webcommand: HttpCommandStream[0x3f8c6b28]: 3455292 bytes free, executing: climatecontrol on 
OVMS# climatecontrol on 
Climate Control 
on
I (244285010) v-nissanleaf: CommandClimateControl
I (244285010) v-nissanleaf: RemoteCommandHandler
I (244285010) v-nissanleaf: Sending Wakeup Frame
I (244285110) v-nissanleaf: RemoteCommandTimer 24
I (244285110) v-nissanleaf: OLD TCU on EV Bus
I (244285110) v-nissanleaf: Sending Wakeup Frame
I (244285110) v-nissanleaf: Enable Climate Control
I (244285210) v-nissanleaf: RemoteCommandTimer 23
I (244285210) v-nissanleaf: OLD TCU on EV Bus
I (244285210) v-nissanleaf: Sending Wakeup Frame
I (244285210) v-nissanleaf: Enable Climate Control
I (244285310) v-nissanleaf: RemoteCommandTimer 22
I (244285310) v-nissanleaf: OLD TCU on EV Bus
I (244285310) v-nissanleaf: Sending Wakeup Frame
I (244285310) v-nissanleaf: Enable Climate Control
I (244285410) v-nissanleaf: RemoteCommandTimer 21
I (244285410) v-nissanleaf: OLD TCU on EV Bus
I (244285410) v-nissanleaf: Sending Wakeup Frame
I (244285410) v-nissanleaf: Enable Climate Control
I (244285510) v-nissanleaf: RemoteCommandTimer 20
I (244285510) v-nissanleaf: OLD TCU on EV Bus
I (244285510) v-nissanleaf: Sending Wakeup Frame
I (244285510) v-nissanleaf: Enable Climate Control
I (244285610) v-nissanleaf: RemoteCommandTimer 19
I (244285610) v-nissanleaf: OLD TCU on EV Bus
I (244285610) v-nissanleaf: Sending Wakeup Frame
I (244285610) v-nissanleaf: Enable Climate Control
I (244285710) v-nissanleaf: RemoteCommandTimer 18
I (244285710) v-nissanleaf: OLD TCU on EV Bus
I (244285710) v-nissanleaf: Sending Wakeup Frame
I (244285710) v-nissanleaf: Enable Climate Control
I (244285810) v-nissanleaf: RemoteCommandTimer 17
I (244285810) v-nissanleaf: OLD TCU on EV Bus
I (244285810) v-nissanleaf: Sending Wakeup Frame
I (244285810) v-nissanleaf: Enable Climate Control
I (244285910) v-nissanleaf: RemoteCommandTimer 16
I (244285910) v-nissanleaf: OLD TCU on EV Bus
I (244285910) v-nissanleaf: Sending Wakeup Frame
I (244285910) v-nissanleaf: Enable Climate Control
I (244286010) v-nissanleaf: RemoteCommandTimer 15
I (244286010) v-nissanleaf: OLD TCU on EV Bus
I (244286010) v-nissanleaf: Sending Wakeup Frame
I (244286010) v-nissanleaf: Enable Climate Control
I (244286110) v-nissanleaf: RemoteCommandTimer 14
I (244286110) v-nissanleaf: OLD TCU on EV Bus
I (244286110) v-nissanleaf: Sending Wakeup Frame
I (244286110) v-nissanleaf: Enable Climate Control
I (244286210) v-nissanleaf: RemoteCommandTimer 13
I (244286210) v-nissanleaf: OLD TCU on EV Bus
I (244286210) v-nissanleaf: Sending Wakeup Frame
I (244286210) v-nissanleaf: Enable Climate Control
I (244286310) v-nissanleaf: RemoteCommandTimer 12
I (244286310) v-nissanleaf: OLD TCU on EV Bus
I (244286310) v-nissanleaf: Sending Wakeup Frame
I (244286310) v-nissanleaf: Enable Climate Control
I (244286410) v-nissanleaf: RemoteCommandTimer 11
I (244286410) v-nissanleaf: OLD TCU on EV Bus
I (244286410) v-nissanleaf: Sending Wakeup Frame
[... 270 lines cut ...]
I (231229) v-nissanleaf: Sending Wakeup Frame
I (231229) v-nissanleaf: Enable Climate Control
I (231329) v-nissanleaf: RemoteCommandTimer 2
I (231329) v-nissanleaf: OLD TCU on EV Bus
I (231329) v-nissanleaf: Sending Wakeup Frame
I (231329) v-nissanleaf: Enable Climate Control
I (231429) v-nissanleaf: RemoteCommandTimer 1
I (231429) v-nissanleaf: OLD TCU on EV Bus
I (231429) v-nissanleaf: Sending Wakeup Frame
I (231429) v-nissanleaf: Enable Climate Control
I (231529) v-nissanleaf: RemoteCommandTimer 0
I (232329) v-nissanleaf: CcDisableTimer
I (232329) v-nissanleaf: OLD TCU on EV Bus
I (232329) v-nissanleaf: Sending Wakeup Frame
I (232329) v-nissanleaf: Auto Disable Climate Control
E (242269) esp32can: can1 stuck bus-off error state (errflags=0x00040000) detected - resetting bus
I (272269) ovms-server-v2: Send MP-0 S86.0,M,0,0.00,stopped,standard,57,34,0,0,0,2,0,21,0,0,0,0,51.36,0,0,0,0,-1,0,0,0,0,0,0,0,0.00,387.00,82,0.00,100.00,0.00,0.0
I (272279) ovms-server-v2: Send MP-0 D0,8,4,2.22222,2.22222,0,0,278649,0,79237,-5,0,0,0,13.23,0,13.28,16,0,0,-9.44444
I (272279) ovms-server-v2: Send MP-0 L39.028305,-95.740013,0,314.3,1,1,0,0,0,0,0.669,0.137,0,100,AA,11,1.0,0.0,100
I (302279) housekeeping: 2023-02-16 08:14:55 CST (RAM: 8b=69100-69724 32b=12352 SPI=3382736-3402632)
I (333269) ovms-server-v2: Send MP-0 S86.0,M,0,0.00,stopped,standard,57,34,0,0,0,2,0,21,0,0,0,0,51.36,0,0,0,0,-1,0,0,0,0,0,0,0,0.00,387.00,82,0.00,100.00,0.00,0.0
I (333279) ovms-server-v2: Send MP-0 D0,8,4,2.22222,2.22222,0,0,278649,0,79298,-5,0,0,0,13.21,0,13.28,16,0,0,-9.44444
I (333279) ovms-server-v2: Send MP-0 L39.028301,-95.740013,0,314.3,1,1,0,0,0,0,0.669,0.137,0,100,AA,13,0.8,0.0,100
I (333279) ovms-server-v2: Send MP-0 F3.3.003-297-gae3bc7fd/ota_0/edge (build idf v3.3.4-849-g6e214dc33 Feb 5 2023 00:05:47),,19,0,NL,BoatGordHizy,-1,-1,OVMS WIFI BLE BT cores=2 rev=ESP32/3; MODEM SIM7600
"

nismoman240sx, 2023-02-17:
As of this morning the OVMS unit has stopped all communication with the app again and I can't log into it via wifi. My wife has the car right now, but when she get back I'll do another hard reset. Is it possible that I have a hardware issue?

nismoman240sx, 2023-02-20:
Doing another factory reset today to get it working again. I pulled this off of the SD card I had installed. It was inside a file called "can.crtd"
1675986139.852675 CXX OVMS CRTD
1675986139.852675 CVR 3.0
1675986139.854104 1CXX Info Type:vfs Format:crtd(discard) Vehicle:NL Path:/sd/can.crtd
1675986211.135301 1CER TX_Fail T11 679 00
1675986211.138845 1CER TX_Fail T11 679 00
1675986211.146034 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986211.253116 1CER TX_Fail T11 679 00
1675986211.253170 1CER TX_Fail T11 679 00
1675986211.253206 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986211.253649 1CER TX_Fail T11 56E 4e
1675986211.352684 1CER TX_Fail T11 679 00
1675986211.352724 1CER TX_Fail T11 679 00
1675986211.352758 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986211.353096 1CER TX_Fail T11 56E 4e
1675986211.453030 1CER TX_Fail T11 679 00
1675986211.453072 1CER TX_Fail T11 679 00
1675986211.453105 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986211.453507 1CER TX_Fail T11 56E 4e
1675986211.552906 1CER TX_Fail T11 679 00
1675986211.552952 1CER TX_Fail T11 679 00
1675986211.552989 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986211.554042 1CER TX_Fail T11 56E 4e
1675986211.652895 1CER TX_Fail T11 679 00
1675986211.652942 1CER TX_Fail T11 679 00
1675986211.652980 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986211.653384 1CER TX_Fail T11 56E 4e
1675986211.753208 1CER TX_Fail T11 679 00
1675986211.753266 1CER TX_Fail T11 679 00
1675986211.753303 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986211.753716 1CER TX_Fail T11 56E 4e
1675986211.853099 1CER TX_Fail T11 679 00
1675986211.853146 1CER TX_Fail T11 679 00
1675986211.853183 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986211.853846 1CER TX_Fail T11 56E 4e
1675986211.954324 1CER TX_Fail T11 679 00
1675986211.954414 1CER TX_Fail T11 679 00
1675986211.954449 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986211.955127 1CER TX_Fail T11 56E 4e
1675986212.053070 1CER TX_Fail T11 679 00
1675986212.053116 1CER TX_Fail T11 679 00
1675986212.053153 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986212.053552 1CER TX_Fail T11 56E 4e
1675986212.152904 1CER TX_Fail T11 679 00
1675986212.152948 1CER TX_Fail T11 679 00
1675986212.152985 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986212.153344 1CER TX_Fail T11 56E 4e
1675986212.253134 1CER TX_Fail T11 679 00
1675986212.253181 1CER TX_Fail T11 679 00
1675986212.253218 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986212.253624 1CER TX_Fail T11 56E 4e
1675986212.352870 1CER TX_Fail T11 679 00
1675986212.352917 1CER TX_Fail T11 679 00
1675986212.352955 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986212.353331 1CER TX_Fail T11 56E 4e
1675986212.452904 1CER TX_Fail T11 679 00
1675986212.452956 1CER TX_Fail T11 679 00
1675986212.452992 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986212.453424 1CER TX_Fail T11 56E 4e
1675986212.553145 1CER TX_Fail T11 679 00
1675986212.553193 1CER TX_Fail T11 679 00
1675986212.553229 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986212.553631 1CER TX_Fail T11 56E 4e
1675986212.652944 1CER TX_Fail T11 679 00
1675986212.652989 1CER TX_Fail T11 679 00
1675986212.653025 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675986212.653381 1CER TX_Fail T11 56E 4e
1675986212.753148 1CER TX_Fail T11 679 00
1675986212.753202 1CER TX_Fail T11 679 00
1675986212.753238 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
[... 638 lines cut ...]
1675989433.320702 1CER TX_Fail T11 679 00
1675989433.320738 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675989433.321146 1CER TX_Fail T11 56E 4e
1675989433.420157 1CER TX_Fail T11 679 00
1675989433.420197 1CER TX_Fail T11 679 00
1675989433.420230 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675989433.420589 1CER TX_Fail T11 56E 4e
1675989433.520547 1CER TX_Fail T11 679 00
1675989433.520594 1CER TX_Fail T11 679 00
1675989433.520631 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675989433.520998 1CER TX_Fail T11 56E 4e
1675989433.620570 1CER TX_Fail T11 679 00
1675989433.620615 1CER TX_Fail T11 679 00
1675989433.620651 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675989433.621014 1CER TX_Fail T11 56E 4e
1675989433.721150 1CER TX_Fail T11 679 00
1675989433.721200 1CER TX_Fail T11 679 00
1675989433.721236 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675989433.721617 1CER TX_Fail T11 56E 4e
1675989434.621314 1CER TX_Fail T11 679 00
1675989434.621364 1CER TX_Fail T11 679 00
1675989434.621400 1CER TX_Fail T11 5C0 00 13 4a 3f cc 54 fb 3f
1675989434.621774 1CER TX_Fail T11 56E 46

Whatever is wong seems to be with the write process to the can. Assuming I can get it working again with a factory reset, I'm not going to send the ClimateControl command and see if the issue comes back or not.

nismoman240sx, 2023-02-20:
I got the OVMS module working again. I didn't have to factory reset, a power cycle was able to get it communicating again. I also got a micro SD card and got it installed and set up so that it's giving verbose logs on everything now. I also altered some other settings. I noticed that it wasn't staying connected to wifi because I had the connections limits set too tightly. I loosened them up to see if that will keep it from switching back and forth between wifi and cellular. Somehow my 12v offset was wrong too, so I got that fixed as well. I also updated to the update that went out today. I don't think any of the updates have anything to do with the Leaf, but I figured it couldn't hurt.  I went ahead and sent the climate control command and I didn't see any errors in the logs this time, so we'll see if I've somehow fixed it. If it happens again, I should have very complete logs now instead of what I was just able to pull from the shell monitor from my phone.

dexterbg, 2023-02-25:
You can trace back the changes made to the CAN interface and Leaf adapter here:
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/commits/master/vehicle/OVMS.V3/components/esp32can/src
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/commits/master/vehicle/OVMS.V3/components/vehicle_nissanleaf/src

I don't see any changes there that could explain your issue, but maybe your new logs can will show something helpful.

Btw, you can attach files instead of copying them into the text.

biohead, 2023-03-09:
[leaflog.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/10936285/leaflog.txt)

I'm also still seeing the issue. Today I was able to get out and spend a bit of time with the car... add an SD card to the OVMS module, and then also give the leaf a proper power cycle just in case that helped (it didn't).
This is a 2015 model... not sure if that makes any difference.

I have noticed recently that when the climate command does work, the it stays running (doesn't timeout after 15 mins when on battery - it used to). I always thought this 15 minute timer was done by the car, not OVMS... so not quite sure why it's being tricky. Hopefully when it happens again the logs will show it.

nismoman240sx, 2023-03-09:
I'm still trying to catch it on the logs. I thought I had it logging correctly, but I screwed up a setting and ended up with close to a hundred files, so finding the fault in the logs was too much work. I deleted everything, got the settings right, and started over again, and now it's working fine. I'm sure it'll mess up again, but when it does, it'll be easier to find it in the logs.

nismoman240sx, 2023-03-10:
Another failure catching it in the logs. I can't see any reason why, but the OVMS stopped logging about a day before I got the failed "climatecontrol on" command. I had the file size set pretty high, so I cycled the power and changed the log file size to about half of what I had it. I'm keeping my fingers crossed that I'll be able to catch it this time.

biohead, 2023-03-11:
What options are you using for logging? I've put an 8GB uSD card in so should have plenty of space.

I've currently got everything set to verbose and a 1mb file size. It's working, but catching a lot of irrelevant things so trying to narrow it down to the more useful items.

nismoman240sx, 2023-03-13:
The logging has failed again. It stopped logging on 3/11 and I tried to send the CC command this morning, 3/13. 

I'm using a 4GB uSD card. I initially had all of the settings set to default and the logs set to verbose. It created a BUNCH of log files so it made digging through all of them to find the faults a PITA. I increased the file size to 100MB from the default 1MB. It got to around 90MB and then stopped logging. I dropped it down to 50MB and it failed again. I've just reset everything back to factory settings again. It was at least working then.

Bevanj86, 2023-03-16:
Hi guys, 
I've just run into exactly this same issue with the remote climate control with a JDM 2015 e-NV200. Configured as a 2015 Leaf, with CAN-write enabled. TCU is definitely disconnected.

Also getting the "OLD TCU on EV Bus" messages. Climate control fires up, but no indication of running climate control on the Android OVMS app. I also note the "Auto Disable Climate Control" at the end of the attempts. I get similar results when trying to use the app based central locking. 

I'll get an SD card and also attempt to log. I'm new to OVMS, so am in for a steep learning curve.

mjkapkan, 2023-04-03:
OVMS initiated climate control time is the same as in the official Nissan app: 15 minutes when the car is unplugged and 2 hours if the car is plugged in.

trinityhades, 2023-05-14:
Just adding to this thread that I am getting the same issue on my 2013 Nissan Leaf with the TCU unplugged. I get the same output in the shell as everyone else in this thread. I bought this OVMS specifically for Climate Control support, so hoping this can be fixed.

warthog9, 2023-07-17:
Don't know that I have a lot to add but I'll relate what I'm seeing on this particular issue.
Car: Nissan Leaf, 2015, SV
Original TCU: completely disconnected, not a wire plugged in
OVMS Firmware: 
```
Running partition: ota_1
Boot partition: ota_1
Factory image: 3.3.001-33-g58d01654
OTA_O image: 3.3.002
OTA_1 image: 3.3.003
```

So I can, seemingly 100% reliably, get the climate to come on if I issue the command shortly after physically pulling it out and plugging the OVMS system back in.  This is obviously not ideal.  After remembering to stop by and see if I was the only one seeing this (and tripping over this issue thankfully), I tried from the app to start the climate control and was seeing the same thing that's reported above right down to the logs, so I won't repeat that.  On a whim, and I feel like I've tried this before successfully, I rebooted the module from the web interface (since I'm home and that's not hard), waited a bit and for it to reconnect to my server and then issued the `A/C ON` and it came right up.  Quickly checking the car it was clearly running.

I haven't tried a newer firmware than the stock OTA 3.3.003 yet, but a hard pull and/or a module reboot both seem to get it back to a working state.  It's unclear if that just causes a different signal on boot to wake up the appropriate CAN bus as it reconnects, or if this is something from a software perspective that can settle itself out.

I will note that when I looked at `can can1 status` there were no error flags marked as well.

fermulator, 2023-07-23:
(following subscribed)

darn was about to purchase but noticed users reporting this bug on forum

if fixed will purchase 4G kit for Leaf

trinityhades, 2023-12-13:
Adding that I have similar issue with my 2013 Nissan Leaf. Except I have never gotten it to work regardless of what firmware I’m on or if I’ve just rebooted etc… 
My shell looks the same as most here, except mine says “NEW TCU on EV Bus”

`OVMS# climatecontrol on
Climate Control 
on
I (702881) webcommand: HttpCommandStream[0x3f8b86c0]: 3473336 bytes free, executing: climatecontrol on
I (702891) v-nissanleaf: CommandClimateControl
I (702891) v-nissanleaf: RemoteCommandHandler
I (702891) v-nissanleaf: Sending Wakeup Frame
E (702891) can: can1: intr=1 rxpkt=0 txpkt=0 errflags=0x8000d9 rxerr=0 txerr=8 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=0 wdgreset=0 errreset=0
E (702891) can: can1: intr=4 rxpkt=0 txpkt=0 errflags=0x8000d9 rxerr=0 txerr=23 rxinval=0 rxovr=0 txovr=0 txdelay=2 txfail=0 wdgreset=0 errreset=0
I (702991) v-nissanleaf: RemoteCommandTimer 24
I (702991) v-nissanleaf: New TCU on CAR Bus
I (702991) v-nissanleaf: Sending Wakeup Frame
I (702991) v-nissanleaf: Enable Climate Control
I (703091) v-nissanleaf: RemoteCommandTimer 23
I (703091) v-nissanleaf: New TCU on CAR Bus
I (703091) v-nissanleaf: Sending Wakeup Frame
I (703091) v-nissanleaf: Enable Climate Control
I (703191) v-nissanleaf: RemoteCommandTimer 22
I (703191) v-nissanleaf: New TCU on CAR Bus
I (703191) v-nissanleaf: Sending Wakeup Frame
I (703191) v-nissanleaf: Enable Climate Control
I (703291) v-nissanleaf: RemoteCommandTimer 21
I (703291) v-nissanleaf: New TCU on CAR Bus
I (703291) v-nissanleaf: Sending Wakeup Frame
I (703291) v-nissanleaf: Enable Climate Control
I (703391) v-nissanleaf: RemoteCommandTimer 20
I (703391) v-nissanleaf: New TCU on CAR Bus
I (703391) v-nissanleaf: Sending Wakeup Frame
I (703391) v-nissanleaf: Enable Climate Control
I (703491) v-nissanleaf: RemoteCommandTimer 19
I (703491) v-nissanleaf: New TCU on CAR Bus
I (703491) v-nissanleaf: Sending Wakeup Frame
I (703491) v-nissanleaf: Enable Climate Control
I (703591) v-nissanleaf: RemoteCommandTimer 18
I (703591) v-nissanleaf: New TCU on CAR Bus
I (703591) v-nissanleaf: Sending Wakeup Frame
I (703591) v-nissanleaf: Enable Climate Control
I (703691) v-nissanleaf: RemoteCommandTimer 17
I (703691) v-nissanleaf: New TCU on CAR Bus
I (703691) v-nissanleaf: Sending Wakeup Frame
I (703691) v-nissanleaf: Enable Climate Control
I (703791) v-nissanleaf: RemoteCommandTimer 16
I (703791) v-nissanleaf: New TCU on CAR Bus
I (703791) v-nissanleaf: Sending Wakeup Frame
I (703791) v-nissanleaf: Enable Climate Control
I (703891) v-nissanleaf: RemoteCommandTimer 15
I (703891) v-nissanleaf: New TCU on CAR Bus
I (703891) v-nissanleaf: Sending Wakeup Frame
I (703891) v-nissanleaf: Enable Climate Control
I (703991) v-nissanleaf: RemoteCommandTimer 14
I (703991) v-nissanleaf: New TCU on CAR Bus
I (703991) v-nissanleaf: Sending Wakeup Frame
I (703991) v-nissanleaf: Enable Climate Control
I (704091) v-nissanleaf: RemoteCommandTimer 13
I (704091) v-nissanleaf: New TCU on CAR Bus
I (704091) v-nissanleaf: Sending Wakeup Frame
I (704091) v-nissanleaf: Enable Climate Control
I (704191) v-nissanleaf: RemoteCommandTimer 12
I (704191) v-nissanleaf: New TCU on CAR Bus
I (704191) v-nissanleaf: Enable Climate Control
I (704291) v-nissanleaf: RemoteCommandTimer 11
I (704291) v-nissanleaf: New TCU on CAR Bus
I (704291) v-nissanleaf: Sending Wakeup Frame
I (704291) v-nissanleaf: Enable Climate Control
I (704391) v-nissanleaf: RemoteCommandTimer 10
I (704391) v-nissanleaf: New TCU on CAR Bus
I (704391) v-nissanleaf: Sending Wakeup Frame
I (704391) v-nissanleaf: Enable Climate Control
I (704491) v-nissanleaf: RemoteCommandTimer 9
I (704491) v-nissanleaf: New TCU on CAR Bus
I (704491) v-nissanleaf: Sending Wakeup Frame
I (704491) v-nissanleaf: Enable Climate Control
I (704591) v-nissanleaf: RemoteCommandTimer 8
I (704591) v-nissanleaf: New TCU on CAR Bus
I (704591) v-nissanleaf: Sending Wakeup Frame
I (704591) v-nissanleaf: Enable Climate Control
I (704691) v-nissanleaf: RemoteCommandTimer 7
I (704691) v-nissanleaf: New TCU on CAR Bus
I (704691) v-nissanleaf: Sending Wakeup Frame
I (704691) v-nissanleaf: Enable Climate Control
I (704791) v-nissanleaf: RemoteCommandTimer 6
I (704791) v-nissanleaf: New TCU on CAR Bus
I (704791) v-nissanleaf: Sending Wakeup Frame
I (704791) v-nissanleaf: Enable Climate Control
I (704891) v-nissanleaf: RemoteCommandTimer 5
I (704891) v-nissanleaf: New TCU on CAR Bus
I (704891) v-nissanleaf: Sending Wakeup Frame
I (704891) v-nissanleaf: Enable Climate Control
I (704991) v-nissanleaf: RemoteCommandTimer 4
I (704991) v-nissanleaf: New TCU on CAR Bus
I (704991) v-nissanleaf: Sending Wakeup Frame
I (704991) v-nissanleaf: Enable Climate Control
I (705091) v-nissanleaf: RemoteCommandTimer 3
I (705091) v-nissanleaf: New TCU on CAR Bus
I (705091) v-nissanleaf: Sending Wakeup Frame
I (705091) v-nissanleaf: Enable Climate Control
I (705191) v-nissanleaf: RemoteCommandTimer 2
I (705191) v-nissanleaf: New TCU on CAR Bus
I (705191) v-nissanleaf: Sending Wakeup Frame
I (705191) v-nissanleaf: Enable Climate Control
I (705291) v-nissanleaf: RemoteCommandTimer 1
I (705291) v-nissanleaf: New TCU on CAR Bus
I (705291) v-nissanleaf: Sending Wakeup Frame
I (705291) v-nissanleaf: Enable Climate Control
I (705391) v-nissanleaf: RemoteCommandTimer 0
I (706191) v-nissanleaf: CcDisableTimer
I (706191) v-nissanleaf: New TCU on CAR Bus
I (706191) v-nissanleaf: Sending Wakeup Frame
I (706191) v-nissanleaf: Auto Disable Climate Control
E (715391) can: can2: intr=614780 rxpkt=616247 txpkt=75 errflags=0x22401c02 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0`

biohead, 2024-02-13:
Just adding a bit more to this... I'm still running Main... and the issues eventually returned as the weather got colder. It's been a frustrating winter with this.

Another issue - which may or may not be part of this issue - is that the CC once enabled will sometime stay on **permanently**. 
Even after the 15 min timer is up - even after the car is taken for a drive and turned off again, the CC kicks in after a few seconds of shutdown. The last time this happened the OVMS module was unresponsive and I was away - it brought the main HV battery down to 29% before we realised and could intervene.

I've bought a new SD card to start logging again - I had some very strange errors with my old card.

nismoman240sx, 2024-02-14:
> Just adding a bit more to this... I'm still running Main... and the issues eventually returned as the weather got colder. It's been a frustrating winter with this.
> 
> Another issue - which may or may not be part of this issue - is that the CC once enabled will sometime stay on **permanently**. Even after the 15 min timer is up - even after the car is taken for a drive and turned off again, the CC kicks in after a few seconds of shutdown. The last time this happened the OVMS module was unresponsive and I was away - it brought the main HV battery down to 29% before we realised and could intervene.
> 
> I've bought a new SD card to start logging again - I had some very strange errors with my old card. I've also reverted to my factory partition 3.3.001 and will see if I start seeing the CAN errors again.

I've had all of the exact same issues. I've stopped using the OVMS completely. I just need to remove it from the car at this point. After it ran the battery completely dead because it never turned off, I gave up on it ever working correctly. It's been over a year since I started posting about this and nothing has ever changed. I'd love for it to work correctly, as it wasn't exactly cheap, but I've written it off at this point that it will ever work correctly again.

biohead, 2024-02-14:
>  After it ran the battery completely dead because it never turned off, I gave up on it ever working correctly

So, the CC being stuck on is what's prompted me to look further at this, knowing a bit about how the Leaf works.
I was under the impression that this 15 min timer (or 2 hours when plugged in) was _entirely_ driven by the car. OVMS would send the command, then the car would turn it off. OVMS was simulating the remote on command only, and the car takes care of the rest. If anyone could confirm that would be great.
I have a suspicion that it is not something peculiar about the 2015 model year, but something that OVMS is sending is causing parts of the CAN bus to lock out, preventing the OFF command from reaching the right parts of the car after the timer has expired. Once I start seeing the errors again I'll look at the logs closer and try to decode the CAN messages - I'm keeping an eye on them to make sure they keep working in the meantime.

I'm running 3.3.001 now, and no errors so far. I also recall this was the case after flashing back to Main and it was a few days later before it started occurring. I don't know if it's an issue with the CAN part of my board or the firmware - but it was only flashing the firmware which would give me a few days grace. A power reset either soft or hard didn't have the same length of effect.

nismoman240sx, 2024-02-14:
I had the same experience. hard and soft resets wouldn't get it to work. Only a firmware update, and then it would only work for a short amount of time before not working anymore. I'm also sure from when I was still trying to get it to work, that it's nothing to do with the phone app, as I could recreate the same issues sending the CC command from the web interface. 

I am not a computer programmer by any means, but I have a firm enough grasp of how computers work that I think that, like you said, the OVMS is doing something that's causing the cars CAN bus to stop responding to certain messages. That being said, I'm not sure why an OVMS firmware update gets it to work again for a while.

Though now that I'm thinking more about it, I feel like maybe that stopped working eventually too. I could be wrong on that. I gave up on it close to a year ago.

biohead, 2024-02-24:
After a few weeks of it working OK, the problem reoccurred this morning 3.3.001.
The only difference this morning over the previous few weeks were - 1) It was cold and 2) the car was plugged in.
It seems that the Leaf is always plugged in when the issues start - without it being plugged in it has worked fine since reflash.

I attempted to start CC with the car plugged in (but charger NOT active at that time of day, approx 08:05). This didn't work, so I rebooted the unit and CC worked upon reboot. The module responds in the OVMS app, but nothing from the car.
I decided to stop CC before the car departed - this also did not work, another reboot followed the CC disable command did the trick.
Then the car departed around 08:27 (from memory) so I've stopped the log just after that.

I've managed to get the logfiles, it looks like there was a time drift issue before - you'll see the unit reboot then the time goes back about 20 mins. The 2nd half of the log is the correct time.

[Leaf.log](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/14392886/Leaf.log)

biohead, 2024-02-27:
As soon as sending the command this morning, the MCP2515 errors started. I wonder if it's hardware related, but seems unlikely at this stage?
The car DID start climate control, but did not turn it off (the module remained responsive, just needed to issue a remote reset).

```2024-02-27 07:26:41.590 GMT I (89476290) ovms-server-v2: Send MP-0 S53.0,M,0,0.00,stopped,,31,18,0,0,0,79,13,21,-1,0,0,0,48.85,0,0,0,100,-1,0,0,0,0,0,34,0,0.00,378.50,74,0.00,100.00,10.50,-7.5
2024-02-27 07:26:41.600 GMT I (89476300) ovms-server-v2: Send MP-0 D64,8,4,5.55556,6.11111,4,0,476940,0,47830,1.5,0,1,1,14.58,0,12.44,144,0,0,0
2024-02-27 07:26:41.600 GMT I (89476300) ovms-server-v2: Send MP-0 L53.587299,-1.324867,0,58.6,1,1,0.0,0,0,3.407,0.679,0.044,0,100,AA,16,0.7,0.0
2024-02-27 07:26:41.610 GMT I (89476310) ovms-server-v2: Send MP-0 F3.3.001-33-g58d01654/factory/main (build idf v3.3.4-848-g1ff5e24b1 Dec 27 2021 00:03:08),SJNFAAZE0U6034500,21,1,NL,Dr-Pepper,-1,-1
2024-02-27 07:26:46.270 GMT W (89480970) mcp2515: can2 EFLG: RX_Err_Passv RX_Err_Warn EWARN 
2024-02-27 07:26:46.270 GMT E (89480970) can: can2: intr=1323106 rxpkt=1330753 txpkt=12 errflags=0xa00b10a0 rxerr=129 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
2024-02-27 07:26:46.270 GMT W (89480970) mcp2515: can2 EFLG: RX_Err_Warn EWARN 
2024-02-27 07:26:46.350 GMT W (89481050) mcp2515: can2 EFLG: RX_Err_Passv RX_Err_Warn EWARN ```

nismoman240sx, 2024-02-27:
Is there a way to emulate a command coming from Nissan to the TCU instead of trying to emulate that command onto the bus? It seems like the commands that are being put into the bus aren't correct and it's causing the CC system to start ignoring them. 

I removed the OVMS this last weekend. I'll hang on to it for a bit to see if the developers can find a solution, but it doesn't look like this is even on their radar.

glynhudson, 2024-02-27:
It's bizarre that so many reports of this issue, however I've never once experienced any issues with my UK built 2014 AZE0. I use CC almost daily in winter. I'm running main 16th Jan 2022 FW. I wonder if there is some subtle variation between different countries? 

Assuming TCU is unplugged?

biohead, 2024-02-28:
Yes, TCU is unplugged (big connector, not the little one). UK based, 2015 UK built AZE0-1. The logs are definitely showing something is not happy on the car2 can. I really don't know if it's hardware - is there a way to loan a module to see if it reoccurs? I updated back to main last night - as the issues were clearly evident on 3.3.001 and 3.3.002 as well.
The only thing I've not done is a full power cycle of the car (drop the 12v battery for a while) - that's my next move.

Nissan have announced today the sunset of the original remote services from 1st April 2024 (what a date to pick!) so hopefully a few more OVMS modules will join the club (but not have these issues!).

nismoman240sx, 2024-02-28:
My TCU was completely unplugged. USA 2015. We lost our 3G network needed for communication about 2 years ago, which is why I got the OVMS system in the first place.  

My OVMS is now removed from the car. I would love to send it to someone who is NOT having problems to see if the issue occurs on their vehicle or not. If the issue doesn't occur on another vehicle, then I know it's something specific to my car. If it does occur, then it's likely I have a bad OVMS module. My hope is that it's the latter, as I feel like that's an easier to resolve issue.

Meatballs1, 2024-03-02:
I've just installed OVMS in my 2014 Tekna, I unplugged the big white connector for the TCU and cant get climate control working with either the factory (3.3.001-33) or OTA images (3.3.003).

When I try and turn on climate control via OVMS I hear audible clunk 3 times (relay?) and then nothing.

Weirdly I can still turn on climate control via the app still (I thought removing the white connector would prevent it). Pretty sure I've unplugged the right module (under/right of drivers side steering wheel on RHS in the UK).

glynhudson, 2024-03-06:
Can you take an updated video for Teka models? The current video was from an Acenta model

Meatballs1, 2024-03-06:
Annoyingly I have video of the camera connector but not the TCU.

Will have a look at the weekend if I can brave the dirt in the wife's foot well again 😱

[image]

biohead, 2024-03-09:
So, it's been about a week and I'm still hesitant to post this, but...

After a full module reset, then setup from scratch (no backup) - enabling _only_ the bare minimum (WiFi, LTE, Server V2, Leaf) - I've not yet had an issue.  Previously I absolutely would have seen the issue at some point over the course of a week, so I'm cautiously optimistic. Running main, not Edge.

I've not calibrated the 12v, nor the cabin temp, nor setup Server v3, nor any logging - but I've tried to use remote CC as much as possible and so far so good. I'm also not logging, so no real means of keeping an eye for Can2 errors.

Now, it's not been _super_ cold just yet, but I'm still doubtful it was a weather-related issue. It was just that's when I've relied upon it the most - so most frustrated with it.

I have attached my OVMS backup zip, excluding SSH files and redacting passwords, just incase it was config related that can be traced - [Backup_redacted.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/14546064/Backup_redacted.zip)

biohead, 2024-03-24:
Another 2 weeks down the line and all remains operational. I've never had this level of success before...
@nismoman240sx - had you ever done a full factory reset and setup only the bare minimum?

I'm hesitant to start adding features back, as I'm just super happy it's working again!

nismoman240sx, 2024-03-25:
I did try that, with the same bare minimum setup you just described. IIRC it worked for a while, but failed eventually.

glynhudson, 2024-04-03:
Has anyone got remote climate working on a ZE1 2018+ 40kWh+ LEAF?

stevebuk1, 2024-05-21:
> I've just installed OVMS in my 2014 Tekna, I unplugged the big white connector for the TCU and cant get climate control working with either the factory (3.3.001-33) or OTA images (3.3.003).
>
> When I [...]

Just to say thanks for this post. I bought an OVMS system a week ago.   I have a 2016 RHD 30kWh Tekna model that exhibits most of the features of this thread - sometimes I can get the ClimateControl invoked and sometimes I can't. Everything else (including lock/unlock) works fine. I also get the 2 or 3 relay clicks when invoking the Climate commands -  every time. 

Following your post I've realized that I've followed the Open Vehicles video for which plug to pull and have indeed pulled the camera unit plug rather than the TCU plug. I proved tonight by just trying to operate the cameras and finding they are blank. Reinserting the plug brings the cameras back on line.

I've yet to actually locate the TCU and try pulling that, but in all my googling I did find what I think is the source C code for the nissan leaf, which includes the following:

```
if (MyConfig.GetParamValueInt("xnl", "modelyear", DEFAULT_MODEL_YEAR) >= 2016)
    {
    ESP_LOGI(TAG, "New TCU on CAR Bus");
    length = 4;
    tcuBus = m_can2;
    }
  else
    {
    ESP_LOGI(TAG, "OLD TCU on EV Bus");
    length = 1;
    tcuBus = m_can1;
    }
  CommandWakeup();
```
So I've concluded the log files don't tell you whether the TCU is actually connected, its just an indication of which Canbus its going to write on. 
Also, given the Open Vehicles documentation states that the TCU was moved from the EV bus to the CAR bus somewhere in 2016, I'd imagine that there will be 2016 Leafs that have the TCU on the EV bus, and from the above one needs to set a model year of 2015, not 2016. 

I'll update this thread with findings and progress.

stevebuk1, 2024-05-21:
> Has anyone got remote climate working on a ZE1 2018+ 40kWh+ LEAF?

You probably want to follow this issue:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/323

stevebuk1, 2024-06-07:
> I'll update this thread with findings and progress.

After locating the right TCU and disconnecting it I find climate control works fine, but my microphone for hands free calling no longer worked. 

I followed the advice in this thread and reinstalled the TCU but disconnected Pins 6 and 7, which is the Canbus connection:
https://www.openvehicles.com/node/3809

Rather than depin the connector I chopped the connector pins within the TCU module, as shown here. (The pinout of the connector is shown on the paper at the very top left of the image)

[image]

Climate control is working fine, shuts itself off after 15 minutes. Yet to check handsfree but can't imagine that will no longer work.
Car is UK Leaf built 2016, 30kWh Tekna model, with all round cameras.

mrt25, 2024-09-21:
hi, 
just wondering if anyone has knowledge of climate control behaviour on a Nissan leaf 2015 Tekna. 

I encountered the same issue as immediately above, i.e. disconnecting the cable on the OVMS manual disabled the rear camera, however by disconnecting the smaller cable (pictured) I can operate CC via OVMS app, albeit only when my outdoor charger is plugged in. 

Of course, ideally I would be able to activate CC via the app anywhere, as used to be the case with the nissan remote app. Can anyone suggest any steps to achieve this?

[image]

Meatballs1, 2024-09-21:
Have you installed and setup the 4G SIM card?

biohead, 2024-09-21:
@mrt25 do you know exactly what model you have? An AZE0-0/1 or AZE0-2?
The behaviour sounds like a -2. but that is pushing the 2015 date.

mrt25, 2024-09-21:
> <openvehicles/Open-Vehicle-Monitoring-System-3/issues/804/2365270492@
> github.com>

Hi

Yes I have the sim running and other functions seem to be working via that
eg the state of charge.

mrt25, 2024-09-21:
> @mrt25 do you know exactly what model you have? An AZE0-0/1 or AZE0-2? The behaviour sounds like a -2. but that is pushing the 2015 date.

Ah, I'm not sure. Can I check somehow? On VIN checker website it came up with 'Year Of Manufacture 2015'

biohead, 2024-09-21:
Is your headunit/satnav "Carwings" branded, or "Nissan Connect" branded?

mrt25, 2024-09-21:
> <openvehicles/Open-Vehicle-Monitoring-System-3/issues/804/2365272592@

It's car wings

[10 more replies]