openvehicles/Open-Vehicle-Monitoring-System-3 #805: Status of locked / unlocked vehicle is reversed

andrejs2, 2023-01-19:
Hi all!
I have Kia Niro EV (MY2023) with OVMS v 3, Android app., `vehicle` is 
`OVMS# vehicle
Vehicle module 'Kia Niro / Hyundai Kona EV' (code KN) loaded and running` 

Problem:
When my car is `unlocked` and if I check the metrics with: `metrics list lock` I get:
`v.e.locked                               yes
v.p.gpslock                              yes
xkn.v.door.lock.front.left               yes
xkn.v.door.lock.front.right              yes
xkn.v.door.lock.rear.left                yes
xkn.v.door.lock.rear.right               yes`

And in the opposite, if car is `locked` and if I check the metrics with `metrics list lock` I get:
``

So just the opposite.
The same goes with android app, if the car is in reallity `unlocked` the app shows green lock - as it would be locked. And also the opposite.

And also the problem 2:
command: `OVMS# unlock <pin>`
the result is: 
`Error: vehicle could not be unlocked`

The same goes for `lock` id the vehicle is unlocked...

Any idea how to resolve this?

Nimdae, 2023-03-23:
2023 Niro is "second generation" and has a new ECU setup. It appears to be similar/same as Ioniq 5. If you've noticed a ton of stuff being broken in the OVMS app, try changing your vehicle to Ioniq 5 and see if you get the same result I have in that all features start working. Including the lock state, charge state, tire information, etc.

I suspect the Kia EV6 will follow the same thing. 2024 Hyundai Kona might also work with this (we'll have to wait for release to find out). If that's the case, the vehicle can probably be updated to be better named/identify that this is what you use for second generation Niro and Kona, as well as EV6.

andrejs2, 2023-03-23:
Thank you @Nimdae for this! Sound reasonable, but I tried to select vehicle type and there is only IoniqFL (besides Kia Niro / Hyundai Kona EV - which is selected now).

How can I manually select Ioniq 5 (HION5)? 
Appreciate for any help...thank you!

Nimdae, 2023-03-23:
The module appears to be I5, not HION5.

I am also using dexter’s server, but running edge. It might be worth checking main to see if I5 is in there before jumping to edge, though.

I have been noticing some small things that are not quite right, also. Ioniq 5 is on E-GMP, which is an 800v system, so the web dashboard shows the voltage and SOC to be low since this is a 400v system. I believe this is only a visual problem as the app seems okay, which is likely using the actual OVMS metrics.

andrejs2, 2023-03-23:
Thank you @Nimdae. With edge I could select Ioniq 5 👍 
Indeed there are some things broken (voltage, can write is unavailable), some things are the same as with old Niro profile type (tyres temps and pressure), but strangely valet mode can be enabled or disabled via app :). But not vehicle lock/unlock.