openvehicles/Open-Vehicle-Monitoring-System-3 #807: Nissan Leaf OVMS unresponsive

nismoman240sx, 2023-01-26:
I made another post about issues with the HVAC not working from the app. The OVMS has stopped communicating via cellular completely and I can't log in via wifi because it's not broadcasting the wifi network anymore. I can get it to work briefly if I unplug the OBD plug and plug it back in, but it doesn't work long enough to get any logs or anything. I'm starting to wonder if I have a hardware issue. I assume I'm going to have to remove the OVMS module and connect it via USB.

glynhudson, 2023-01-27:
Yes, connecting via USB to view the serial output should give you an idea of what the problem is. You could try a factory reset? https://docs.openvehicles.com/en/latest/userguide/factory.html

nismoman240sx, 2023-01-27:
There are a few other people complaining of similar issues on the OVMS forums. It seems like this started with the mid-January OTA update. I'll try to get mine reverted back to the factory firmware image and see how it does.

nismoman240sx, 2023-02-07:
I finally found time to get the module pulled and got a factory reset done. It seems to be working normally now. I'll leave this open for now. If it continues to work for the next week or so, I'll come back and close this.

nismoman240sx, 2023-02-16:
I have another ticket open that's more relevant to the issue. I'm still able to connect after the factory reset, so I'm going to close this one.

closed 2023-02-16 by nismoman240sx