openvehicles/Open-Vehicle-Monitoring-System-3 #814: Script sendlivedata2abrp.js

andrejs2, 2023-02-02:
I have trouble running script in /store/scripts/sendlivedata2abrp.js
I c/p scripts from [docs](https://docs.openvehicles.com/en/latest/plugin/abrp/README.html), added missing info about tokens, car data...
Saved it, reloaded and if I run EvaluateJS I get:
```
ERROR: ReferenceError: identifier 'exports' undefined
    at [anon] (duk_js_var.c:1236) internal
    at eval (eval:308) preventsyield (line 308)
```
My module info:
```
OVMS# module summary
OVMS MODULE SUMMARY

Module
  Version:  3.3.003/ota_0/main (build idf v3.3.4-848-g1ff5e24b1b Sep  3 2022 11:32:12)
  Hardware: OVMS WIFI BLE BT cores=2 rev=ESP32/3; MODEM SIM7600
  12v:      14.6v

MODEM Status
  Model: SIM7600
  Network Registration: RegisteredRoaming
    Provider: ******* Hologram
    Signal: -65 dBm
    Mode: LTE,Online
  State: NetMode
    Ticker: 179
    User Data: 0
  UART:
    FIFO overflows: 0
    Buffer overflows: 0
    Parity errors: 0
    Frame errors: 0
    Driver Buffer overflows: 0
  Mux: Status up
    Open Channels: 4
    Framing Errors: 0
    RX frames: 280
    TX frames: 35
    Last RX frame: 0 sec(s) ago
  PPP: Connected on channel: #2
  GPS: Connected on channel: #1
     Status: enabled
     Time: enabled

Configuration
  [auto]
    dbc: no
    egpio: no
    ext12v: no
    init: yes
    modem: yes
    obd2ecu: 
    scripting: yes
    server.v2: yes
    server.v3: no
    vehicle.type: KN
    wifi.mode: apclient
    wifi.ssid.ap: *****
    wifi.ssid.client: ******
  [can]
  [dbc]
  [egpio]
    monitor.ports: 
  [http.plugin]
  [http.server]
  [locations]
    Home: 46.038348,14.611619,30
  [log]
    file.enable: no
    file.path: /store/log
    level: info
  [modem]
    apn: hologram
    apn.password: 
    apn.user: 
    enable.gps: yes
    enable.gpstime: yes
    enable.net: yes
    enable.sms: yes
    pincode: 
  [module]
    cfgversion: 2020053100
    init: done
  [network]
    dns: 
  [notify]
    log.trip.minlength: 20
    report.trip.enable: yes
    report.trip.minlength: 20
  [obd2ecu]
  [obd2ecu.map]
  [ota]
    http.mru: https://ovms.dexters-web.de/firmware/ota/v3.3/main/ovms3.bin
    server: https://ovms.dexters-web.de/firmware/ota
  [password]
    module: **redacted**
    pin: **redacted**
    server.v2: **redacted**
    server.v3: **redacted**
  [plugin]
  [plugin.disabled]
  [plugin.enabled]
  [plugin.repos]
  [power]
    12v_shutdown_delay: 30
    enabled: yes
    modemoff_delay: 96
    wifioff_delay: 24
  [pushover]
  [sdcard]
  [server.v2]
    port: 
    server: api.openvehicles.com
    tls: yes
    updatetime.connected: 
    updatetime.idle: 
  [server.v3]
    port: ******
    server: *******
    tls: no
    topic.prefix: 
    updatetime.charging: 120
    updatetime.connected: 120
    updatetime.idle: 600
    updatetime.on: 120
    user: *****
  [ssh.info]
    fingerprint: **********
  [ssh.keys]
  [ssh.server]
    key: **redacted**
  [system.adc]
    factor12v: 195.7
  [tpms]
  [usr]
  [vehicle]
    12v.alert: 11
    12v.ref: 
    bms.alerts.enabled: no
    id: 62KHC
    minsoc: 20
    name: Kia Niro EV
    registered.phone: +38641383576
    stream: 10
    timezone: CET-1CEST,M3.5.0,M10.5.0/3
    timezone_region: Europe/*******
    units.distance: K
  [wifi.ap]
    62-KHC: **redacted**
  [wifi.ssid]
    A63574: **redacted**
    Pixel: **redacted**
    TP-Link_2D46: **redacted**
  [xkn]
    canwrite: yes
    cap_act_kwh: 
    consoleKilometers: yes
    leftDrive: yes
    maxrange: 440
    suffrange: 0
    suffsoc: 0

REPORT ENDS
```

markwj, 2023-02-03:
Where did you get this version 2.0.1 from? The standard docs reference a version 1.3.

Anyway, from what I can see it is working fine:

`script eval abrp.onetime()
...
Vehicle Response:
  (2023-02-03 01:44:01.000+01:00) INFO: Sending telemetry to ABRP {"utc":1675385041;"soc":80;"power":0;"speed":0;"lat":46.0384;"lon":14.6116;"is_charging":false;"is_dcfc":false;"is_parked":true;"kwh_charged":0;"soh":100;"heading":0;"elevation":291.4;"ext_temp":0;"batt_temp":5;"voltage":393;"current":0;"odometer":3283;"est_battery_range":282}`

and similarly:

`script eval abrp.info()
...
Vehicle Response:
  Plugin Version:   2.0.1
  State of Charge:  80 %
  ...`

Probably easier to ask this on the www.openvehicles.com support ticket, as GitHub is for firmware issues.

dexterbg, 2023-02-03:
Apparently there is a bug in the documentation for the old ABRP plugin, the file "sendlivedata2abrp.js" should be located in the "/store/scripts/lib/" plugin directory, not in the main scripts directory.

Also the ABRP plugin development has been continued by Jason at Iternio, the latest version and documentation is available from here: https://github.com/iternio/ovms-link

An update request for our plugin directory is WIP here: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/718

andrejs2, 2023-02-08:
Thanks for help!
The linked abrp script works!

closed 2023-02-08 by andrejs2