openvehicles/Open-Vehicle-Monitoring-System-3 #837: Not able to build

kimmilde, 2023-02-23:
There is a build error at ovms_webserver

EMBED build/ovms_tls/baltimore_cybertrust.crt.bin.o
/bin/sh: dos2unix: fant ikke kommando
make[1]: *** [/home/kimmi/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/ovms_webserver/component.mk:53: /home/kimmi/Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/ovms_webserver/assets/charts.js] Error 127
make: *** [C:/OVMS/home/kimmi/esp-idf/make/project.mk:561: component-ovms_webserver-build] Error 2
make: *** Waiting for unfinished jobs....
EMBED build/ovms_tls/isrg_x1.crt.bin.o

dexterbg, 2023-02-25:
You need to install the `dos2unix` command.

llange, 2023-05-13:
Hi @kimmilde, can you confirm that the build is OK now for you after installing `dos2unix` command?
If so, please close this issue. Thanks!

closed 2023-08-20 by kimmilde