openvehicles/Open-Vehicle-Monitoring-System-3 #858: OTA update using https

beimoe, 2023-03-15:
Hi all,
I have a OVMS running in my car and I am having my own update server, where I would like to use https for security reasons.
Otherwise there is the possibly risk for overtaking the cars firmware via man-in-the-middle attack.
Could you add such a feature?
Thanks much in advance
Beimoe

leres, 2023-03-15:
I think the lack of ota https support is due to the age of our mbedtls. Switching your sdkconfig to wolfssl used to allow ota https work with the downside that connecting to the V2 server over ssl would break.

Given the renewed effort by to bring the project up to ESP-IDF v5 I think it's just a matter of time before we'll be able to use ssl for everything, including ota https.

I also do ota updates to my own server. But I run my ovms modules in wifi client-only mode and only do updates when my cars are at home, connected to my home network.

dexterbg, 2023-03-25:
Actually, the reason isn't the TLS library, but the OTA download using our own, very basic, synchronous HTTP client (class `OvmsHttpClient`, module `ovms_http`), which doesn't support HTTPS.

HTTPS support is no issue with mbedTLS when using mongoose, as I did for the Javascript API `HTTP.Request()`. But that induces the necessity for an asynchronous download handling. Mark tried to implement exactly this in the `for-v3.3` branch, last state can be found by checking out that branch, or browsed online here:
- https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/tree/for-v3.3/vehicle/OVMS.V3/components/ovms_netlib/src

This lead to a deadlock issue with the mongoose context, discussed on the developer list in this thread…
- http://lists.openvehicles.com/pipermail/ovmsdev/2021-March/015239.html

We then tried using the synchronous client for the firmware download, which lead to this issue…
- http://lists.openvehicles.com/pipermail/ovmsdev/2021-September/015465.html

…with final state being:

> The code now uses the old ovms_http library, based on normal sockets (not mongoose). This should also workaround the ‘ota flash http’ issue we had with ssh client connections. I would prefer to use mongoose (apart from uniformity, this also gave us support for HTTPS), but it is quite tricky to do this in a blocking way with the ssh client also over mongoose. Would need to re-work to be non-blocking for these commands.

**TL;DR**: the OTA dowload needs to be reworked towards an asynchronous non-blocking mongoose scheme. Possibly some of the duktape `HTTP.Request` code can be reused.

Another option could be to skip mongoose here completely and use the esp-idf `esp_https_ota` API, as shown in the "advanced_https_ota" example.
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_https_ota.html

fowi4hjte, 2024-03-24:
Why have an old wolfssl and wolfssh version been forked that does not include the latest security issues?
https://github.com/openvehicles/wolfssl
https://github.com/openvehicles/wolfssh

There are already some high risk CVE fixed in the main wolfssl branch that stay unfixed in the fork.