openvehicles/Open-Vehicle-Monitoring-System-3 #862: Cellular commands?

dragon2611, 2023-04-03:
Are the modem commands documented anywhere as I didn't see them in the docs.

Had problems with the modem getting stuck PPP no carrier, it seemed to have registered on 3G, after a couple of reboots it registered on LTE with a much stronger signal and managed to connect.

I could see any obvious command to force it to re-scan/re-register, I was also hoping there was a way to force the use of LTE like some devices support.

dexterbg, 2023-04-07:
Modem commands are documented in the respective modem manufacturer model documentation, simply searching for model name + "at commands" will turn these up. We may not include these into the OVMS repository, as they are copyrighted.

closed 2023-04-07 by dexterbg

dragon2611, 2023-04-07:
@dexterbg Found the command to switch modes, Tested with GSM/WCDMA and LTE and it seems It only gets a connection in LTE mode.

With GSM/WCDMA mode it registers to the Network ( UK rwgmobile (EE) data sim ) but fails PPP with NOCARRIER

```
 (855336) cellular: PPP Connection is ready to start
D (856276) cellular: State transition NetStart => NetMode
I (856276) cellular: State: Enter NetMode state
I (856276) gsm-ppp: Initialising...
D (856276) events: Signal(system.modem.netmode)
D (859496) cellular: mux-rx-line #3: +CGREG: 1
D (859496) cellular: mux-rx-line #3: +CEREG: 4
D (861496) cellular: mux-rx-line #3: NO CARRIER
D (861496) cellular: mux-rx-line #3: +PPPD: DISCONNECTED
I (861496) cellular: PPP Connection disconnected
```

I do have one of their Vodafone sims but my OVMS module is a bit of a pain to get to.

I'm tempted to order another one of their EE based sims (They're cheap enough) and test it in another device to see if that can whilst on GPRS/3G to see if it's a carrier issue or an OVMS issue.