openvehicles/Open-Vehicle-Monitoring-System-3 #863: 7600G dropping connection and not reconnecting

dragon2611, 2023-04-09:
Now i've locked the Modem to LTE it will connect to the network (EE/rwgmobile) but it appears if I leave an LTE coverage area the modem will drop but never manages to reconnect.

I may try another carrier's sim to see if it's a problem with the carrier/reseller but in the meantime I was wondering if it's possible to script a powercycle of the modem.

I know the command is cellular setstate PowerOffOn but I'm not sure what would be a good way to trigger it, is there an event for loss of PPP i could use, or is that likely to lead to excessive powercycling?

dexterbg, 2023-04-10:
The log can tell you lots of details about what is happening. On level debug, all events will be logged as well. For standard events see the user guide: https://docs.openvehicles.com/en/latest/userguide/events.html

dragon2611, 2023-04-15:
RWGMobile offer data sims on Both EE and Vodafone MNO's here, I actually have one of each (I do not have any of their multinet sims as those are more expensive)

Just swapped the Sim from the EE based one to the Vodafone based one, Took a little while for the sim to activate but the Modem connected on 3G (Not sure why it seems to prefer 3G over LTE when in auto but whatever)

I will have to see if this also has problems reconnecting when it loses coverage.

I'm not sure if it's a case of OVMS not liking EE or OVMS not liking the rwgmobile APN on EE specifically.