openvehicles/Open-Vehicle-Monitoring-System-3 #879: ESP idf v5 compilation documentation.

priyankguddu, 2023-04-21:
@llange 
Thanks for porting it to ESP IDF V5, I cloned your fork and main fork on Esp idf. Hello world compiles fine, but the idf.py build on OVMS gives error.
Can someone please update their steps 
'`CMake Error: Error required internal CMake variable not set, cmake may not
  be built correctly.

  Missing variable is:

  CMAKE_FIND_LIBRARY_PREFIXES

  CMake Error: Error required internal CMake variable not set, cmake may not
  be built correctly.
`

llange, 2023-04-21:
Hi @priyankguddu ,
thanks for taking the time to test this build !

Just to be sure, I confirm that the branch [experimental-esp-idf-build-workflow](https://github.com/llange/Open-Vehicle-Monitoring-System-3/tree/experimental-esp-idf-build-workflow) is the best starting point.
It has all the latest patches, is updated with the main from time to time, AND is [automatically built](https://github.com/llange/Open-Vehicle-Monitoring-System-3/actions) (by GitHub CI).
This way you can confirm that the build is possible indeed, plus you can use the [automated build config file](https://github.com/llange/Open-Vehicle-Monitoring-System-3/blob/experimental-esp-idf-build-workflow/.github/workflows/build-ovms.yml) as a guideline if your setup is not working properly.

Additionally, some have found that building with [the official docker image](https://hub.docker.com/r/espressif/idf) can remove the complexity of maintaining / tuning / tweaking your local ESP-IDF install. I'd suggest using the `v5.0.1` tag of this image as a starting point.

If you still have issues, please highlight the detailed steps (if Docker: `docker run ...` included ; indicating the ESP-IDF version, and your O/S version), the git branch and version of the OVMS3 you're using, and the output / error message, so that we can help you. The first step would be that we manage to reproduce the error message, so please reproduce it in a clean environment (a VM for example) and share the steps with us.

llange, 2023-04-30:
Hi @priyankguddu,
how does the compilation work for you ? Did you have time to fix it, or setup the docker image ?
Let us know !

mk11047, 2023-05-26:
Hi @llange,
Got the following error when build using normal command line. 
```
CMake Error: Error required internal CMake variable not set, cmake may not
be built correctly.
Missing variable is:
CMAKE_FIND_LIBRARY_PREFIXES
CMake Error: Error required internal CMake variable not set, cmake may not
be built correctly.
```
When tried with docker, gives the following error:
```
CMake Error at /opt/esp/idf/tools/cmake/component.cmake:312 (message):
  Include directory '/project/components/zip/libzip/lib' is not a directory.
Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/component.cmake:486 (__component_add_include_dirs)
  components/zip/CMakeLists.txt:10 (idf_component_register)
```
**esp-idf version: 5.0.1**

Please suggest resolution.

llange, 2023-05-26:
Hello @mk11047 ,
It seems that (at least for the docker try), you did not fully follow the instructions. It looks like your local git repo is not fully initialized, in particular the submodules.
Have you followed the developper guide https://docs.google.com/document/d/1q5M9Lb5jzQhJzPMnkMKwy4Es5YK12ACQejX_NWEixr0/edit#heading=h.iq9ytb4muvpt ? In particular the `git submodule ...` part.

In any case, it's always better to give full (reproducible) context, including your command line so that we can follow exactly the steps you've made.

For the non-docker try, it seems you did not launch the `export` script : https://docs.espressif.com/projects/esp-idf/en/v5.0.2/esp32/get-started/linux-macos-setup.html#step-4-set-up-the-environment-variables

So please start from scratch (in a temporary folder), reproduce your build error, and then copy **all** the steps starting from `git clone ...` so that we can see exactly what happens.

Let us know.

mk11047, 2023-05-29:
Hi @llange,
I tried the compilation from scratch but got the same error, attaching the file for your reference and please let me know.
[test_ovms.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/11591117/test_ovms.txt)

mk11047, 2023-05-29:
Hi @llange
As suggested I followed the steps mentioned in [link](https://docs.espressif.com/projects/esp-idf/en/v5.0.1/esp32/get-started/linux-macos-setup.html#step-4-set-up-the-environment-variables) and observing the same issue. The hello world compilation is fine but getting issue with ovms compilation.

Attached is the log file of my execution of the commands and their received output. Log File: [test_ovms.txt](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/11591179/test_ovms.txt)

llange, 2023-05-29:
Hi @mk11047,

thanks for the attached output log, it helped me see what parts are missing:
- You're using `master` branch of my fork, which is not the best starting point. I recommend starting with branch [`build-master`](https://github.com/llange/Open-Vehicle-Monitoring-System-3/tree/build-master)
- The [developper guide](https://docs.google.com/document/d/1q5M9Lb5jzQhJzPMnkMKwy4Es5YK12ACQejX_NWEixr0/edit#heading=h.iq9ytb4muvpt) is not optional! You really should read it, in particular the git submodule parts as I explained previously.
- Then, there are a few manual patches that are still needed.

Let's try this sequence of operations: (please adapt when I guessed wrong)
```
. /home/yantrr/Videos/esp-idf/export.sh
cd /home/yantrr/Videos/Open-Vehicle-Monitoring-System-3
git checkout build-master
git fetch --all
git pull
git submodule update --init --recursive   # Just do that one time, or when you update with git fetch
git apply --directory="vehicle/OVMS.V3/components/mongoose/mongoose" "vehicle/OVMS.V3/support/mongoose-espv5.patch" # Just do that one time
cd vehicle/OVMS.V3
cp support/sdkconfig.defaults.esp5.0.1 sdkconfig # Just do that one time
idf.py build
```
I believe it should then build.

Please be warned, that this branch is not (yet) ready for consumption, not working as it should, lacking some pieces (in particular TLS / SSL is not working, etc...)

This is really a developer testbed, with all its quirks, and if you feel that you're not ready to deal with all these issues then it's best waiting that we complete the port. (But of course every help is useful !)

Let us know.

priyankguddu, 2024-05-19:
Hi @llange  is there any updated documentation on IDF 5.x compilation here. We have been now stuck to use this as 3.3x all the python 2.7.x dependency are deprecated. Kindly help with the basic compilation and update the dev guide