openvehicles/Open-Vehicle-Monitoring-System-3 #899: Provide ability to select destination OTA when flashing from file

GreyPeter, 2023-05-13:
When I am writing and testing code, I prefer to flash the new code to OTA_1 in case I have a problem after rebooting.
I can then reset back to OTA_0 from the terminal if the module is locked up.
Is it possible to add a destination OTA (Auto/OTA_0/OTA_1) on the "Firmware setup & update" page when flashing from a file.

markwj, 2023-05-13:
This is part of the bigger issue of hitting the current 4MB flash size limit.Perhaps it is time to change from the original design factory+ota0+ota1, to a more modern ota0+ota1 (giving us 6MB for each)?Mark

GreyPeter, 2023-05-13:
Yes, I have noticed that we were gradually creeping to the 4MB file size. Less than 12 months left in my calculation. Probably a good time to get rid of the factory partition if that’s possible. Perhaps new firmware could go to OTA_1 with OTA_0 on a known stable build. Similar to the factory now I guess. As for my original question, would it be possible to have a destination selection for flashing from a file or would you prefer to concentrate on fixing the 4MB flash limit?

markwj, 2023-05-13:
The current approach is:Run from ota 0 or 1.If running from 0, flash to 1 and reboot to that.If running from 1, flash to 0 and reboot to that.You can’t flash to the current running partition (as that is where the code you are current running is stored). We could reboot to factory and then flash to either ota 0 or 1.Regards, Mark

GreyPeter, 2023-05-13:
Yes. Of course you can’t flash the running partition. Silly me.

closed 2023-05-13 by GreyPeter