openvehicles/Open-Vehicle-Monitoring-System-3 #908: ESP-IDF-v5.1next: new error following a possible upgrade of toolchain

llange, 2023-05-23:
The following new error appears on the `release-v5.1` branch of ESP-IDF:

```
In file included from <OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/vehicle_kianiroev.h:32,
                 from <OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/vehicle_kianiroev.cpp:61:
<OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/vehicle_kianiroev.cpp: In member function 'void OvmsVehicleKiaNiroEv::HandleCharging()':
<OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/../../vehicle_kiasoulev/src/kia_common.h:256:137: error: use of deleted function 'std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::basic_string(std::nullptr_t) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>; std::nullptr_t = std::nullptr_t]'
  256 | #define SET_CHARGE_STATE(n,m)           StdMetrics.ms_v_charge_state->SetValue(n); if(m!=NULL) StdMetrics.ms_v_charge_substate->SetValue(m)
<OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/vehicle_kianiroev.cpp:649:25: note: in expansion of macro 'SET_CHARGE_STATE'
  649 |                         SET_CHARGE_STATE("topoff", NULL);
      |                         ^~~~~~~~~~~~~~~~
In file included from <ESP-IDF-TOOLS-ROOT>/release-v5.1/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/string:53,
                 from <OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle/vehicle.h:36,
                 from <OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/../../vehicle_kiasoulev/src/kia_common.h:4:
<ESP-IDF-TOOLS-ROOT>/release-v5.1/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/bits/basic_string.h:734:7: note: declared here
  734 |       basic_string(nullptr_t) = delete;
      |       ^~~~~~~~~~~~
<OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/../../vehicle_kiasoulev/src/kia_common.h:256:137: error: use of deleted function 'std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::basic_string(std::nullptr_t) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>; std::nullptr_t = std::nullptr_t]'
  256 | #define SET_CHARGE_STATE(n,m)           StdMetrics.ms_v_charge_state->SetValue(n); if(m!=NULL) StdMetrics.ms_v_charge_substate->SetValue(m)
<OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/vehicle_kianiroev.cpp:653:25: note: in expansion of macro 'SET_CHARGE_STATE'
  653 |                         SET_CHARGE_STATE("topoff", NULL);
      |                         ^~~~~~~~~~~~~~~~
<ESP-IDF-TOOLS-ROOT>/release-v5.1/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/bits/basic_string.h:734:7: note: declared here
  734 |       basic_string(nullptr_t) = delete;
      |       ^~~~~~~~~~~~
<OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/../../vehicle_kiasoulev/src/kia_common.h:256:137: error: use of deleted function 'std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::basic_string(std::nullptr_t) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>; std::nullptr_t = std::nullptr_t]'
  256 | #define SET_CHARGE_STATE(n,m)           StdMetrics.ms_v_charge_state->SetValue(n); if(m!=NULL) StdMetrics.ms_v_charge_substate->SetValue(m)
<OVMS-ROOT>/vehicle/OVMS.V3/components/vehicle_kianiroev/src/vehicle_kianiroev.cpp:693:25: note: in expansion of macro 'SET_CHARGE_STATE'
  693 |                         SET_CHARGE_STATE("charging",NULL);
      |                         ^~~~~~~~~~~~~~~~
<ESP-IDF-TOOLS-ROOT>/release-v5.1/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/12.2.0/bits/basic_string.h:734:7: note: declared here
  734 |       basic_string(nullptr_t) = delete;
      |       ^~~~~~~~~~~~
[960/1425] Building C object esp-idf/duktape/CMakeFiles/__idf_duktape.dir/src/duktape.c.objninja: build stopped: subcommand failed.
```

May be we will have to rewrite a little the macro construct to appease the compiler.

**_Just noting this here so that we can investigate it - no urgent action needed._**

fowi4hjte, 2024-03-24:
The ESP-IDF 3.3 is now out of support. IDF 4 is still supported but would stop soon. IDF 5 is the main supported version.
I am using aarch64 CPU architecture and ESP-IDF 3.3 does not support aarch64. IDF 5 support aarch64, but with version 5 i cant build this OVMS because of the build errors. Would be great if OVMS could get ported to the latest IDF 5 in some future.

priyankguddu, 2024-05-19:
++
Can someone please help here, IDF v5+ compilation