openvehicles/Open-Vehicle-Monitoring-System-3 #911: Nissan Leaf 2014 - No reading of battery temperature by OVMS after converting from 24 to 40 kWh

szakal333, 2023-06-05:
Hi. After converting the battery from 24 to 40 kWh, OVMS stopped seeing the temperature of the battery. Leaf Spy sees the temperature correctly. Can someone help solve the problem? Thanks!

dalathegreat, 2023-07-12:
@szakal333 Which CAN-bridge are you using? If you are using my variant, update to the latest version!

szakal333, 2023-07-12:
Hi. Thank you very much for your answer. I am using your version with firmware V4.13. Should I update the software?

dalathegreat, 2023-07-12:
@szakal333 yes, latest version here fixes it! https://github.com/dalathegreat/Nissan-LEAF-Battery-Upgrade

szakal333, 2023-07-13:
@dalathegreat Thank you for your help so far. Do I have to disconnect the 12V power supply in the car to update the controller so that the controller is not powered by the car? I am full of admiration for your work, I decided to become your micro patron and buy you the proverbial coffee as a thank you. Thanks!

dalathegreat, 2023-07-13:
@szakal333 12V should be off yes. Best to disconnect the lead acid battery. Thanks for the Patreon support, we can continue the discussion on the Discord server!

@dexterbg this issue can be closed, not related to OVMS

malyjaponec, 2025-04-29:
The car is ZE0 converted to 30kWh battery.  App constantly shows battery temperature 11C if car is parked for a while:

v.b.temp is 11C
but 
xnl.bms.temp.int is 16;16;255;15;15;0C
v.b.c.temp is 15.912;16.116;15.3C

then I switch on AC that wakes up the car and I see this
v.b.temp is -2C
and 
xnl.bms.temp.int is still 16;16;255;15;15;C that corresponds to the temperature in garage now also v.e.temp shows 16,5C
v.b.c.temp is 15.912;16.116;15.3C

So where ovms gets values for v.b.temp (11 or -2C)? I expected it should somehow average the bms temperatures. 
Is this problem of OVMS or can bridge?

I think I am using latest can bridge sw. I was fiddling with some betas that did not deliver results regarding SOH that promised but I think I have v.4.24

dalathegreat, 2025-04-29:
OVMS uses the Muxed 0x5C0 historic data message. This is not optimal, since it will assume wrong LEAF generation on battery upgraded vehicles. The 0x5C0 message has not been fully decoded on all LEAF generations, and they moved around a lot of data in it. So when OVMS is trying to use the ZE0 method to interpret the temperatures, it goes wrong.

This issue can be solved in two ways:

A: Change OVMS to use the polled temperature values from 0x7BB instead of periodically transmitted 0x5C0. This method works the same on all LEAF generations.
B: Code the Battery-Upgrade firmware better, to reformat the new 0x5C0 message into the correct shape for ZE0. Requires extensive R&D

malyjaponec, 2025-05-11:
I found that also dashboard shows always 2 bars that means something between -10 and 0 C. I usually do not look too much to display as the car is mostly driven by my wife. It was winter so I expected it is ok but now it seems it should not be staying on 2 bars with temperatures we have now. It seems that since update to new simpler can bridge software the car lost also availability to display correct temperature. But I remember that with ancient old can bridge sw it worked. I have seen bars up to red when I "tried" to bake the battery in summer long trip. So maybe the coding and R&D would not need to be so much extensive, if there must be some old code what probably makes 0x5C0 messages somehow. It might not be 100% as the temperature coding was not fully discovered but would be working somehow.

This is not priority for me at all. I just want to publish what I found to others know it.

malyjaponec, 2025-05-15:
Please accept my deepest apologies, but today I found that my previous post is not true. Today I have seen 4 temperature bars on dashboard. So the dashboard seems to be working somehow with CAN bridge.

The problem is really only OVSM app and what i wrote about v.b.c.temp and xnl.bms.temp.int is still valid. If the app use different values for "Car" page (that I can request on demand) it would be fine. I can write in terminal "me li v.b.c.temp" to get real values but it is quite inconvenient.