openvehicles/Open-Vehicle-Monitoring-System-3 #917: Unit conversion for temperature standard deviation not correct

kezarjg, 2023-06-17:
The unit conversion of temperature units for standard deviation is not correct. The standard deviation should be converted without the offset factor of the equation.

kezarjg, 2024-09-25:
## Proposed Solution:

In order to properly handle temperature standard deviations without causing errors due to unit conversions, I propose the addition of new units specifically for storing temperature standard deviations in both Celsius and Fahrenheit. 

These new units will be used for metrics that represent the standard deviation of temperatures, ensuring correct scaling during calculations and conversions.

### Files Affected:
- `vehicle/OVMS.V3/main/ovms_metrics.h`

### Proposed Changes:

```cpp
// Add new units for temperature standard deviations
StDevCelcius    = 22,  // Standard deviation in Celsius
StDevFahrenheit = 23,   // Standard deviation in Fahrenheit

dexterbg, 2024-09-29:
Can you please add an example of the error you see?

kezarjg, 2024-09-29:
Yes, of course. I should have included them earlier.

[image]

[image]

The standard deviation in Fahrenheit is incorrect. It should be 0.184995 °F

kezarjg, 2024-09-29:
To be clear, I understand the proposed solution will touch a few more files. I wanted feedback that I was headed down the right path before I propose the changes to the other places.

dexterbg, 2024-09-29:
I see. OK, your proposed solution looks valid to me, but you should not name the units "StDev…", but rather "Diff…" or "Delta…".

That's because this applies to any temperature differences or ranges, not only to standard deviations.