openvehicles/Open-Vehicle-Monitoring-System-3 #933: filesystem corruption

Stoatwblr, 2023-09-10:
This is a semi-regular occurrance (all firmware versions, currently running 

Thanks to the 12V battery going flat, I'm no longer able to make configuration backups

using the GUI, I get this when clicking "save"

Creating config backup '/sd/backup/cfg-230910.zip'...
..add 'ovms_config'
..add 'events'
..add 'scripts'
..add 'obd2ecu'
..add 'dbc'
..add 'plugin'
..add 'tls'
..add 'trustedca'
..add 'usr'
[ 1-2 minute delay ]
Request error, please retry

At that point the OVMS box reboots - judging by uptimes it looks like the error message happens when the browser gets a "connection refused" during the bootup cycle and the crash happens within seconds of "..add 'usr'"

I've dealt with this this twice in the past by doing a cold reinstall, but this is _not_ an ideal way to fix what appears to be simple filesystem corruption

The corruption doesn't occur every time battery voltage drops out, but it happens often enough to be a clear and present bug

Logging is to the SD card and all plugin writes are sent there too. That part is testing ok and fsck finds no issues

dexterbg, 2023-09-10:
The fat filesystem corruption bug is more than 5 years old and still unsolved. It's currently tracked here:
- https://github.com/espressif/esp-idf/issues/8197

Our issue thread on this is https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/151

I meanwhile expect this to turn out as another ESP32 hardware bug.

Stoatwblr, 2023-09-16:
I rechecked everything and the microsd card (less than 6 months old 128GB Samsung galaxy EVO) has toggled read-only

Replacing it solved the crashing issue

closed 2023-09-19 by dexterbg