openvehicles/Open-Vehicle-Monitoring-System-3 #955: Toyota Hiace EV and Temperatures

ECHEngineering, 2023-12-05:
I have a project for 10 Toyota Hiace EV vans.  As part of a research project i'd like to track and monitor them with OVMS but note they are not listed in the compatible vehicles list... will it be possible.
And just to make it a whole level harder, I'd really like to add in some temperature and humidity sensing.  Can this be built onto the OVMS or can a unit such as the open energy monitor or their heat pump monitor be used for the temperature and send all of the data to the same remote server (I'm using emoncms in another project, hence stumbling upon OVMS in the open energy shop).    I'd like to use the same IOT sim if I do need two modules, I guess that may need a mobile data router.....     All thoughts welcome. 
Thanks 
Ed

dexterbg, 2024-01-07:
Sure. You'll need to add some support for the Toyota Hiace, and you'll need to add some hardware for the humidity sensors.

ECHEngineering, 2024-01-08:
I've requested the register of Parameters from Toyota (they are providing the vans to the project and "will help if they can" ); no response yet but fingers crossed.  May just have to plug in and see what we can access with one of the existing configs.  I think I'll end up with a separate system for the additional temperature and Humidity though.  Just for simplicity.