openvehicles/Open-Vehicle-Monitoring-System-3 #964: Fiat 500e not showing charging status

pfwolbers, 2024-01-11:
I've just installed the ovms module in my fiat 500e (2015).
I can see the soc and lock doors etc.
But when i charge the car (granny charger) it is not detected by ovms. 

Not charging
SOC: 64.0%
Est. range: 65km
ODO: 74587.0km

pfwolbers, 2024-01-13:
State is not chowing correctly in app and webpage, but is showing correctly at mqtt server.

dexterbg, 2024-01-14:
You should check your server v2 connection & configuration then, as there is no difference regarding state data source between v2 & v3 (MQTT) transport.

dexterbg, 2024-01-15:
`v.c.charging` is only one part of the charging state info. The web page displays `v.c.state`, and for a minimum charge state record, the vehicle also needs to supply the charge port state `v.d.cp`.

The Fiat 500e module source has logic to supply these. If that doesn't work correctly for your model, you should contact the module developer, Günther Huck. Check out his blog: http://fiat500e.eu/

pfwolbers, 2024-01-15:
Ok, thank you. Both metrics have no data. So that will be the problem.

ndfranczyk, 2026-10-27:
Additional reproduction on a 2014 Fiat 500e using FT5E and the Kia Soul cable, running 3.3.006-172-g362f8742.

While the owner confirms the car is charging, the web Vehicle panel reports "Not charging." Read-only shell checks returned:
```
v.c.charging  yes
v.d.cp        no
v.c.state     topoff
v.c.pilot     (unset)
```

Source inspection identifies a concrete decoding defect in vehicle_fiat500e.cpp, IncomingFrameCan2, case 0xA194040:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/362f8742/vehicle/OVMS.V3/components/vehicle_fiat500/src/vehicle_fiat500e.cpp#L346-L380

The code compares (d[3] & 0x30) with 0x1, 0x2 and 0x3. The masked expression can only produce 0x00, 0x10, 0x20 or 0x30, so all three nonzero branches are unreachable. The zero branch sets v.d.cp=false and v.c.state="topoff". This is consistent with the observed metric combination, although no raw CAN capture was taken to establish the complete causal sequence.

The generic CommandStat implementation gates charging output on v.d.cp and a nonempty v.c.state, rather than v.c.charging:
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/362f8742/vehicle/OVMS.V3/components/vehicle/vehicle.cpp#L1592-L1599
Consequently the observed v.d.cp=false explains the "Not charging" summary despite v.c.charging=yes.

The Fiat driver fetched from upstream master on October 26, 2026 was byte-for-byte identical to the installed-revision source. Its charging flag is also written independently from two CAN1 signals (0x820A040 and 0x640A046), which may merit review for inconsistent transitions.

This establishes the unreachable branches, but not a validated replacement decoder. The bit position/width, enum meanings and charge-port semantics still need confirmation against a reliable signal definition and captures covering unplugged, plugged/idle, actively charging and completed states. Merely shifting the existing mask is not yet a vehicle-tested fix.

No metric overrides, firmware changes or vehicle-control commands were used during this diagnostic check.