openvehicles/Open-Vehicle-Monitoring-System-3 #972: OVMS Module Full Reflash

Szlovakricsi, 2024-01-21:
Hello i had a bad ESP32 chip with i changed but i cannot relash the new chip its readable by the computer but it kepps comeing back with an error. Can you help? Thank you.
The Log:

esptool.py v4.7.0
Serial port COM5
Connecting....
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme 3/4
Crystal is 40MHz
MAC: 30:ae:a4:4c:ea:38
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00010000 to 0x0038cfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 22960 bytes to 14682...
Writing at 0x00001000... (100 %)
A fatal error occurred: Guru Meditation Error detected (IllegalInstruction)

markwj, 2024-01-29:
Precisely which ESP32 WROVER module did you purchase for this replacement? Please provide details such as chip type, revision, SPIRAM size, Flash size. A picture of the module would be helpful.

Also, please provide the exact flash command you are using, ESP IDF version (if building from source), etc.

Your output seems strange, as the partition table offsets shown don't seem to match what we have in our project partitions.csv file.

markwj, 2024-01-29:
Please also provide the output of espyool chip_id and flash_id commands.

closed 2025-04-12 by Szlovakricsi