openvehicles/Open-Vehicle-Monitoring-System-3 #975: vehicle.charge.pilot.on/off storm on Leaf causing device to stop respond or even reboot.

malyjaponec, 2024-02-25:
When I connect AC cable or when I switch on AC on on Nissan Leaf 2012 it started to generate events charge-pilot-on and off messages very quickly and device is overloaded, is irresponsible, doing reboots and sometimes it freeze completely.
[image]

I blamed the ABRP script that I later disabled and it is happening again. I blamed my one line scripts that changes charging limits when I arrive/leave home. Happened also after deleting them. I tried to several combinations.

Adding long files I got from device.
[logs.zip](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/files/14396038/logs.zip)

This behaviour render the device unusable as i can ot read SOC on app while charging most of times or it drops connection after setting on preheating, that are two main reasons why I bought and install the unit. I can test whatever needed.

malyjaponec, 2024-03-07:
New information: I found that the problem is not happening only during charging but during normal drive. This is what OVMS unit receives when car is on. Just stays parked ON, nobody is doing nothing,, no charging cables in socket.

_D (356530) events: Signal(vehicle.charge.pilot.on)
D (356540) events: Signal(vehicle.charge.pilot.off)
E (356540) can: can2: intr=590918 rxpkt=597993 txpkt=3 errflags=0x22401c02 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (356540) can: can2: intr=590918 rxpkt=597994 txpkt=3 errflags=0x23401c01 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
D (356630) events: Signal(vehicle.charge.pilot.on)
D (356640) events: Signal(vehicle.charge.pilot.off)
D (356730) events: Signal(vehicle.charge.pilot.on)
D (356740) events: Signal(vehicle.charge.pilot.off)
D (356830) events: Signal(vehicle.charge.pilot.on)
D (356840) events: Signal(vehicle.charge.pilot.off)
D (356930) events: Signal(vehicle.charge.pilot.on)
...
D (372150) events: Signal(vehicle.charge.pilot.off)
V (372240) v-nissanleaf: Range: ideal=141 km, est=92 km, full=119 km
D (372240) v-nissanleaf: Poll state: 1
E (372240) can: can2: intr=617067 rxpkt=624500 txpkt=3 errflags=0x22401c02 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
D (372250) events: Signal(vehicle.charge.pilot.on)
D (372250) events: Signal(vehicle.charge.pilot.off)
I (372300) ovms-duk-util: [lib/abrp.js:25:log] (2024-03-07 21:48:24.000+01:00) INFO: Not parked or DC fast charging
D (372350) events: Signal(vehicle.charge.pilot.on)
D (372350) events: Signal(vehicle.charge.pilot.off)_

The problem is not in server task that kills the unit because want to transfer all data. The problem is that unit does not know CAN messages of this car and thinks that some message that travels on can several times per second if car is on is something about charging. Not it is not. 

**And I cant tell you just logging this kills the unit time to time.**

malyjaponec, 2024-03-22:
Today the unit works very well (it restarts during charging once in 5 minutes but to the mobile app it sends every secoond charging start and charging stop message. No way to see how much was chargerd as the notificaiton in android change forth and back as it charging or not charging. There is probably wrong interpretation of CAN messages what is charging and what is end of charging. This is the only reason why whole device can restart 5 times in 5 minutes. I am not expert but there should be CAN message database for this car. 
I always pray that device does not freeze and will be able to stop charging at 80% and I will not get car to 100%. 
Web interface is almost all the charging time dead, if it opens it stops responding in several seconds.

velihukka, 2024-03-25:
Version 3.4 seems to fix this issue, it is available currently through the eap channel for testing.

malyjaponec, 2024-03-25:
Updating right now. I will monitor it and give feedback in a few days.