openvehicles/Open-Vehicle-Monitoring-System-3 #976: Support for FIAT e-Doblo (same as Peugeot Rifter, Opel Combo, Citroen Berlingo?)

ramack, 2024-02-25:
I would like my Fiat e-Doblo being supported by OVMS. Technically it is a FIAT-branded Peugeot e-Rifter as the Opel Combo-e and Citroen e-Berlingo are also, so I guess all of them would be usable with the same communication.

My plan is to help out here and dig into the details of the CAN frames to get this working.

ramack, 2024-04-02:
I wonder how the vehicle support should be named 🤔.

I own a Fiat e-Doblo and will use it for testing. I am 99.99% sure it will work perfectly for the sister branded cars Opel Combo-e, Peugeot Rifter, Citroën Berlingo and the Toyota ProAce City, which are claimed to be identical in construction. All are using the [PSA/Stellantis EMP2 platform](https://de.wikipedia.org/wiki/Efficient_Modular_Platform_2) and I could imagine that all those electric vehicles will also work, but this is neither sure nor can I test it.

So what do you think? Is it better to go for something generic like

1) generic
**short type code:** E2
**long vehicle name:** Stellantis EMP2 (components/vehicle_stellantis_emp2)
**unique log tag:** v-semp2
**unique namespace prefix:** semp2
with the drawback of potential incompatibilities with other cars in this platform and more hard to identify the name for an end user who owns something like an "Fiat e-Doblo"?

2) specific
**short type code:** FD
**long vehicle name:** Fiat e-Doblo (components/vehicle_fiat_e-doblo)
**unique log tag:** v-fedob
**unique namespace prefix:** fdob
with the high likelihood that many other cars will work with it?

or any mixed form of them? - I'm not enthusiastic about both options, so any proposal, idea or comment is warmly welcome.

dexterbg, 2024-04-02:
It's up to you as the developer, but if you can only test & guarantee it will work for the Fiat, then I'd use that code scheme. You can always add further vehicles to the long name & documentation later on. If you already know it will work on a number of brands, I'd choose the one that will be most common for users.

Btw, if you need code variations based on the vehicle brand, that can be done later on by adding alias codes or code extensions, or by some vehicle variant user config. Code switches sometimes are necessary even on the same platform, if VINs are encoded differently for example.

But please note that both `E2` and `FD` are not acceptable now for new vehicles. Use a code that has at least two characters for the brand and two for the vehicle. And as we already have `FT5E` for the Fiat 500e, e.g. `FTED` or `FTDO` could be the Fiat e-Doblo.

You can be more explicit on the log tag, for example the Fiat 500e uses `v-fiat500e`.

Finally, the namespace prefix needs to begin with `x`, so could e.g. be `xfd`.

ramack, 2024-04-02:
thanks, so best seems to be 
short type code: **FTDO**
long vehicle name: **Fiat e-Doblo** (components/vehicle_fiat_e-doblo)
unique log tag: **v-fiatedob**
unique namespace prefix: **xfd**

rmivdc, 2024-07-28:
From some other's German Forum we can get those commands from a standard ELM237 : 
"
> 
> #######
> ####### Setup commands
> #######
> ATWS ---> Warm reset set back to defaults
> ATI ---> Show interface ID
> ATE0 ---> Turn off terminal echo
> AT@1 --->
> ATSP6 ---> Set to protocol 6: ISO 15765-4 CAN (11 bit ID 500 kbaud)
> ATAT1 --->
> ATL0 ---> Disable line feed
> ATS0 ---> Disable sending spaces between hex characters
> ATH1 ---> Turn on odb2 header
> ATCAF1 ---> Turn on CAN auto formatting
> ATSH79B ---> Set request message header 79B
> ATFCSH79B ---> Set flow control header (CAN ID)
> ATFCSD300000 ---> Set flow control data
> ATFCSM1 ---> Set flow control mode; 1 = User defined CAN ID and data
> 
> #######
> ####### Data commands
> #######
> ATSH6A2 ---> Set the header of transmitted OBD messages to header 6A2
> ATFCSH6A2 ---> Set flow control header (CAN ID)
> ATCRA682 ---> Set the CAN hardware filter to 682
> 22D4341 ---> Ambient temperature
> 22D8EF1 ---> Battery temperature
> 22D4021 ---> Speed
> ATSH6B4 ---> Set the header of transmitted OBD messages to header 6B4
> ATFCSH6B4 ---> Set flow control header (CAN ID)
> ATCRA694 ---> Set the CAN hardware filter to 694
> 22D8161 ---> Battery current
> 22D8151 ---> Battery voltage
> 22D86F1 ---> Battery minimum voltage
> 22D8701 ---> Battery maximum voltage
> 22D4101 ---> SOC calibrated
> 22D8651 ---> kWh available
> 22D8601 ---> SOH
> 22D8101 ---> SOC
> 
> And the already decoded values:
> SoC calibrated (22D4101) = value / 512
> bat min voltage (22D86F1 ) = value in mV
> bat max voltage (22D8701) = value in mV"

Another info from French Forum : 

> On the OBD socket (located on the driver's side, remove the cover under the steering wheel on the left), only the pins 1,3,4,5,6,8,14,16 are wired.
> 
> The standard/regulatory CAN bus is on pins 6 and 14 (6: high, 14: low)
> The Diagnostic CAN Peugeot bus is on pins 3 and 8 (3: high, 8: low)
> 

> Details:
> pin 1 + 12v after power
> pin 3 can high Peugeot
> pin 4 body ground
> pin 5 electro ground
> pin 6 can high standard
> pîn 8 can low Peugeot
> pin 14 can low standard
> pin 16 + 12v OBD
> 
> OVMS side:
> Can1 = regulatory/standard can
> Can2 = pins 3 and 8 of the OBD socket must be re-mapped on connectors 13 and 12 respectively for OVMS
> 

it seems that some advanced commands (other's than getting informations...) requires some specific OBD wiring.

ramack, 2024-07-28:
That looks interesting. I already read the SOC from a CAN frame. Beside that I am currently busy with some other things, but surly will come back to the topic. @rmivdc do you also have such a car?

ramack, 2024-12-08:
I tried to read the vehicle speed on the OVMS shell
```
obdii can1 req dev 6a2 682 22d402
6a2[682] 22d402: ERROR: timeout waiting for poller/response
```

But I guess I have to start a diagnostic session first or send a tester present. Does anyone have an idea on how I could do that from the OVMS shell?

dexterbg, 2024-12-09:
See Wikipedia or ISO 14229, diagnostic session control is command `10`, with extended diag session being value `03`, so in theory for your device:
```
obdii can1 req dev 6a2 682 1003
```

The either send a "tester present" (`3e`) manually…
```
obdii can1 req dev 6a2 682 3e80
```
…or start a tester timer:
```
OVMS# re obdii tester start ?
Usage: re obdii tester start <bus> <ecu> <interval>
```

rmivdc, 2024-12-09:
> That looks interesting. I already read the SOC from a CAN frame. Beside that I am currently busy with some other things, but surly will come back to the topic. @rmivdc do you also have such a car?

Yes, psa ev car.

Some folks here have completely reversed engineered the firmware and offers some arduino based tools and flashing platform : 

https://vlud.net/nac-telecodage.html

ramack, 2024-12-22:
I tried for several hours today, but I am lost.
Using a BT-OBD2 adapter and (and old version of) the Android App Car Scanner I can view a lot of data (e. g. HV battery voltage) and this one has only the standard CAN pins connected on OBD 6+14.
With the BT-Adapter the UDS commands from OVMS seem to get answered, also but without it I cannot manage to make the car answer my requests.
I tried to analyze the CAN dumps but without success.

dexterbg, 2024-12-23:
Not sure if you've done that ("CAN dumps"?): as you've got the working Car Scanner App, you can use the OVMS to capture it's CAN traffic with the car by using a simple OBD2 Y-cable. You should be able to see the session init commands you need to use. Possibly just a car specific session control argument (command `10`), but the ECU might also need more.

When recording a CAN session, use custom events to annotate your actions, or take notes with timestamps.

ramack, 2024-12-23:
I also tried to use the pins 3 and 8 (Peugeot CAN?) as can2 on the OVMS, but as soon as I activate the CAN my OVMS crashes.
Measured with a multimeter (yes I know, but the scope in the car is not convenient this evening :-) against GND I see
PIN  Voltage
3  3.2V
6 2.7V
8 1.8V
14 2.2V
16 12.4V

...
hours later
...

maybe I am just too dumb? I have seen some logs from my own code handling UDS responses
`v-fiatedoblo: poll reply 0xd870 `

as soon as I switched on the ignition and yes, I am sorry to waste your time before. Let me see how this goes...

ramack, 2025-11-27:
Oh oh. Good that you ask. I have it running so far that I see SoC, SoH, Battery voltages, and some other boring stuff like door state.

I always wanted to make the VIN read work and cleanup before sumitting a PR. And cool would be to support preconditioning and charge state.

Every now and then OVMS restarts, and in rare cases I read an SoC of 0 for short periods so there are some bugs I should investigate.

As often with such kind of things I got distracted and as it somehow works for me I lost focus. Please ping me in case you want me to hurry up!

dexterbg, 2025-11-29:
Generally, mostly all vehicles and the OVMS framework itself have occasional crashes, so that needn't be your fault, and your current state may still & already be useful for other Fiat e-Doblo owners. Having the basic metrics available already enables quite a lot of integration.

Regarding the SOC 0 reading, if you create a draft PR from your current state, I can help you with some code suggestions.

trollkarlen, 2025-12-01:
Could not find any specific information for PSA.

You don't have a dump for the retuned value and the readout value from some of tool I can help you reverse engineer it. If you dotn have the real value at least a guess of what it could be like in idle.

Normally values I have seen för HV battery current is
1. Signed 16 bit integer(2 bytes) * either 0.1 or 0.01A
2. Unsigned 16 bit with offset
3. unsigned 16 bit with scale and offset.

closed 2025-12-07 by ramack