openvehicles/Open-Vehicle-Monitoring-System-3 #978: Kia Niro EV (eNiro) 2019. Aux battery high power drain

Toug19, 2024-03-03:
Hi!
I have a 2019 Kia Niro EV (named as eNiro in France). I'm having issue with 12V power drain. I was thinking that the OVMS module was consumming a lot of power. So to be sure I measured the 12V battery output current.
In fact the module consumes max 100mA, so the difference when car is idle is only 50%. But what is causing the battery drain is that the module somehow blocs the car to go to idle state.
I think that because the module sends frames or because it is plugged in the obd, some calculator stays awake, draining the battery. The current stays at around 2 A instead of 150 mA.
When I disconnect the module, the current keeps at 2A for around 2 minutes then goes to 150 mA.

My firmware is "3.3.003-610-g29cc0734/ota_0/eap (build idf v3.3.4-849-g6e214dc33 Jan 8 2024 00:06:40".

I don't know if this is a bug or a configuration issue. But I tried some configuration changes without success.

Regards,

Wireheadbe, 2024-03-05:
It's normal unfortunately with the e-Niro. By having OVMS active, the car doesn't go to sleep. You can see a workaround here: https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/275#issuecomment-754598211

Toug19, 2024-03-06:
Thanks, I have some question that I post in the topic you provided.