openvehicles/Open-Vehicle-Monitoring-System-3 #980: Connecting OVMS unit to running LEAF ZE0 caused orange car symbol on dashboard.

malyjaponec, 2024-03-06:
Connecting OVMS to running car immediately trigger an orange car symbol on dashboard as it generates some errors and it does not disappear with car restart. these errors are not temporary. I had to manually clear them.

Because of other issue the unit restarts during charging time to time (if it does not freeze completely) and these restarts causing more rerros so in one moment after 6-8th of charging that day the car went to not drivable condition. It did not switched on HV battery at all. I had to take a tool and remove all errors to drive out of charging place.

It seems to me that during startup it must block CAN bus somehow that some messages between car and battery are missing or must ask such requests that it prevents normal traffic on the CAN. 

the symbol on dash is almost instant. I do not know how long it takes unit to send first data to CAN but it looks like it must be before it. 

Car is upgraded ZE0 LEAF with 30kWh battery, using Dala's CAN bridge. There are always some non fatal errors in the car due to not perfect implementation of communication with battery bms but generally it did not caused any long / short term problems than occasional orange car symbol after DC charging that always disappear after car restart as the charger errors were temporary.

Connecting OVMS to the switched off car usually made no problem, but the car must be switched off for longer time when the all units sleeps, the symbol as i can remember lights up when I plug it in less then 5s after switching car off.

What else I can test or measure to find why this is happening?

art103, 2025-01-12:
Thank you for raising this issue! 
Given the cold weather recently starting to thaw, I thought I had an electrical fault with my OVMS unit causing *all* of the CAN systems in the car to fail simultaneously!
Turns out there was a firmware update whilst I was ChaDeMo charging at 2am (don't ask!). 

I am assuming that when OVMS rebooted, that is when the car's dashboard lit up like a Christmas tree.
It had "---" for range, car wouldn't drive, even the A/C wouldn't turn on, so this is a pretty major issue IMHO.
As you've said, the only way to fix is to unplug OVMS, clear all of the DTCs with LeafSpy, power off the car, then plug in OVMS.

FWIW, this is the firmware version that triggered it for me:
3.3.004-339-gf88eb24f/ota_0/edge (build idf v3.3.4-851-gfa4f07fb3 Jan 12 2025 00:05:24)

michalstral, 2025-02-15:
> Thank you for raising this issue, it saved me days of HW debugging! Given the cold weather recently starting to thaw, I thought I had an electrical fault with my OVMS unit causing _all_ of the CAN systems in the car to fail simultaneously! Turns out there was a firmware update whilst I was ChaDeMo charging at 2am (don't ask!).
> 
> I am assuming that when OVMS rebooted, that is when the car's dashboard lit up like a Christmas tree. Charging terminated, it had "---" for range, car wouldn't drive, even the A/C wouldn't turn on. This is a pretty major issue IMHO. As you've said, the only way to fix is to unplug OVMS, clear all of the DTCs with LeafSpy, power off the car, then plug in OVMS.
> 
> FWIW, this is the firmware version that triggered it for me: 3.3.004-339-gf88eb24f/ota_0/edge (build idf v3.3.4-851-gfa4f07fb3 Jan 12 2025 00:05:24)
> 
> 2015 AZE0 24kWh Tekna, Stock, other than disconnection of Telemetry unit and addition of OVMS.

I experienced the same problem during a night firmware update. It's the same Leaf year and model as yours.
What firmware version do you use now and do you have any errors since the upgrade?

dexterbg, 2025-02-16:
The issues described can be related to the brief high current peaks the module (the modem) produces during startup (when announcing itself to cell towers nearby).

Combined with an aged 12V battery and winter conditions, a high 12V current can lead to a substantial voltage drop. The VCU probably detects either of this, classifies this as a major fault and shuts down all running operations.

If you boot the module while the VCU is off, the 12V battery recovers before the fault gets triggered.

To check if the modem is the culprit: disable the modem auto start, then try if rebooting still produces the issue.

I suggest checking the car's 12V battery. You could also try supplying some extra juice to the module by connecting a USB power bank -- it may still draw current from the 12V supply, but that may help reducing that.

art103, 2025-03-11:
[image]

art103, 2025-03-11:
Ok, bonfire was an exaggeration.
I've had the OVMS on the workbench and I can confirm that the CAN bus (can1) goes crazy at reboot.

Setup: 
Spare Leaf instrument cluster
Trimmed CAN capture played back on loop to make the cluster work
OVMS hooked up (shows speed on the status page)
Oscilloscope on the CAN H and L lines
Click reset on the OVMS web UI: 100% reproducible output flooding the CAN bus.

Please see the attached video: 
https://github.com/user-attachments/assets/1860b890-0e93-41b3-a940-cb1a753cced5

dexterbg, 2025-03-11:
Thank you for the very thorough analysis and report, very helpful.

From some first tests with the before/after versions, my conclusion is the CAN bus shutdown wasn't done at all on a reset before the change. So fixing the bus shutdown actually caused this issue.

As you see a brief bus noise period during cold boot as well, I assume the noise is present while/whenever the SN65 transceiver is off. The transceiver is switched on at boot when configuring and starting the CAN controller (in `esp32can::Start()`). When stopping the bus, the SN65 is switched off after stopping the CAN controller (in `esp32can::Stop()`).

If you'd like to try, you can take manual control of the can transceiver any time by setting the egpio output port 2:
- `egpio output 2 1` = switch off SN65
- `egpio output 2 0` = switch on SN65

Assuming switching off the transceiver leaves the bus connection in an open state, and assuming either signal reflections or radio noise then cause the issues: maybe enabling the OVMS internal bus termination would solve this for the Leaf?

You can enable the internal 120 ohm termination for CAN1 by closing J4 on the PCB (see image below).

Alternatively you can attach an external 120 ohm resistor to the CAN H/L pins of the DB9 connector.

Could you give that a try?

[image]

dexterbg, 2025-03-11:
As a software workaround, we can add a config to suppress switching off the transceiver on CAN stop, or suppressing the CAN stop operation altogether.

That won't help for the cold boot, but at least for regular reboots.

You can try if keeping the transceiver on helps by commenting out this line:

https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/cfdccef8c5727c06254dcd5950edb57416bfa782/vehicle/OVMS.V3/components/esp32can/src/esp32can.cpp#L574

art103, 2025-03-11:
Thanks - I should find some time tomorrow evening to take another look.

I already have a 120 Ohm termination resistor on the DB9 connector and the bus is being actively driven (correct differential signals observed on the scope). Sorry, I was misleading calling it noise - it's a driven signal of some sort, I just don't know if it's from the ESP32 CAN peripheral, or just a floating TX input to the sn65hvd233d.

What you've said about the transceiver being turned off seems counter intuitive. A high on the sn65hvd233d Rs pin should put it in listen only mode, which isn't consistent with what I'm seeing.
For the cold boot case that makes sense as it will default to egpio 1 = 0 (TX enabled) and the ESP32 IO25 will be input, floating (i.e. noisy)?
For reboot, assuming the max7317 keeps its state whilst the ESP32 is rebooting, I'd expect it to be clean (no TX)?

Anyhow, more experiments tomorrow 🙂.

dexterbg, 2025-03-12:
The MAX7317 defaults to all ports input/high level = SN65 off.

You have other participants on the bus, the signal may originate at the Leaf instrument cluster, possibly an error triggered by a signal reflection. That could also explain why this seems to be a Leaf specific issue.

art103, 2025-03-12:
> You have other participants on the bus, the signal may originate at the Leaf instrument cluster, possibly an error triggered by a signal reflection. That could also explain why this seems to be a Leaf specific issue.

Agreed.

I can confirm that during cold boot and soft reboot, pin 1 (TXD) of U12 does not move, so we can rule out noise on that signal, or anything coming from the ESP32.
Also, the state of pin 8 (Rs) is as expected: high at cold boot until booted, and toggles on the 'not working' FW during reboot as expected.

I need to move the experiments into the car, as the 20Hz bursts must be originating from the instrument cluster (they're differential signals and it's the only other bus device) but only when the bus is in an unloaded state (SN65HVD233D in RX only mode).
This screams termination issues like you say, but I have 120Ohms on each end and twisted pairs in between - admittedly on a very short setup.

In the car, the only thing I can think of is a Y-splitter on the OBD port and the long(ish) DB9 - OBD cable.
I wonder if the ECUs in the Leaf have a bus health check mechanism so that if it sees a reflection, it throws a wobbly?!

A simple way to find out would be to connect the OBD - DB9 cable without OVMS and see what happens: should be reflection city...
Will try that when my wife gets back.

dexterbg, 2025-03-21:
Did you test keeping the transceiver enabled?
https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/980#issuecomment-2715879015

@markwj Any other ideas?

dexterbg, 2025-03-22:
> I hooked up a Seleae logic analyser to the 3.3v side of the can1 transceiver (TX and RX). I see an unexpectedly high amount of TX ACKs (single bits), but assume that is something to do with the ODBII connection?

I'm wondering if you're aware of the discussions running on the developer list, and if you're currently using my test patch to the bus timing?
http://lists.openvehicles.com/pipermail/ovmsdev/2025-January/016651.html
[can1-sae.diff.txt](https://github.com/user-attachments/files/19401952/can1-sae.diff.txt)

A slightly shifted CAN timing could cause unexpected ACKs and errors. If you're running my patch, that could mean it makes things worse, if not maybe you could give that a try. Either way, you possibly could actually measure a difference the patch makes, where the other Leaf developers need to rely on observing behaviour (inconclusive results up to now).

frogonwheels, 2025-03-22:
This is great work and beyond what I imagined.

Looking at your 'edit 5' - the calls you have there to shutdown the power on the busses are when CONFIG_OVMS_COMP_POLLER is undefined .. though a similar thing happens in MyPollers.ShuttingDownVehicle() - so essentially the same. (unloading the vehicle module causes any busses that were turned on by the vehicle to be turned off).

So. it seems that the contribution the poller change made was to always shutdown the cans on shutdown? And that doing so has made it break in the same/similar way to the cold boot? I can certainly provide a switch to disable turning off the busses as suggested by @dexterbg.  Either a config or some kind of call from the specific Vehicle module (or both). 

If it helps I've been running the new can timing on my ioniq 5 for a while now.

frogonwheels, 2025-03-25:
> As a software workaround, we can add a config to suppress switching off the transceiver on CAN stop, or suppressing the CAN stop operation altogether.
> 
> That won't help for the cold boot, but at least for regular reboots.
> 
> You can try if keeping the transceiver on helps by commenting out this line:
> 
> [Open-Vehicle-Monitoring-System-3/vehicle/OVMS.V3/components/esp32can/src/esp32can.cpp](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/cfdccef8c5727c06254dcd5950edb57416bfa782/vehicle/OVMS.V3/components/esp32can/src/esp32can.cpp#L574)
> 
> Line 574 in [cfdccef](/openvehicles/Open-Vehicle-Monitoring-System-3/commit/cfdccef8c5727c06254dcd5950edb57416bfa782)
> 
>  MyPeripherals->m_max7317->Output(MAX7317_CAN1_EN, 1);

I have added a config  but also a parameter which allows the Leaf implementation to suppress the CAN stop operation on shutdown. (which has been merged now ty).

This needs to be a temporary work-around... since the bug still exists from cold as you say... but I guess it would at least put it back to where it was at before the poller code.

art103, 2025-03-25:
Sorry for the radio silence - it's been, and continues to be, a busy time here.

@dexterbg, thank you for the link to the timing change. I need to get set up to build from source so I can capture the bus changes. It's my wife's car so doing any analysis involves some overhead.

@frogonwheels, That's great, thank you!

> This needs to be a temporary work-around... since the bug still exists from cold as you say... but I guess it would at least put it back to where it was at before the poller code.

Agreed.

I feel as though there are 2 issues here: 

1) Instability when the CAN transceiver is in high impedance mode.
I've only seen this on the bench and could not reproduce it in the car; however it fits with @malyjaponec's original post and what I saw during the overnight ChaDeMo session.
Assuming this is one of the causes, I think your workaround helps a 'normal' reboot (e.g. after an OTA update).
The case in [this post](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/980#issuecomment-2713696099) was due to a crash + reset. Do you know if the shutdown sequence is triggered here? 

Assuming not, then:

2) Combination of electrical instability (long spur from OBD2 port to OVMS unit?) and CAN timings?
Given the difficulty reproducing the issue in the car, I'm inclined to think there is another unknown variable somewhere and need to do more experimentation. @dexterbg's link to the CAN timing offers another avenue to explore.

dexterbg, 2025-03-25:
Att: at boot, `Housekeeping::Init()` calls `MyPeripherals->m_esp32can->SetPowerMode(Off)`, which in turn powers off the SN65. The vehicle module will be loaded shortly after and power on the transceiver, but that means even without powering down the bus before a regular shutdown, and also with a reboot due to a crash, there will be a brief period with the SN65 being off.

So the config controlling the SN65 shutdown behaviour also needs to be effective for the Housekeeping init procedure.

wayne-love, 2025-03-25:
I'm getting the same issue on a AEZ1 Leaf when the OVMS module crashes (due to a buffer overrun).

frogonwheels, 2025-03-25:
> <openvehicles/Open-Vehicle-Monitoring-System-3/issues/980/2750494005@
> github.com>

I think this should do the trick.. The using the max buffer in the
reserve() should prevent any of the called functions from dipping into
unchecked memory if the packet is too short.

https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/1118

dexterbg, 2025-06-14:
The MAX7317 does not do an automatic reset on a crash/reset. The only defined reset condition according to the spec sheet is the initial power up, there is no reset command as well. Our driver does not do any pin mode/level change during init, so on a crash, the pins should retain their pre-crash state, until components request a change. We probably should add setting all pins to high to the MAX7317 init, to get a consistent init result and correct reflection of the state variables, but I think that's not relevant here.

Looking at our init sequence lead me to this idea: what if the ESP32CAN has an issue with it's reset mode, an issue that does not allow the controller to be set to reset mode for *any* time while the transceiver is in active mode?

Our ESP32CAN driver assumes the controller will be perfectly silent while in reset mode. Based on that assumption, the controller is set to reset mode within the class constructor (assuming that's safe), and the MAX7317 calls in Start/Stop do not care about keeping a specific sequence with the reset mode change.

So on a crash, the ESP32CAN constructor will set reset mode, while the SN65 is still in active mode, leaving ~430 ms until `Start()` is called. And on a regular shutdown/reset, `esp32can::Stop()` will still first set reset mode, then set the SN65 to listen mode, which leaves a couple milliseconds (SPI!) of potential mayhem.

I've reworked the init sequence to test this theory, see attached patch. I've tested this on my vehicle, the ESP32CAN init & shutdown work as before. I cannot test if this has any effect on the issue, as my vehicle (Seat Mii) doesn't have that, and I don't have any CAN debug equipment. So please test & report.

Regards,
Michael

[esp32can-initsequence-patch.txt](https://github.com/user-attachments/files/20739266/esp32can-initsequence-patch.txt)

dexterbg, 2025-06-15:
If the theory is correct, we might still have an issue with crashes, if the ESP32CAN subsystem automatically enters reset mode on an ESP32 master reset. We would need to switch the SN65 to listen mode from within the crash handler, which won't work with the standard SPI library.

zbchristian, 2025-06-15:
If the crash handler would be able to switch to listen-only mode via a regular GPIO, would be a very good solution. 

Yes, the pullup of the RS pin should definitely be kept.

dexterbg, 2025-06-15:
As the MAX7317 sets a pin with output level high to high impedance / input mode, the connection to EGPIO_1 doesn't even need to be cut, and the connection to EXP_1/2 can be made simply by connecting the pins at the expansion port (pins 7/9 & 17) -- a simple jumper cable will do. That's probably doable for most users, and easily reversible.

The software side is pretty simple. I can implement that as soon as I know it will actually solve the issue.

Christian, as you already use a standard GPIO port to control your SN65, you can easily test this: add switching the SN65 to listen mode within `Boot::ErrorCallback()`. Combine this with my previous patch here, to make sure starting the bus also won't switch to active mode before the controller is released from reset mode.

dexterbg, 2025-06-19:
The crash handler doesn't attempt any recovery, that's outside its scope. The crash handler only tries to carry some crash data over the reset for post-mortem analysis (see `boot status` and the server table `*-OVM-DebugCrash`). It could be used to do some final low level IO state changes before reset, but not much more. Also, a module being unresponsive to network connections doesn't imply the module wasn't doing some reset(s), it could for example have been stuck in a crash/reboot loop. To know for sure, you need to watch the module's USB output in that state.

wk4084987-cmd, 2026-08-21:
When exploring technical problems, clear step-by-step resources can make complex subjects easier to understand and troubleshoot. The same learning approach applies beyond automotive diagnostics, especially when developing patience and logical problem-solving skills. For beginners interested in structured puzzle solving, [rubik's cube tutorials](https://cubemastery.com/rubiks-cube-tutorial-step-by-step/) can provide a simple way to practice following sequences, understanding patterns, and working through challenges systematically.