openvehicles/Open-Vehicle-Monitoring-System-3 #1042: Can I monitor the car's charge when it is turned off?

Gabriele-tomai00, 2024-08-02:
Hi, it is not clear to me if and when the monitoring system can check the status of the car even when it is turned off. 
I am interested in seeing the battery growth during charging when the car is off.
I know that the OBD-II port is disabled if the car is off.

I have an Opel Corsa-e (2020). I know it's not supported but maybe I can reverse engineer it. 

Thank you

glynhudson, 2024-08-02:
OVMS can only monitor the SoC when the car is on or actively charging. However, the SoC will only change when the car is on or actively charging.

Gabriele-tomai00, 2024-08-02:
How is it possibile if the OBD-II il turned off during the carge?

frogonwheels, 2024-08-09:
AFAIK The OBDII port does not get turned off (though maybe on some cars?)  but on my car at least, there is very limited access to ECUs.  Some are just not available when the car is locked.  
Also, if you keep polling the ECUs that are there, then you tend to get an issue with battery drain.

glynhudson, 2024-08-10:
The ODB2 port is never turned off, but the ECU modules that OVMS is reading from will only be awake when the EV is switched on or driving. Some EVs may wake up some modules to periodically engage the DC-DC converter to top up the 12V batter. OVMS will be able to read the SoC while AC and DC charging is in progress.