openvehicles/Open-Vehicle-Monitoring-System-3 #1053: Ioniq 38kWh Support

Alco4998, 2024-08-19:
Hey Guys, 

(Possibly a duplicate of #255 )
There appears to be a lot broken with the newer Ioniqs.
As the owner of one, i am looking to put in the work to get some of these issues resolved.

some of the initial issues are: 
12V Battery recharge is triggering Charge Notification.
and a whole bunch of metrics are out of wack or non-existent

Still evaluating what needs to be done but if there are other owners that can corroborate issues I'd be happy to hear them out

Alco4998, 2025-03-25:
Hey guys
I have finally started work on this;
12v amperage is broken but using other scan tools the there are valid data points containing that data but this is still WIP

I have refactored IGMP, which includes general vehicle security data points so now all seatbelt and doors respond correctly on my branch

If any other owners are watching this, I’d love to hear from you as to what issues you are having!

dexterbg, 2025-03-25:
I suggest posting this on Ioniq forums and on the developer mailing list, most developers and users do not watch/follow github issues.