openvehicles/Open-Vehicle-Monitoring-System-3 #1061: P317E-00 thrown when charging stopped by OVMS

Lothsahn, 2024-09-02:
I have a 2017 Nissan Leaf with a factory replaced 40 kWh battery.  It is throwing a P317E-00 08 EV/HEV HV Battery System EVC-249 code, I believe when ovms stops charging.  I have feature #10 (Sufficient SOC) set to 68 to stop charging early.

I read the report by @markfinn in https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/issues/287 that it might be triggering relay degraded counts, so I was concerned this could be damaging my car.

I have a 2017 S with charging timer set.  Charging from a 40A Juicebox EVSE.

Running version 3.3.004-4-gab8f16b7 of ovms.

Lothsahn, 2024-09-02:
Looking through the service manual, it's hard to say, but it says the P317E code is when the VCM detects an error signal from the LBC.  So I think the VCM is shutting things down, which it would be able to do in a normal way.  I think the relay degradation raised in #287 is when the battery disconnects abruptly under high current.

I don't *think* this should raise the count, but it's software engineering, so who knows.

Dala says he's been using this since 2018 on his own car (7 years) without issue, and is not concerned.

bignerdmonster, 2025-11-11:
I have the same thing,l (2017 AZE0, replaced 40kwh battery) with the target SOC at 80%, and I recently discovered the same issue, for testimonial. Car has also felt sluggish since then, but… I’m likely making that up.