openvehicles/Open-Vehicle-Monitoring-System-3 #1062: Main functions missing on 2020 Nissan e-NV200

TehPhoenix, 2024-09-03:
This is probably the wrong place to be asking questions about functions not working after following the installation guide and I have posted questions on the OVMS site forum, but get no response.  Where is the best place to seek assistance please?

TehPhoenix, 2024-09-05:
I received appropriate assistance on the OVMS form, so this can now be closed.
You may however like to note and publish that contary to the instruction given on the OVMS site in respect of installation on this year and model (40kWh), full functionallity was only acheived after setting the model year to 2016.

markwj, 2024-09-06:
What is you exact make, model, year, and battery size for your vehicle? We can add that as an example on the user guide.

TehPhoenix, 2024-09-11:
Nissan, e-NV200, 2020, 40kWh.

glynhudson, 2024-09-13:
This is already mentioned in the e-NV200 OVMS docs here: https://docs.openvehicles.com/en/latest/components/vehicle_nissanleaf/docs/index.html#e-nv200-models

The reason for the quirk is that a post 2018 e-NV200 uses the same platform as a 2014-2016 Nissan LEAF

TehPhoenix, 2024-09-13:
I did check the docs very carefully when installing and they say to set the year to 2015 for the 2020 e-NV200.  This did not work for me. It was suggested by another on the OVMS forum that I try what they did in order to get their system working, and set the year to 2016. Now everthing functions as expected.
I was just reporting that anomaly here.

glynhudson, 2024-12-02:
What functions were not working?

pjogorman, 2025-04-06:
Hi ,
Same issue here env200 2015 
No main function no soc battery percentage cant turn on off ac charging etc.

I followed the 2015 yr setting no joy also did it for 2016 no joy. 

Location working and 12v battery info working.  

Did the reset left it at factory firmware. 

I have enabled can write also. 

Open to suggestions. I also opened on issue on ovms. 

[image]

moffetx, 2025-04-08:
On the leaf the TCU (and therefore some of the functions such as air con) moved from the ev bus to the car bus in 2016 and the following is hardcoded into the firmware to set which to use depending if the year is before or after the move:

if (MyConfig.GetParamValueInt("xnl", "modelyear", DEFAULT_MODEL_YEAR) >= 2016)
    {
    ESP_LOGI(TAG, "New TCU on CAR Bus");
    length = 4;
    tcuBus = m_can2;
    }
  else
    {
    ESP_LOGI(TAG, "OLD TCU on EV Bus");
    length = 1;
    tcuBus = m_can1;
    }

The documentation states that the env200 should always be set to 2015 (to send on can1) but i guess it must have been moved onto can2 at some point because my 2018 env200 needs this date to be 2016 or later. I guess this means that a post 2018 env200 cannot be using exactly the same platform as the 2014-2016 leaf as Glyn states above.

The documentation needs to be corrected but I do not know which year this changed for the env200 - I would guess when they introduced the 40kwh model in 2018? We will need someone with a 2017 model to verify that it is working using 2015 as the date to check.

I don't think that pjogorman is having the same problem - the location and 12v readings are not read from the vehicle and even when the setting was wrong things like the SoC displayed fine. Their vehicle is also a 2015 model which will definately need the 2015 setting. I would suggest updating to the latest stable firmware (main) and setting it up again to double check the rest of the settings are correct because it does not seem to be communicating with the vehicle at all.