openvehicles/Open-Vehicle-Monitoring-System-3 #1071: PubSub.unsubscribe('ticker.1') Unsubscribes from Similar Topics (e.g., ticker.10)

kezarjg, 2024-09-29:
**Description**
When calling `PubSub.unsubscribe('ticker.1')`, the current implementation may unintentionally unsubscribe from similarly named topics like `ticker.10`. This occurs because the code checks for any topic that starts with the string `ticker.1`, causing `ticker.10` to match and be unsubscribed as well, even though it's a different topic.

**Steps to Reproduce**

1. Set up a messages object with the following structure:
`PubSub.subscribe('ticker.1', someFunction);`
`PubSub.subscribe('ticker.10', someOtherFunction);`
2. Call the unsubscribe method for the topic 'ticker.1':
`PubSub.unsubscribe('ticker.1');`
3. Observe that both `ticker.1` and `ticker.10` are unsubscribed from.

**Analysis**
The issue arises from this part of the code:
[`vehicle/OVMS.V3/components/ovms_script/jsmodsrc/pubsub.js`](https://github.com/kezarjg/Open-Vehicle-Monitoring-System-3/blob/2b17ee0d7153709c503db2ad8f1cc831e8db9c04/vehicle/OVMS.V3/components/ovms_script/jsmodsrc/pubsub.js#L169)

`    if (Object.prototype.hasOwnProperty.call(messages, m) && m.indexOf(topic) === 0)
      {
      delete messages[m];
      }
    }`

This logic matches any topic that starts with the provided topic string, causing unintended matches like `ticker.10` when trying to unsubscribe from `ticker.1`.

**Proposed Solution**
Update the detection logic for matches and hierarchical children to;
`(m.indexOf(topic + '.') === 0 || m === topic)`

dexterbg, 2024-09-29:
The PubSub framework is a widely used standard implementation: https://github.com/mroderick/PubSubJS

That means you're free to check for an update at the source, that incorporates a change of that kind. But unless you find one, we should not alter the inherited API without good reason, and if we find such a reason, we should discuss that change upstream first.

So, is this a good reason? No.

Unsubscribing by event name is a **last resort** option for a system running wild due to a bug. It's a **global** unsubscribe, i.e. **all** subscribers matching the name prefix will be removed. That is never to be used by any proper JS plugin during normal operation, it's meant for a developer / OVMS user to use from the shell, primarily during plugin development in case something went wrong.

The proper way to unsubscribe your handler is to pass the subscription ID you got when subscribing to `.unsubscribe()`.

kezarjg, 2024-09-29:
OK. Thank you.

closed 2024-09-29 by kezarjg