openvehicles/Open-Vehicle-Monitoring-System-3 #1088: [MG ZS EV 2021] Can you track SoC while charging? (Docs unclear)

RayGenWurm, 2024-12-26:
Hey everyone,

I recently bought a used MG ZS EV 2021.
So now i found out: there is no APP for this car (at least here in europe) and after some research I came across OVMS.
I read the documentation but I am a bit confused...

1. Can someone explain to me what zombie mode is or why i need it?

2. the most important thing: is OVMS able to show me current SoC while plugged in and charging?
   or is this only working when the car is ON?
   
   thanks,
   RayGenWurm

GreyPeter, 2024-12-26:
Hello RayGenWurm,

First of all, yes it will show SOC when charging. I had a 1st Gen MG ZS EV here in Australia and OVMS works quite well. It shows SoC both when in READY and Charging. I often used it to check SoC whilst public charging.
As for Zombie mode, it is part of the vehicle software and it is either part of your vehicle or not. You should probably just select MG ZS EV (UK/EU) as the vehicle and all should be fine.

Regards,
Peter

RayGenWurm, 2024-12-26:
alright, thank you very much for your fast response!
so i have to get an OVMS system :)