openvehicles/Open-Vehicle-Monitoring-System-3 #1090: authenticating to AWS IOT CORE

ralphmackinder, 2025-01-03:
Hi

Quick issue / question - is it possible to connect OVMS to AWS IOT core using x509 certs? 

Ive loaded the root authority and client cert, but AWS also needs the private key and I have no idea where to load that or even if OVMS can see it / use it? 

IOT core does not allow unauthenticated connections over port 8883 / TLS. 

Thanks

dexterbg, 2025-01-03:
Not yet, but support for a client cert & key for MQTT can be added. See `struct mg_connect_opts`, fields `ssl_cert` & `ssl_key`. You'll need to add the cert & key config UI (copy & adapt that from the webserver), and you then need to load & apply these in `OvmsServerV3::Connect()`.

ralphmackinder, 2025-01-03:
thanks the the quick reply.  all above my level - im just a user really. 

Probably should have asked before spending a few hours learning IOT core :) 

thanks again - will wait and hope to see an update to include keys.

markwj, 2025-01-03:
Note that IoT core in Amazon goes way beyond a simple client certificate. It is a complex ecosystem intended to support hundreds / thousands of devices. We had investigated it for OVMS, but the per-device costs are not possible to support without charging monthly fees (which is not desirable for a project such as ours).

ralphmackinder, 2025-01-04:
> Note that IoT core in Amazon goes way beyond a simple client certificate. It is a complex ecosystem intended to support hundreds / thousands of devices. We had investigated it for OVMS, but the [...]

understood - but for individual users , the free limits look ok for a single device. I setup a bridge from an EMQX local broker and managed to republish messages from IOT core's test client.

closed 2025-01-04 by ralphmackinder