openvehicles/Open-Vehicle-Monitoring-System-3 #1091: Repeated connection drops on V3 / MQTT

ralphmackinder, 2025-01-04:
Hi, 

I managed to implement a free cloud based serverless MQTT solution for the V3 server on OVMS, but im seeing frequent re-connects. The server connects, drops and reconnects **every minute or so.** 

Shell Entries related: 

```
W (974806) websocket: WebSocketHandler[0x3f8402b4]: job queue overflow detected
W (974926) websocket: WebSocketHandler[0x3f8402b4]: job queue overflow resolved, 116 drops
```

Its doing a send all every time it does this too which is obviously not good. 

Other clients connect to the same EMQX cloud server ok and don't drop. 

Behaviour is same over wifi as over 3G. 

Appreciate any advice. Hopefully just something I can tweak in the OVMS config. 

Thanks

dexterbg, 2025-01-04:
The websocket isn't related to the v3 server, but the log entries may indicate your Wifi connection is unstable or slow.

To trace the v3 server's connection level events, raise the log level to `verbose` for component tag `ovms-server-v3`.

I suggest also enabling verbose logging for the EMQX instance to see if that indicates a cause.

If you're running the latest OVMS edge build, you may need to revert the keepalive timeout to the previous default (60 seconds) for the EMQX broker. If not running the edge build yet, give that a try and test lower or higher keepalive values.

ralphmackinder, 2025-01-04:
Hi, just to update. this is resolved. it may have been the broker rate limits or some other quirk of the serverless environment. OVMS or the broker didnt like the data dump of all the topics at once on 'send all. '

I used exclude / include to cut the topics down to 10-15 metrics that i use for sensors,  and now it works perfectly with a much reduced data burden per send. 

thanks

closed 2025-01-05 by ralphmackinder