openvehicles/Open-Vehicle-Monitoring-System-3 #1092: Leaf CAN2 error

developerfromjokela, 2025-01-05:
Hi.

I noticed on my 2011/2012 LEAF, that it doesn't get VIN and charging count from CAN2. Tried investigating logs, shows errors for can2:

```
E (159030) can: can2: intr=234638 rxpkt=236353 txpkt=9 errflags=0x23401c01 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (157980) can: can2: intr=232895 rxpkt=234596 txpkt=9 errflags=0x22401c02 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (115491) can: can2: intr=189260 rxpkt=190732 txpkt=6 errflags=0x22401c02 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (115491) can: can2: intr=189260 rxpkt=190733 txpkt=6 errflags=0x23401c01 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (172830) can: can2: intr=257630 rxpkt=259564 txpkt=9 errflags=0x22401c02 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (172830) can: can2: intr=257631 rxpkt=259565 txpkt=9 errflags=0x23401c01 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (173240) can: can2: intr=258300 rxpkt=260242 txpkt=9 errflags=0x22401c02 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (173240) can: can2: intr=258301 rxpkt=260243 txpkt=9 errflags=0x23401c01 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (174160) can: can2: intr=259837 rxpkt=261790 txpkt=9 errflags=0x22401c02 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (174160) can: can2: intr=259837 rxpkt=261791 txpkt=9 errflags=0x23401c01 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (192820) can: can2: intr=290938 rxpkt=293143 txpkt=9 errflags=0x23401c01 rxerr=0 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
```

These errors are present when OVMS is polling from CAN, when car is on or it is charging. And data that's supposed to come from CAN2 is not present. Using LeafSpy the data is readable. 

How could I try to resolve the issue? Seems like there is something wrong in CAN2 with OVMS. As a control test, Leafspy works fine, no problems with it. Same OBD port, different results.

CrashOverride2, 2025-01-05:
Hi,

I owned a Leaf from 2013 a couple of days now. Of course I throwed a OVMS into it. The Leaf behaves like a 2012 one, it need the 12V wake up and so on. I connected the CAN from the TCU plug to CAN 1 (EV CAN) and like you CAN2 the CarCAN through the OBD Jack (PIN 6/14) and I have the same problem.

I posted in the Nissan Leaf forum, because I dont have time to investigate whats different on my Leaf than others, because I dont heared anything problematic with the OVMS and the Leaf.

https://mynissanleaf.com/threads/tcu-replacement-with-ovms-using-its-case.37084/#post-655916

You are the first one with the exact same problem. I hope the Leaf integration dev reads this and have a solution for us.

Cheers

developerfromjokela, 2025-01-05:
> Hi,
>
> I owned a Leaf from 2013 a couple of days now. Of course I throwed a OVMS into it. The Leaf behaves like a 2012 one, it need the 12V wake up and so on. I connected the CAN from the TCU plug to [...]

I guess people don't notice since it doesn't use it for anything else than reading VIN and charge counts. But I'm using only the OBDII port, no wiring to original TCU. I've actually completely disconnected it since it was interfering with remote AC.

developerfromjokela, 2025-01-05:
Also would be nice to understand the error flags. Went through source code and seems like it's coming from a framework/driver/library? Couldn't find any info or docs about it.

CrashOverride2, 2025-01-06:
Yes its coming from the CAN component, the error rises on different situations. I wrote the integration for the Zoe Ph2, on this everything must be polled, because OBD Port is silent (Can Gateway). I get the error if the CAN interface is down and I want to poll.

The VIN is not interested by me, but I have no ranges and the SOC from the battery is missing, I switched to manual GID entry, to have some percentage, but because battery is gettinger weaker, I need to set the value lower time to time, so I can reach 100% charge level display by OVMS.

dexterbg, 2025-01-06:
> Also would be nice to understand the error flags. Went through source code and seems like it's coming from a framework/driver/library? Couldn't find any info or docs about it.

The error flags are a combination of the status bits of the CAN hardware, in case of can2 that's the MCP2515. You'll find details on the flags in the MCP2515 documentation. You can find some example discussions on error flags in the developer mailing list archive, e.g. this post by Mark: http://lists.openvehicles.com/pipermail/ovmsdev/2021-June/015347.html

developerfromjokela, 2025-01-06:
Thanks. Seems like exact errors happen on leaf as shown in that mailing list.

developerfromjokela, 2025-01-06:
Here is can2 status. Same error flag comes back even when stopping can2 and starting manually.
```
CAN:       can2
Mode:      Active
Speed:     500000
DBC:       none

Interrupts:              162848
Rx pkt:                  163396
Rx ovrflw:                    0
Tx pkt:                       9
Tx delays:                    0
Tx ovrflw:                    0
Tx fails:                     0

Err flags: 0x01000001
Rx err:                       0
Tx err:                       0
Rx invalid:                   0
Wdg Resets:                   0
Wdg Timer:                    0 sec(s)
Err Resets:                   0
```

developerfromjokela, 2025-01-06:
Seems like the buffer 0 gets filled with a flood of messages and something happens with buffer 1 and messages don't get through?

developerfromjokela, 2025-01-06:
> Yes its coming from the CAN component, the error rises on different situations. I wrote the integration for the Zoe Ph2, on this everything must be polled, because OBD Port is silent (Can Gateway). [...]

Which model year leaf is it? I think older models without SOC display do not send SOC info on EVCan. There is a SOC xnl.v.b.soc.nominal from BMS, but left unused because of its inaccuracy (?)

developerfromjokela, 2025-01-06:
While charging, getting error:

```
E (21519108) can: can2: intr=26 rxpkt=25 txpkt=0 errflags=0x80001080 rxerr=9 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (21519118) can: can2: intr=27 rxpkt=25 txpkt=0 errflags=0x80001080 rxerr=45 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (21519118) can: can2: intr=28 rxpkt=25 txpkt=0 errflags=0x80001080 rxerr=63 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (21519118) can: can2: intr=29 rxpkt=25 txpkt=0 errflags=0x80001080 rxerr=72 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (21519118) can: can2: intr=30 rxpkt=26 txpkt=0 errflags=0x81001001 rxerr=80 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
E (21519258) can: can2: intr=44 rxpkt=39 txpkt=0 errflags=0x80001080 rxerr=76 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
W (21519258) mcp2515: can2 EFLG: RX_Err_Warn EWARN 
E (21519258) can: can2: intr=45 rxpkt=39 txpkt=0 errflags=0xa00310a0 rxerr=103 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
W (21519258) mcp2515: can2 EFLG: RX_Err_Passv RX_Err_Warn EWARN 
E (21519258) can: can2: intr=46 rxpkt=39 txpkt=0 errflags=0xa00b10a0 rxerr=130 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
W (21519258) mcp2515: can2 EFLG: RX_Err_Warn EWARN 
```

CrashOverride2, 2025-01-11:
Hi,

my Leaf got first license plate in Germany mid 2013, but this is definitively a Gen1 (ZE0? Or AZE0? Cant tell the difference) because it has the bar in the trunk which houses the charger.

In Dashboard I have only bars no percent value. Nominal SOC is the real soc of battery I think, so is never getting 100% or 0% and I drive the Leaf unti BMS stopped discharging for testing.

I get these CAN errors on CAN2 as well from time to time, have no VIN and locking/unlock is not possible (because it needs to send to CAN2 or its not working with this Gen).

I just scrolled through the code, some data from CAN2 is working, like odometer, throttle, soh, footbrake and gear for example.

No range no time remaining and the current is negative If I charge it, so the App never shows the kW while charging.

developerfromjokela, 2025-01-14:
> Hi,
>
> my Leaf got first license plate in Germany mid 2013, but this is definitively a Gen1 (ZE0? Or AZE0? Cant tell the difference) because it has the bar in the trunk which houses the charger.
>
> In [...]

Seems like Can2 errors occur when polling, aka transmitting data. Time remaining doesn't show even when adding manual battery capacity in settings. I've fixed quite few issues in my repo: https://github.com/developerfromjokela/Open-Vehicle-Monitoring-System-3/. I can share a compiled firmware if you want for test

CrashOverride2, 2025-04-11:
Hi developerfromjokela,

thanks for your reverse engineering work on the old Leaf TCUs, really nice! I had the TCU still laying around and if we have still 3G network here in Germany, I would test it.

So I was very busy with the Zoe Ph2 Integration, now I looked closer to the Leaf one. 

I flashed the latest master from this repo and nothing changed. The errors from CAN2 interface in this scope is negligible, polls are getting through and response is coming in.

I understand that old leafs has no SOC on Instrument, but somewhere the "User SOC" must be. I dont know the original Carwings App, but if the Leaf for example is fully charged, I think it says 100% or?

Fully charged SoC nominal says 93%, this is also not very useful. Range from instrument is filled, but not used in the OVMS metric, to show on app.
chargebars and chargeminutes3kW also not working. During charging I see only the 70V charge voltage offset and If charge is aborted or car is full, I get timerwait even no charge timer is programmed.

Because of all these missing or wrong important metrics, using OVMS with this is a bit annoying.
I dont want to cry around rather than helping improving the quality of the integration with focus on ZE0 2012 model Leafs.

So first thing is to gather informations, I found some service manuals which is good for hardware issues. OE Software seems like Consult III is the key here, but it requires a special interface which I dont have. My cheap Renault VCI doesnt seem to be recognized. There are some DDT files but only for newer models I think. The DBC files from dalathegrate is somewhat useful, I need to check against dumps.

developerfromjokela, 2025-04-12:
> Hi developerfromjokela,
>
> thanks for your reverse engineering work on the old Leaf TCUs, really nice! I had the TCU still laying around and if we have still 3G network here in Germany, I would test [...]

ZE0 2012 I did some modifications to OVMS code to get for example charging info etc working. Have a look at my fork: https://github.com/developerfromjokela/Open-Vehicle-Monitoring-System-3/. It's not up-to-date but there is some modifications made to leaf, ie. detecting AC state, charging plug status, etc. that are broken on master for ZE0 vehicles. Hope it helps!

developerfromjokela, 2025-04-12:
From the stock TCU I also learned one thing, to query latest battery status, aka. "Wake up", you need to pull EV activation signal to VCM. I'm not sure if TCU issues a special command on canbus to not start charging the vehicle but only refresh latest data. 

Also for ZE0 vehicles I cleared a little bit the activation part, using proper canbus command and pulling on EV activation signal.

CrashOverride2, 2025-04-12:
Thanks for your answer, if you have fixes why dont do a PR? Sync your fork to latest master and check against incompatibilities and then Go :-)

I made a small change on the Leaf code to get range calcuation at least working. Without that it was nearly useless. I think the origin developer of the integration has no old Leaf or it may have dependencies on firmware versions of various ECUs?

In my fork I added a simple trip calculation too, I store odometer value in a temporary var and subtract it from actial odo. I dont see any trip distance metric filled, is this probably unwanted?

https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/pull/1128

developerfromjokela, 2025-05-03:
> So I finally tested your code, you added trip calc like me, nice. You calced the capacity with Amphours instead of reading from config, thats nice too. The max_gids is still neccessary for SoC in my case. Charge plug status is now correct.
> 
> Location streaming is still not working, I get a position update on start/stop only. If I look at all metrics v.e.on and v.e.awake is correctly set.

Streaming has been working nicely on mine but haven't used OVMS for around 3 months now :) it was somewhat buggy with it missing can frames quite often making it not so pleasant. I'm using the stock TCU for now.

developerfromjokela, 2025-05-03:
> So I finally tested your code, you added trip calc like me, nice. You calced the capacity with Amphours instead of reading from config, thats nice too. The max_gids is still neccessary for SoC in my [...]

For SOC on ZE0 CANBUS reports LB_NEW_FULL_CAPACITY if you check dala's dbc file. Technically if you just convert it to GIDs and calculate based on that.

dexterbg, 2025-05-04:
Btw, in case you missed it, I meanwhile added error flag decoding to the framework. That includes a command interface, so you can now do…

> While charging, getting error:
> 
> ```
> W (21519258) mcp2515: can2 EFLG: RX_Err_Passv RX_Err_Warn EWARN 
> E (21519258) can: can2: intr=46 rxpkt=39 txpkt=0 errflags=0xa00b10a0 rxerr=130 txerr=0 rxinval=0 rxovr=0 txovr=0 txdelay=0 txfail=0 wdgreset=0 errreset=0
> W (21519258) mcp2515: can2 EFLG: RX_Err_Warn EWARN 
> ```

```
OVMS# can can2 explain 0xa00b10a0
Bus error flags 0xa00b10a0:
 | IR.7 RX/TX Message error | IR.5 Error state change
 | EF.3 Receive Error-Passive (REC >= 128) | EF.1 Receive Error Warning (REC >= 96) | EF.0 Error Warning (TXWAR or RXWAR set)
 | IF.5 Error & wakeup interrupts cleared
```

You'll of course still need to read the controller manual for the technical background. That for example seems to have been a continuous stream of bus errors for more than 128 frames, so the controller entered error-passive mode.