openvehicles/Open-Vehicle-Monitoring-System-3 #1097: Wrong mapping of tyre pressure values with Smart EQ

lhitc, 2025-01-12:
Tyre pressures are assigned to the wrong wheels in the module (see also https://github.com/openvehicles/Open-Vehicle-Android/issues/178 )

On the "car" page, the app displays a tyre pressure of 2.5 bar on both left wheels and 2.2 bar on both right wheels.
Issuing "tpms" on the shell displays the same values.

I checked and in reality it is 2.5 bar on both rear wheels and 2.2 bar on both front wheels.

Ping @Dimitrie78

Dimitrie78, 2025-01-12:
Hi, 
I can tell you that it's not the app or the OVMS. The sensors are read correctly.
The fault lies with the car. The sensors are not assigned correctly in the car, the car doesn't know where each sensor is.
Other users have already noticed this. The sensors are assigned incorrectly in different ways.
Therefore, it makes no sense to change them in the OVMS if the problem lies with the car.

I could possibly build in a function to assign the sensors to the appropriate tire.

Regards Thomas

lhitc, 2025-01-12:
So it's German engineering hitting once again :laughing: 

Having a possibility to assign the wheels/sensors correctly would highly satisfy my inner Monk.