openvehicles/Open-Vehicle-Monitoring-System-3 #1105: Module does not recover from losing cellular connection

joshuajung, 2025-02-14:
Hi there, I am facing issues with my OVMS module not recovering from a connection loss. This is a reasonably reproducible issue for me, so I'd be happy to help debug this!

# Environment
* Smart EQ
* OVMS module as supplied by @Dimitrie78 via [smart EMOTION](https://www.smart-emotion.de/shop/product/75-ovms-unabhängiger-appzugriff/)
* Firmware `3.3.004-382-ge83be0e7`
* dexters-web.de server
* Telekom SIM card 

# Issue
* I am driving through an area with limited cellular connectivity a few times per week. There is a time span of around 30 minutes during which cellular signal quality is pretty bad and fluctuating. 
* It's totally fine that the module loses connectivity here, that's not the issue.
* However, in about 50% of the cases the module does not correctly recover from this connection loss. Instead, it ends up in a state where
  * ✅ `cellular` is in `NetMode` state
  * ✅ `gsm-ppp` is `Connected` and has an IP
  * ❌ `ovms-server-v2` is stuck in `ready to start` state
  * ❌ `ssh` remains stopped
  * ❌ `webserver` remains stopped

# Sample Logs
I limited the logs below to the services mentioned above, log level `Info` and above, and removed all `UART Frame Errors` (there are a lot) and `MP-0` messages by `ovms-server-v2`. Let me know if you want the full thing, I can send it in private. 

_I start driving at around 7:15, connection is lost after about a minute:_
```
Feb 14 07:16:17  ovms  cellular INFO  Network Registration status: Searching
Feb 14 07:16:17  ovms  cellular WARN  Lost network connection (NetworkRegistration in NetMode)
Feb 14 07:16:17  ovms  cellular INFO  State: Enter NetLoss state
Feb 14 07:16:17  ovms  gsm-ppp INFO  Shutting down (hard)...
Feb 14 07:16:17  ovms  gsm-ppp INFO  StatusCallBack: User Interrupt
Feb 14 07:16:17  ovms  gsm-ppp INFO  PPP is shutdown
Feb 14 07:16:17  ovms  ovms-server-v2 ERROR  Status: Connection lost, reconnecting
Feb 14 07:16:17  ovms  webserver INFO  Stopping Web Server
Feb 14 07:16:17  ovms  ssh INFO  Stopping SSH Server
Feb 14 07:16:18  ovms  ovms-server-v2 INFO  Status: Server is ready to start
Feb 14 07:16:27  ovms  cellular INFO  State: Enter NetWait state
Feb 14 07:17:18  ovms  cellular INFO  Network Mode: GSM,Online
Feb 14 07:17:27  ovms  cellular INFO  Network Registration status: RegisteredHome
Feb 14 07:17:28  ovms  cellular INFO  State: Enter NetStart state
Feb 14 07:17:29  ovms  cellular INFO  PPP Connection is ready to start
Feb 14 07:17:30  ovms  cellular INFO  State: Enter NetMode state
Feb 14 07:17:30  ovms  gsm-ppp INFO  Initialising...
```

_At this point, there is a 20 minute break in the logs, which may be related or another issue?_ 

```
Feb 14 07:38:39  ovms  cellular INFO  State: Enter PowerOffOn state
Feb 14 07:38:45  ovms  cellular INFO  State: Enter PoweringOn state
Feb 14 07:38:51  ovms  cellular INFO  State: Enter Identify state
Feb 14 07:38:51  ovms  cellular INFO  Identified cellular modem: SIM7600/Experimental support for SIMCOM SIM7600
Feb 14 07:38:51  ovms  cellular INFO  Set modem driver to 'SIM7600'
Feb 14 07:38:52  ovms  cellular INFO  State: Enter PoweredOn state
Feb 14 07:38:52  ovms  cellular INFO  SIM card PIN code required
Feb 14 07:38:52  ovms  cellular INFO  Using PIN code "REDACTED"
Feb 14 07:39:11  ovms  cellular INFO  State: Enter MuxStart state
Feb 14 07:39:12  ovms  cellular INFO  State: Enter NetWait state
Feb 14 07:39:23  ovms  cellular INFO  Network Registration status: RegisteredHome
Feb 14 07:39:23  ovms  cellular INFO  State: Enter NetStart state
Feb 14 07:39:24  ovms  cellular INFO  Network Mode: LTE,Online
Feb 14 07:39:25  ovms  cellular INFO  PPP Connection is ready to start
Feb 14 07:39:25  ovms  cellular INFO  State: Enter NetMode state
Feb 14 07:39:25  ovms  gsm-ppp INFO  Initialising...
Feb 14 07:39:26  ovms  gsm-ppp INFO  StatusCallBack: None
Feb 14 07:39:26  ovms  gsm-ppp INFO  status_cb: Connected
   our_ipaddr  = 10.164.254.187
   his_ipaddr  = 10.64.64.64
   netmask     = 255.255.255.255
   DNS#0       = 10.74.210.210
   DNS#1       = 10.74.210.211
   our6_ipaddr = ::
Feb 14 07:39:57  ovms  cellular INFO  Network Provider is: Telekom.de Telekom.de
```

_At 7:45, I am back home and cellular connection is stable again. However, `webserver`, `ssh` and `ovms-server-v2` don't come up again. There are no more logs in the meantime. I waited another hour, then pulled the plug and the module came back online fine._ 

Please make sure to let me know if there is anything else I can supply or test. Thank you!

joshuajung, 2025-02-17:
I spent some more time digging through the logs and code trying to trace this a bit further. Here is what I found:

`ovms-server-v2`, `ssh`, and `webserver` shut down because the `network.mgr.stop` event is fired. They are then waiting for the `network.mgr.init` event to start again, but this never happens.

So now I'm wondering: Is the issue that the network manager stops in the first place, or that it doesn't initialize again once `gsm-ppp` is connected again? 

I'll continue digging into this, but maybe someone else has more insights about the intended behavior. 🙂

joshuajung, 2025-02-17:
> So now I'm wondering: Is the issue that the network manager stops in the first place, or that it doesn't initialize again once gsm-ppp is connected again?

I guess that the network manager stopping in the first place is intended, e.g. via this call stack:
1. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L725
2. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L1012
3. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L978

So I assume the issue is that `network.mgr.init` isn't called again.

This should in theory probably happen via this call stack: 
1. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L942
2. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L936
3. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L995

This method doesn't seem to be called, as I cannot find `MongooseTask starting` in my logs.

joshuajung, 2025-02-17:
> So now I'm wondering: Is the issue that the network manager stops in the first place, or that it doesn't initialize again once gsm-ppp is connected again?

I guess that the network manager stopping when the connection is lost in the first place is intended, e.g. via this call stack:
1. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L725
2. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L1012
3. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L978

So I assume the issue is that `network.mgr.init` isn't called again after the connection has been reestablished.

This should in theory probably happen via this call stack: 
1. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L678
1. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L995
1. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L936
1. https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/e83be0e7e8a7de03def65907d8acba428a274720/vehicle/OVMS.V3/main/ovms_netmanager.cpp#L942

`ModemUp()` (1.) is definitely still being called and completed, as the `network.interface.change` gets fired. We never end up in `MongooseTask()` (4.) though, so at some place between 1 and 4 we get lost.

dexterbg, 2025-02-17:
You can log events by either setting the log level to debug for the `events` tag, or by command `event trace on` (= log at info level).

Not running Wifi by default (at least in fixed client mode) is a possible, but unusual setup. Nevertheless, I've just tested this on my bench:
1. Boot with modem only into full operation, simulate network loss by `cellular setstate NetLoss`.
2. Boot with modem only into full operation, power off modem, wait for 5 minutes, power modem on again.

In both cases the network gets all up & running just fine.

```
…
I (767218) cellular: PPP Connection is ready to start
I (768168) cellular: State: Enter NetMode state
I (768168) gsm-ppp: Initialising...
I (768168) events: Signal(system.modem.netmode)
I (768298) gsm-ppp: StatusCallBack: None
I (768298) gsm-ppp: status_cb: Connected
I (768298) gsm-ppp:    our_ipaddr  = x.x.x.x
I (768298) gsm-ppp:    his_ipaddr  = x.x.x.x
I (768298) gsm-ppp:    netmask     = 255.255.255.255
I (768298) gsm-ppp:    DNS#0       = 8.8.8.8
I (768298) gsm-ppp:    DNS#1       = 8.8.4.4
I (768298) gsm-ppp:    our6_ipaddr = ::
I (768298) events: Signal(network.interface.up)
I (768308) events: Signal(system.modem.gotip)
D (768308) netmanager: Saved DNS#0 8.8.8.8
D (768308) netmanager: Saved DNS#1 8.8.4.4
D (768308) netmanager: Saved DNS#2 0.0.0.0
I (768308) netmanager: Set DNS#2 0.0.0.0
I (768308) netmanager: MODEM up (with WIFI client down): starting network with MODEM
I (768308) events: Signal(network.modem.up)
I (768308) events: Signal(network.up)
I (768318) events: Signal(network.interface.change)
D (768318) netmanager: MongooseTask starting
I (768338) events: Signal(network.mgr.init)
I (768358) ssh: Launching SSH Server
I (768368) ovms-server-v2: Status: Network is up, so attempt network connection
…
I (775738) ovms-server-v2: Status: OVMS V2 login successful, and crypto channel established
I (775748) events: Signal(server.v2.connected)
```

So something is special about your cellular network loss or module setup.

You should enable debug logging for events and netmanager, so you can get a more detailed view of the processing sequence. To get rid of the UART frame error logs, simply set log level error for `cellular`.

joshuajung, 2025-02-17:
Thanks for looking into this, @dexterbg. I'll try some more things and report back. Being able to simulate NetLoss will make this a lot easier. 

Just for clarification, I am running my module with WiFi in client mode, and the issue still occurs.

joshuajung, 2025-02-19:
To provide an update, I have created a debugging build of the firmware with additional logging in NetManager. 

My initial observation is that `OvmsNetManager::StartMongooseTask` ends early because `m_network_any` is supposedly `false`, despite `m_connected_modem` having been set to `true` by `ModemUp` very shortly before. Maybe this is some kind of race condition/concurrency issue due to `tcpip_callback_with_block`, which would also explain why this is not 100% reproducible?

I have added some more specific logging in a new build and will report back. Unfortunately, this is a bit tricky to simulate at home.

Also, I want to add that there seem to be numerous reports of this issue from the rapidly growing user base at https://www.smart-emotion.de/forum/board/79-ovms-unabhängiger-appzugriff/, so this may not be a super exotic edge case.

dexterbg, 2025-02-19:
You just triggered my memory of a potential issue I saw with `tcpip_callback_with_block`, discussed only in an email conversation with @markwj and Derek Caudwell following this thread: http://lists.openvehicles.com/pipermail/ovmsdev/2024-March/016369.html

Dereks effect resembles yours. Derek suspected some DNS timing issue blocking the reconnect. Mark suspected an error in my rework of the V2 server 60 second reconnect interval. I verified the code and found no issue with that, but along that found this:

> My impression from checking the code now is, that it's potentially race critical the server v2 relies on the `MyNetManager.m_connected_any state` variables in its event handling. The state variables can now be changed _after_ the network event has been processed, because the netmanager's `PrioritiseAndIndicate()` controlling these state variables is now an async callback in the TCP/IP context.
> 
> Note: `tcpip_callback_with_block()` is a bit misleading -- "block" here means it will block until the callback request is posted to the queue, _not_ until the callback has been executed.
> 
> I.e.: netmanager does `WifiConnect()`, starts `PrioritiseAndIndicate()` then emits `network.mgr.init`. Server V2 `NetmanInit()` checks `MyNetManager.m_connected_any`, which hasn't been updated yet, so doesn't do anything.
> 
> That could happen especially when the TCP/IP stack is busy, which seems to apply to Derek's situation.

Our email discussion ended there, so the issue got forgotten.

Chances are the refactoring of `PrioritiseAndIndicate()` to be executed in the LwIP context was meant to just change the context, not to have it executing asynchronously in the first place, as it was synchronous before (just within the netmanager task).

So a potential solution could be to simply introduce a semaphore for `PrioritiseAndIndicate()` to block until the callback has been executed.

I've just tested this locally, it doesn't seem to produce any unwanted effect. Please apply the attached patch & try if that solves your issue.

Regards,
Michael

[patch-netman.diff.txt](https://github.com/user-attachments/files/18875655/patch-netman.diff.txt)

joshuajung, 2025-02-20:
Ha, that sounds promising!

My additional logging hinted at the same thing this morning:

```
2025-02-20 07:26:23.754 CET D (83924) netmanager: JAJ Checkpoint: ModemUp 1
2025-02-20 07:26:23.754 CET D (83924) netmanager: JAJ Checkpoint: OvmsNetManager::SafePrioritiseAndIndicate
2025-02-20 07:26:23.754 CET D (83924) netmanager: JAJ Checkpoint: ModemUp 2
2025-02-20 07:26:23.754 CET I (83924) netmanager: MODEM up (with WIFI client down): starting network with MODEM
2025-02-20 07:26:23.754 CET D (83924) netmanager: JAJ Checkpoint: StartMongooseTask 1
2025-02-20 07:26:23.754 CET D (83924) netmanager: JAJ Checkpoint: StartMongooseTask 2
2025-02-20 07:26:23.754 CET D (83924) events: Signal(network.modem.up)
2025-02-20 07:26:23.754 CET D (83924) events: Signal(network.up)
2025-02-20 07:26:23.754 CET I (83924) time: Starting SNTP client
2025-02-20 07:26:23.754 CET D (83924) events: Signal(network.interface.change)
2025-02-20 07:26:23.764 CET D (83934) netmanager: JAJ Checkpoint: SafePrioritiseAndIndicate
2025-02-20 07:26:23.764 CET D (83934) netmanager: JAJ Checkpoint: DoSafePrioritiseAndIndicate 1
2025-02-20 07:26:23.764 CET D (83934) netmanager: JAJ Checkpoint: DoSafePrioritiseAndIndicate 2
2025-02-20 07:26:23.764 CET D (83934) netmanager: JAJ Checkpoint: DoSafePrioritiseAndIndicate 3
```

So the whole `SafePrioritiseAndIndicate` didn't run until `StartMongooseTask` had already given up.

Just flashed my module with your diff and will let you know how it goes in the next days.

joshuajung, 2025-02-21:
So it's only been a day, but I've been on the road a lot and the module always recovered perfectly fine. 👍

Logs also indicitate that`DoSafePrioritiseAndIndicate` now always completes before `StartMongooseTask` runs.

joshuajung, 2025-02-21:
More positive feedback: https://www.smart-emotion.de/forum/thread/3076-lte-wlan-und-gsm-der-allgemeine-fragen-thread-rund-um-die-einrichtung-aller-funk/?pageNo=18

dexterbg, 2025-02-21:
Joshua, that link points to a ressource that's not available for visitors:
> Dieses Thema enthält 247 weitere Beiträge, die nur für registrierte Benutzer sichtbar sind.

Maybe you can give a summary of the feedback?

joshuajung, 2025-02-21:
Ah, apologies. I supplied my patched firmware to another member who had been struggling for weeks with OVMS not reconnecting after the vehicle had been parked in an underground parking lot for a few hours. With the patched firmware, the issue disappeared for him.

Our module has now been having a stable connection for two days as well, I have rarely seen this before. So I'm pretty confident that we found the root cause and a fix.

dexterbg, 2025-02-21:
OK, I'll go ahead and commit the change.

dexterbg, 2025-02-21:
Thanks for helping in debugging this.

dexterbg, 2025-02-21:
Published in edge build 3.3.004-383-ge057c404

closed 2025-02-21 by dexterbg