openvehicles/Open-Vehicle-Monitoring-System-3 #1107: LogTask: writing to file failed, terminating

joshuajung, 2025-02-25:
Hello again, back with another issue that I couldn't find earlier mentions about.

## Issue
Since initial installation, my module has regularly aborted logging. Today, I was able to capture a related error message in the web shell by chance:
```
E (14028190) command: LogTask: writing to file failed, terminating
```

This happens...
* irrespective of the log target (SD/store partition)
* irrespective of the log level
* sometimes within seconds of enabling the log, sometimes after many hours
* even after a complete module factory reset

After logging has aborted, I can easily restart it with `log open`, and then it will abort again after a while.

## Full sample log (from the web shell)

```
OVMS# log status
Log listeners      : 3
File logging status: inactive
  Log file path    : /store/logs/logs
  Current size     : 896.2 kB
  Cycle size       : 1024 kB
  Cycle count      : 0
  Dropped messages : 0
  Messages logged  : 2
  Total fsync time : 0.0 s

OVMS# log open
File logging to '/store/logs/logs' started
I (14023470) command: OpenLogfile: now logging to file '/store/logs/logs'

// This happened very quickly after by chance, usually logging stay active for a while and doesn't abort immediately
E (14028190) command: LogTask: writing to file failed, terminating

OVMS# log status
Log listeners      : 3
File logging status: inactive
  Log file path    : /store/logs/logs
  Current size     : 896.2 kB
  Cycle size       : 1024 kB
  Cycle count      : 0
  Dropped messages : 0
  Messages logged  : 2
  Total fsync time : 0.1 s

OVMS# log open
File logging to '/store/logs/logs' started
I (14135510) command: OpenLogfile: now logging to file '/store/logs/logs'

// Logging is now on
```

## Environment
* Smart EQ
* OVMS module as supplied by @Dimitrie78 via [smart EMOTION](https://www.smart-emotion.de/shop/product/75-ovms-unabh%C3%A4ngiger-appzugriff/)
* Newest edge firmware
* dexters-web.de server
* Telekom SIM card

Kindly let me know in case there is anything I can do to help debugging this, I'm a bit at a loss.

dexterbg, 2025-02-26:
Well, first of all, logging on `/store` is a really bad idea:
```
OVMS# vfs df
/store:  Size:    976 kB  Avail:    424 kB  Used:  57%
/sd   :  Size:  30424 MB  Avail:   8118 MB  Used:  74%
```

Your achieved log size is about as high as it gets on a mostly empty `/store`.

dexterbg, 2025-02-26:
If you get errors when logging to `/sd`, you should check the SD card filesystem for errors, and try if tuning the SD timing helps:
https://docs.openvehicles.com/en/latest/userguide/logging.html#performance-impact

joshuajung, 2025-02-27:
Thanks @dexterbg, you were obviously right about the `store` partition being full. 🙄 So it may still be an SD card issue. 

I'll try as you suggested and report back in the next days.

joshuajung, 2025-02-27:
Okay, assuming that logging to the store only failed due to lack of space (duh), I'm back to triaging the issue with SD logging.

I was able to catch this message in the web shell:

```
E (9483601) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (9483601) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (9483611) command: LogTask: writing to file failed, terminating
```

As mentioned before, this happens with various SD cards. My current one is a brand new class 10 card, freshly formatted. Health check on my Mac found no issues. 

```
OVMS# sd status
SD CARD is inserted
Status: available
Name: SD
Size:   3727 MB
Free:   3711 MB

Card type: SDHC/SDXC
Max speed: 25000 kHz
Capacity: 3728 MB
CSD: ver=1, sector_size=512, capacity=7634944 read_bl_len=9
SCR: sd_spec=2, bus_width=5
```

You talked about tuning SD timing – were you referring to `maxfreq.khz`? Documentation suggests raising it to 20000 kHz, but I seem to be at 25000 kHz already.

dexterbg, 2025-02-27:
`sd status` tells about the SD card properties, while `maxfreq.khz` sets the host controller limit. The host limit default is 16000 kHz, as documented you can try 20000. You need to `sd unmount` & `sd mount` for the change to apply.

Att: values higher than 20000 may result in crashes with an SD card inserted, so don't try that without physical access to the module.

joshuajung, 2025-03-03:
Thank you @dexterbg. I raised `maxfreq.khz` to 20.000 kHz and am still seeing the same issue:

```
E (1189631) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x109
E (1189631) diskio_sdmmc: sdmmc_write_blocks failed (265)
E (1189641) command: LogTask: writing to file failed, terminating
```

Do you think there is any value in pushing this even higher? Could this be a hardware issue as well? I'm way out of my depth here.

dexterbg, 2025-03-03:
Yes, if that happens with multiple different SD cards, that's most probably a hardware issue, unless you are experimenting with some new code involving long critical sections (long interrupt blocking) or high loads for tasks running at high priority.

If you don't, I'd first have a very close look at the module PCB and connectors. Maybe something has become lose, or you've got a cold soldering point.

joshuajung, 2025-03-03:
Thank you! It's a very plain setup otherwise and I've been trying four cards at this point. I will try another module with the same configuration/card and report back or close the issue.

joshuajung, 2025-03-10:
This was solved by a hardware replacement (same config/SD card). I'll leave the hardware post mortem to people smarter than me. 🙂

Thanks again, closing the issue.

closed 2025-03-10 by joshuajung