openvehicles/Open-Vehicle-Monitoring-System-3 #1109: Problem with cellular connection failing repeatedly

rbebeau, 2025-03-05:
I have been having issues with the cellular connection failing even in an area where there should be good signal quality.

### Environment

    2013 Nissan Leaf
    OVMS module
    Firmware 3.3.004-396-geb3d1636
    OVMS v3 MQTT server to my own system
    Hologram SIM card

### Issue

 I have been experiencing a problem ever since I received the new OVMS system a month ago.   I am using a hologram SIM card and have tried to debug the issue with them, but they were not able to help.  It seems that OVMS tries to establish a PPP connection but after establishing the connection 'NO CARRIER' is received and then after that a loop occurs and the connection is not reestablished.

I set 'log level debug cellular' to try to see what's going on, but I don't see anything obvious on my side to know why 'NO CARRIER' happens after it does establish the PPP session sucessfully.

### Sample Logs

I am showing a session where I start up the modem (from shut down state) to see the PPP session establishing and then immediately getting a 'NO CARRIER' followed by the looping.

If there are further commands or logs that would help to you (or me) to debug the issue, please let me know.

````
D (55193464) cellular: State timeout PowerOffOn => PoweringOn
I (55193464) cellular: State: Enter PoweringOn state
I (55193464) cellular-modem-auto: Power Cycle 4000ms
D (55197464) cellular: tx-cmd: AT
D (55197464) cellular: tx-cmd: AT
D (55197464) cellular: tx-cmd: AT
D (55197464) cellular: tx-cmd: AT
D (55198464) cellular: tx-cmd: AT
D (55199464) cellular: tx-cmd: AT
D (55200464) cellular: tx-cmd: AT
D (55201464) cellular: tx-cmd: AT
D (55202464) cellular: tx-cmd: AT
D (55203464) cellular: tx-cmd: AT
D (55204464) cellular: tx-cmd: AT
D (55205464) cellular: tx-cmd: AT
D (55206464) cellular: tx-cmd: AT
D (55207464) cellular: tx-cmd: AT
D (55208464) cellular: tx-cmd: AT
I (55208554) cellular: State: Enter Identify state
D (55209464) cellular: tx-cmd: AT+CGMM
I (55209474) cellular: Identified cellular modem: SIM7600/Experimental support for SIMCOM SIM7600
D (55209474) cellular: Remove old 'auto' modem driver
I (55209474) cellular: Set modem driver to 'SIM7600'
I (55209474) cellular: State: Enter PoweredOn state
D (55210664) cellular: mux-rx-line #0 (2/16): OK
D (55210664) cellular: mux-rx-line #0 (12/0): +CPIN: READY
D (55211884) cellular: mux-rx-line #0 (8/0): SMS DONE
D (55219464) cellular: tx-cmd: AT+CPIN?;+CREG=1;+CGREG=1;+CEREG=1;+CTZU=1;+CTZR=1;+CLIP=1;+CMGF=1;+CNMI=1,2,0,0,0;+CSDH=1;+CMEE=2;+CSQ;+AUTOCSQ=1,1;E0;S0=0
D (55219494) cellular: mux-rx-line #0 (124/37): AT+CPIN?;+CREG=1;+CGREG=1;+CEREG=1;+CTZU=1;+CTZR=1;+CLIP=1;+CMGF=1;+CNMI=1,2,0,0,0;+CSDH=1;+CMEE=2;+CSQ;+AUTOCSQ=1,1;E0;S0=0
D (55219494) cellular: mux-rx-line #0 (12/21): +CPIN: READY
D (55219494) cellular: mux-rx-line #0 (11/6): +CSQ: 99,99
D (55219494) cellular: mux-rx-line #0 (2/0): OK
D (55221464) cellular: tx-cmd: AT+CGMR;+ICCID
D (55221484) cellular: mux-rx-line #0 (22/38): +CGMR: LE20B05SIM7600G
D (55221484) cellular: mux-rx-line #0 (28/6): +ICCID: 89464XXXXXXXXXXXXX
D (55221484) cellular: mux-rx-line #0 (2/0): OK
D (55229464) cellular: tx-cmd: AT+CMUX=0;+CATR=6
D (55229474) cellular: mux-rx-line #0 (2/0): OK
I (55229474) cellular: State: Enter MuxStart state
I (55229474) gsm-mux: Start MUX
I (55229484) gsm-mux: Channel #0 is open
I (55229494) gsm-mux: Channel #1 is open
I (55229504) gsm-mux: Channel #2 is open
I (55229514) gsm-mux: Channel #3 is open
I (55229524) gsm-mux: Channel #4 is open
D (55230464) cellular: State transition MuxStart => NetWait
I (55230464) cellular: State: Enter NetWait state
[... 131 lines cut ...]
D (55323464) cellular: mux-tx #3: AT+CREG?;+CGREG?;+CEREG?;+CCLK?;+CSQ
D (55323484) cellular: mux-rx-line #3 (10/84): +CREG: 1,2
D (55323494) cellular: mux-rx-line #3 (11/69): +CGREG: 1,2
D (55323494) cellular: mux-rx-line #3 (11/54): +CEREG: 1,4
D (55323494) cellular: mux-rx-line #3 (29/21): +CCLK: "25/03/05,10:52:31-20"
D (55323494) cellular: mux-rx-line #3 (11/6): +CSQ: 99,99
D (55323494) cellular: mux-rx-line #3 (2/0): OK
D (55324464) cellular: mux-tx #3: AT+CPSI?
D (55324474) cellular: mux-rx-line #3 (24/0): +CPSI: NO SERVICE,Online
D (55324474) cellular: mux-rx-line #3 (2/0): OK
D (55325464) cellular: mux-tx #3: AT+COPS?
D (55325474) cellular: mux-rx-line #3 (8/6): +COPS: 0
D (55325474) cellular: mux-rx-line #3 (2/0): OK
D (55333464) cellular: mux-tx #3: AT+CREG?;+CGREG?;+CEREG?;+CCLK?;+CSQ
D (55333484) cellular: mux-rx-line #3 (10/84): +CREG: 1,2
D (55333484) cellular: mux-rx-line #3 (11/69): +CGREG: 1,2
D (55333484) cellular: mux-rx-line #3 (11/54): +CEREG: 1,4
D (55333484) cellular: mux-rx-line #3 (29/21): +CCLK: "25/03/05,10:52:41-20"
D (55333484) cellular: mux-rx-line #3 (11/6): +CSQ: 99,99
D (55333484) cellular: mux-rx-line #3 (2/0): OK
D (55334464) cellular: mux-tx #3: AT+CPSI?
D (55334474) cellular: mux-rx-line #3 (24/0): +CPSI: NO SERVICE,Online
D (55334474) cellular: mux-rx-line #3 (2/0): OK

.... loop continues....

zorgms, 2025-03-06:
In the community www.smart-EMOTION.de reported some user the same issue since #1105 .

dexterbg, 2025-03-06:
That's probably not a software issue. `+CPSI: NO SERVICE,Online` shouldn't normally happen, my Hologram SIM immediately starts from a cold boot with a GSM connection (`+CPSI: GSM,Online,262-02,0x011f,18121,6 EGSM 900,-67,0,0-0`) and switches to LTE within 30 seconds (`+CPSI: LTE,Online,262-02,0xB092,13250579,154,EUTRAN-BAND1,100,5,5,-149,-1056,-709,8`).

The signal level isn't too bad, which is puzzling. I wouldn't exclude an issue with the SIM card / provider. I assume you double checked the SIM's registration & account balance, so try a different SIM card. And try the Hologram SIM in another device, e.g. your mobile phone.

Another possible cause is some half broken/lose wire at the antenna connection. Try a different antenna, and check the internal antenna connection ("pigtail cable") running from the connector to the modem.

Regards,
Michael

dexterbg, 2025-03-27:
AFAICT your log says you've got a good cellular signal level (+CSQ), but none of the cellular providers offer to register you (+CREG,+CGREG,+CEREG,+CPSI). Even with a zero balance, I normally see the network registration happen.

What bugs me is your modem's reported revision code:
```
D (731389) cellular: mux-rx-line #0 (22/38): +CGMR: LE20B05SIM7600G
```

That doesn't match my two units, which show `LE20B03SIM7600M21-A`. Yours _may_ be just a more recent firmware version, but let's compare the full modem info, just to be sure you haven't got a wrong SIMCOM7600 variant by accident:

```
OVMS# cell cmd ATI
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G
Revision: SIM7600M21-A_V2.0
IMEI: 8688xxxxxxxxxxxx
+GCAP: +CGSM
OK

OVMS# cell cmd ATI
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G
Revision: SIM7600M21-A_V2.0.1
SVN: 01
IMEI: 8688xxxxxxxxxxxx
+GCAP: +CGSM
OK
```

The only page I can find on the web mentioning your revision code is another OVMS user's metrics dump, and that also doesn't show a cellular network…

dexterbg, 2025-03-27:
Can you also please query the preferred mode & band selection?
```
OVMS# cell cmd AT+CNMP?
+CNMP: 2
OK

OVMS# cell cmd AT+CNBP?
+CNBP: 0x100200000EE80380,0x480000000000000000000000000000000000000000000042000007FFFFDF3FFF,0x000000000000003F
OK
```

rbebeau, 2025-03-27:
Thanks for the speedy followup.

Here are the results of the commands...

```
OVMS# cell cmd ATI
Manufacturer: SIMCOM INCORPORATED
Model: SIMCOM_SIM7600G
Revision: SIM7600G_V2.0.2
IMEI: 8688XXXXXXXXXXX
+GCAP: +CGSM
OK
```

```
OVMS# cell cmd AT+CNMP?
+CNMP: 2
OK
```

```
OVMS# cell cmd AT+CNBP?
+CNBP: 0x100200000EE80380,0x480000000000000000000000000000000000000000000042000007FFFFDF3FFF,0x000000000000003F
OK
```

dexterbg, 2025-03-27:
That's the correct modem version (7600G = global edition, supports all frequency bands) and mode & band selections are correct as well (auto = allow all).

You've got a more recent firmware, but I've meanwhile found other users running that version without any issues. I've run out of ideas for now, except some internal HW issue of your modem/module.

closed 2025-09-27 by rbebeau