openvehicles/Open-Vehicle-Monitoring-System-3 #1115: 2015 Leaf stop charging issue

umbjm1, 2025-03-20:
After setting up OVMS, I tested the charging stop several times by increasing "sufficient SOC". While I got feedback with this value appearing on the Battery page in the Android app, charging repeatedly failed to stop at the chosen percentage. Rebooting OVMS helped once.

However, in normal use (disconnect wall charger, drive car, return home, reconnect wall charger) it has worked every time.

It therefore seems that something (OVMS, the car, or the charger in the car, or some combination of these) needs to be power cycled for this to work. Can anyone tell me which one is the key? 

I ask because my failures in testing are relevant to increasing the sufficient SOC while charging, something I see myself doing frequently.

wayne-love, 2025-03-28:
I just tried to replicate this on a ZE1 2019 40kWh via the web interface (I dont use the app) and was not able to do so.  Do you get a different result if you use the web UI?

umbjm1, 2025-03-29:
No, it happened both interfaces. I have since done a factory reset and not had the problem.

enoch85, 2025-03-29:
> not had the problem

So I guess the issue can be closed then?

umbjm1, 2025-03-30:
I think so.

enoch85, 2025-03-31:
@dexterbg

closed 2025-04-04 by dexterbg