openvehicles/Open-Vehicle-Monitoring-System-3 #1175: Smart EQ periodically sending climate notifications after update to 3.3.005-1

dexterbg, 2025-07-27:
@zorgms / @zbchristian

A user just reported another issue/regression on the Smart EQ, occuring on the "main" release:

Since OTA firmware 3.3.005-1-geed58d6f was installed on my Smart ForTwo EQ, I am constantly getting messages from the server stating:
```
Climate values:
  on: no
  time: 5:15 h
  weekly: no
  start Day: Mon
  end Day: Fri
  runtime: 5 min
```
This appears to be giving me no useful information and using up my sim card data.  How do I stop this message?

If you need more details, contact me by mail to get the user's email.

zorgms, 2025-07-27:
This is a warning: If you use or have enabled time-controlled climate control, you'll see this message because the settings reset to default after rebooting.
You can enable/disable it on the Features website -> Diff Settings -> Enable Climate System

denewbery, 2025-07-28:
I don't know what you are referring to by the "Features website". I have been using the OVMS app on my phone and the dexters-web.de website.  Where is the "Features" website.

zorgms, 2025-07-28:
You can also change the setting via shell:
Log in to dexters-web.de
Click the shell button
Enter this:
`config set xsq climate.system no`
Then click send.

Or send it via the app via message page:
`config set xsq climate.system no`

More information about smart EQ app topics can also be found here:
https://www.smart-emotion.de/forum/board/79-ovms-unabh%C3%A4ngiger-appzugang/

denewbery, 2025-07-28:
Thanks – I’ve done that and ‘ll see if it works.

Thank you.

 

Regards

 

David Newbery

 

https://www.smart-emotion.de/forum/board/79-ovms-unabh%C3%A4ngiger-appzugang/

zorgms, 2025-08-25:
#1185

zorgms, 2025-08-25:
No, but most smart users who had problems with crash-restarts were in APclient mode and using Wi-Fi/modem. After switching to client mode, the problems were often resolved. I personally manage four modules in smarts with only a modem connection; they don't have any restarts and run continuously without problems.

zorgms, 2025-09-06:
I don't understand what to do right now. Maybe the app rework was a bit too much.

dexterbg, 2025-09-07:
I just did the backport for you, please verify my commit to branch "release-3.3.005" or test the updated "main" build `3.3.005-3-g4f3f7688`.