openvehicles/Open-Vehicle-Monitoring-System-3 #1203: [3.3.005-130-gd2e63edb] MQTT Data are too old

lorbetzki, 2025-09-19:
Hi,
Since the changes to server v.3, it takes forever (20-30 minutes) for data such as v.b.soc to be sent to the MQTT server.

zorgms, 2025-09-19:
#1204 will fix this.

zorgms, 2025-09-19:
I checked again because of monotonic, for me, it's transmitted at every interval. 
Note: only modified data is transmitted. 
Starting with #1202, you can query specific metrics via MQTT, and starting with #1204, you can prioritize your own metrics and retrieve them with `server v3 update priority`. 
Would that be a solution for you?

lorbetzki, 2025-09-21:
Yes, my host cannot be reached via ping. That is correct. I use eclipse-mosquitto v.2.0.22 on docker. But you could be right about a routing problem. 

Today, I noticed that the values arrive regularly when I'm on the move. The errors only occur when the vehicles are within range of the Wi-Fi. So it could really be a routing/DNS problem. It has nothing to do with MQTT.

Please excuse me and thank you for pointing me in the right direction.